Igalia/meta-webkit #182: wpewebkit + qtwpe + linuxfb is it a supported combination.

kraj, 2020-05-08:
I am trying a case where I do not have OpenGL but a simple LCD shield connected over SPI interface, and I am using linuxfb QT backend, I then added `qt-wpe-simple-browser` to my image ensuring `qtwpe` PACKAGECONFIG is enabled in wpewebkit, it all builds fine, however when I launch it on device, it ends up with abort.

Here is a gdb trace

```
Starting program: /usr/bin/qt-wpe-simple-browser https://webkit.org
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/libthread_db.so.1".
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
qt.qpa.input: libinput: event2  - Logitech USB Optical Mouse: is tagged by udev as: Mouse
qt.qpa.input: libinput: event2  - Logitech USB Optical Mouse: device is a pointer
qt.qpa.input: libinput: event3  - Macally Peripherals Macally ICEKey keyboard: is tagged by udev as: Keyboard
qt.qpa.input: libinput: event3  - Macally Peripherals Macally ICEKey keyboard: device is a keyboard
qt.qpa.input: libinput: event4  - Macally Peripherals Macally ICEKey keyboard: is tagged by udev as: Keyboard
qt.qpa.input: libinput: event4  - Macally Peripherals Macally ICEKey keyboard: device is a keyboard
qt.qpa.input: libinput: event0  - aml_vkeypad: is tagged by udev as: Keyboard
qt.qpa.input: libinput: event0  - aml_vkeypad: device is a keyboard
qt.qpa.input: libinput: event5  - ADS7846 Touchscreen: is tagged by udev as: Touchscreen
qt.qpa.input: libinput: event5  - ADS7846 Touchscreen: device is a touch device
qt.qpa.input: libinput: event1  - cec_input: is tagged by udev as: Keyboard
qt.qpa.input: libinput: event1  - cec_input: device is a keyboard
qt.qpa.input: Using xkbcommon for key mapping
qt.qpa.input: libinput: registerDevice /dev/input/event5 - ADS7846 Touchscreen
[New Thread 0x7fb5fe7180 (LWP 3966)]
[Switching to Thread 0x7fb5fe7180 (LWP 3966)]

Thread 2 "QQmlThread" hit Breakpoint 1, 0x0000007fb47e51a4 in WPEQmlExtensionPlugin::registerTypes(char const*)@plt () from /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
(gdb) c
Continuing.

Thread 2 "QQmlThread" hit Breakpoint 1, 0x0000007fb47e876c in WPEQmlExtensionPlugin::registerTypes(char const*) () from /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
(gdb)
Continuing.

Thread 1 "qt-wpe-simple-b" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fb648afa0 (LWP 3962)]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000007fb6d22cc0 in __GI_abort () at abort.c:79
#2  0x0000007fb47e8758 in WPEQmlExtensionPlugin::registerTypes(char const*) () at /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
#3  0x00000000003f4800 in  ()
(gdb)
```

The launch is done like this

```
export QT_QPA_EGLFS_FB=/dev/fb4
export QT_QPA_PLATFORM=linuxfb:fb=/dev/fb4
export QT_QPA_DEBUG=1
export QT_LOGGING_RULES="qt.qpa.*=true"
export QML2_IMPORT_PATH=/usr/lib/qt5/qml
#qt-wpe-simple-browser -platform linuxfb:fb=/dev/fb4 http://webkit.org
gdb --arg qt-wpe-simple-browser https://webkit.org
```

I could use some help in finding if this usecase is something which is supported in wpewebkit?
since I know qtwebengine does work in this setup, I was hoping wpewebkit would work too since
I dont want to use qtwebengine.

philn, 2020-05-08:
The trace is incomplete, please enable qtwpe debug symbols.

kraj, 2020-05-08:
I saw that but it does not see anything there but if I disassemble 4 bytes before 0x0000007fb43c0a04 at 0x0000007fb43c0a00 then it points to 
```
Dump of assembler code for function _ZN9WPEQtView13createWebViewEv:
   0x0000007fb43a7500 <+0>:     stp     x29, x30, [sp, #-144]!
   0x0000007fb43a7504 <+4>:     mov     x29, sp
   0x0000007fb43a7508 <+8>:     stp     x19, x20, [sp, #16]
   0x0000007fb43a750c <+12>:    adrp    x20, 0x7fb43d1000
   0x0000007fb43a7510 <+16>:    mov     x19, x0
   0x0000007fb43a7514 <+20>:    ldr     x1, [x20, #3840]
```
which seems to be `WPEQtView::createWebView()` when unmangled

when I disassemble at 0x0000007fb43c0a10 I see
```
Dump of assembler code for function _ZN9WPEQtViewC2EP10QQuickItem:
   0x0000007fb43a7a10 <+0>:     stp     x29, x30, [sp, #-96]!
   0x0000007fb43a7a14 <+4>:     mov     x29, sp
   0x0000007fb43a7a18 <+8>:     stp     x19, x20, [sp, #16]
   0x0000007fb43a7a1c <+12>:    adrp    x20, 0x7fb43d1000
   0x0000007fb43a7a20 <+16>:    mov     x19, x0
   0x0000007fb43a7a24 <+20>:    ldr     x20, [x20, #3840]
   0x0000007fb43a7a28 <+24>:    stp     x21, x22, [sp, #32]
   0x0000007fb43a7a2c <+28>:    add     x22, sp, #0x30
   0x0000007fb43a7a30 <+32>:    ldr     x2, [x20]
   0x0000007fb43a7a34 <+36>:    str     x2, [sp, #88]
   0x0000007fb43a7a38 <+40>:    mov     x2, #0x0                        // #0
   0x0000007fb43a7a3c <+44>:    mov     w21, #0x1                       // #1
   0x0000007fb43a7a40 <+48>:    bl      0x7fb43a3600 <_ZN10QQuickItemC2EPS_@plt>
```
which points to `WPEQtView::WPEQtView(QQuickItem*)`

I wonder if its some sort of PLT area or something.

philn, 2020-05-08:
Maybe this assert is hit? `RELEASE_ASSERT_WITH_MESSAGE(backend, "EGL initialization failed");`

kraj, 2020-05-08:
A bit of debugging info here

```
Thread 1 "qt-wpe-simple-b" hit Breakpoint 2, QPlatformNativeInterface::nativeResourceForIntegration (this=0x5555585b58, resource=...)
    at /usr/src/debug/qtbase/5.14.99+5.15-beta4+gitAUTOINC+806bc5fc05-r0/git/src/gui/kernel/qplatformnativeinterface.cpp:59
59      /usr/src/debug/qtbase/5.14.99+5.15-beta4+gitAUTOINC+806bc5fc05-r0/git/src/gui/kernel/qplatformnativeinterface.cpp: No such file or directory.
(gdb) s
0x0000007fb43a758c in WPEQtView::createWebView() () from /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
(gdb)
Single stepping until exit from function _ZN9WPEQtView13createWebViewEv,
which has no line number information.
QArrayData::deallocate (data=0x555564daf0, objectSize=1, alignment=8) at /usr/src/debug/qtbase/5.14.99+5.15-beta4+gitAUTOINC+806bc5fc05-r0/git/src/corelib/tools/qarraydata.cpp:264
264     /usr/src/debug/qtbase/5.14.99+5.15-beta4+gitAUTOINC+806bc5fc05-r0/git/src/corelib/tools/qarraydata.cpp: No such file or directory.
(gdb)
270     in /usr/src/debug/qtbase/5.14.99+5.15-beta4+gitAUTOINC+806bc5fc05-r0/git/src/corelib/tools/qarraydata.cpp
(gdb)
__GI___libc_free (mem=0x555564daf0) at malloc.c:3092
3092    malloc.c: No such file or directory.
(gdb) finish
Run till exit from #0  __GI___libc_free (mem=0x555564daf0) at malloc.c:3092
0x0000007fb43a7828 in WPEQtView::createWebView() () from /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
(gdb) s
Single stepping until exit from function _ZN9WPEQtView13createWebViewEv,
which has no line number information.
QQuickItem::window (this=0x55555a9ac0) at /usr/src/debug/qtdeclarative/5.14.99+5.15-beta4+gitAUTOINC+6555642db7-r0/git/src/quick/items/qquickitem.cpp:2858
2858    /usr/src/debug/qtdeclarative/5.14.99+5.15-beta4+gitAUTOINC+6555642db7-r0/git/src/quick/items/qquickitem.cpp: No such file or directory.
(gdb)
0x0000007fb43a75c4 in WPEQtView::createWebView() () from /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
(gdb)
Single stepping until exit from function _ZN9WPEQtView13createWebViewEv,
which has no line number information.
QQuickWindow::openglContext (this=0x5555592800) at /usr/src/debug/qtdeclarative/5.14.99+5.15-beta4+gitAUTOINC+6555642db7-r0/git/src/quick/items/qquickwindow.cpp:3840
3840    /usr/src/debug/qtdeclarative/5.14.99+5.15-beta4+gitAUTOINC+6555642db7-r0/git/src/quick/items/qquickwindow.cpp: No such file or directory.
(gdb) finish
Run till exit from #0  QQuickWindow::openglContext (this=0x5555592800) at /usr/src/debug/qtdeclarative/5.14.99+5.15-beta4+gitAUTOINC+6555642db7-r0/git/src/quick/items/qquickwindow.cpp:3840
0x0000007fb43a75c8 in WPEQtView::createWebView() () from /usr/lib/qt5/qml/org/wpewebkit/qtwpe/libqtwpe.so
Value returned is $2 = (QOpenGLContext *) 0x0
```
As we can see `openglContext` returns null and is then used in WPEQtViewBackend::create call
which then causes SIGABRT

below is relevant code from Source/WebKit/UIProcess/API/wpe/qt/WPEQtView.cpp

```c++
void WPEQtView::createWebView()
  {
      if (m_backend)
          return;

      auto display = static_cast<EGLDisplay>(QGuiApplication::platformNativeInterface()->nativeResourceForIntegration("egldisplay"));
      auto* context = window()->openglContext();
      std::unique_ptr<WPEQtViewBackend> backend = WPEQtViewBackend::create(m_size, context, display, QPointer<WPEQtView>(this));
      RELEASE_ASSERT_WITH_MESSAGE(backend, "EGL initialization failed");
      if (!backend)
          return;

```

kraj, 2020-05-08:
@philn I am using `linuxfb` QT backend not `eglfs` so the question is does qtwpe port supposed to work with linuxfb ?

philn, 2020-05-08:
I have no idea. All I know is that it works with EGLFS and Wayland-EGL QPAs.

philn, 2020-05-08:
is there any EGL support in that linuxfb thing?

kraj, 2020-05-08:
> is there any EGL support in that linuxfb thing?

linuxfb uses the linux frame buffer in fullscreen mode. 
there is directfb which uses the linux frame buffer with OpenGL ES via the directfb layer.
do you think that stands a better chance ?

since qtwebegine works fine with linuxfb I thought it was workable in wpewebkit as well but perhaps needs some work maybe

philn, 2020-05-08:
Well, if the linuxfb qpa has an egldisplay, i guess it could work https://trac.webkit.org/browser/webkit/trunk/Source/WebKit/UIProcess/API/wpe/qt/WPEQtView.cpp#L89

Otherwise some people are working on a DFB backend: https://github.com/nagra-opentv/WPEBackend-rdk

closed 2021-05-16 by github-actions[bot]