Igalia/meta-webkit #183: Failed to render on the iMX8

pcjustin, 2020-05-18:
Hi,

I wanna use webkit to play streaming on the iMX8 but the render was fail.

The following command will work.
```
$ gst-launch-1.0 filesrc location=shinjuku.mp4 ! video/quicktime ! aiurdemux ! queue ! h264parse ! v4l2h264dec ! imxvideoconvert_g2d ! queue ! waylandsink
```

When trying to use cog to play a h264 video fails.
```
$ cog https://people.igalia.com/clopez/wkbug/video/simplevideo.html
```

Here's my configuration:
```
DISTRO ?= 'fsl-imx-xwayland'
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-rdk"
PACKAGECONFIG_pn-wpebackend-rdk = "wayland"
IMAGE_INSTALL_append = " wpewebkit cog"
```

How can I add the `imxvideoconvert_g2d` into the pipeline?

philn, 2020-05-18:
Do you have compelling reasons to use the proprietary vivante driver instead of etnaviv?

pcjustin, 2020-06-04:
Hi phiin,

Sorry for late to reply.

I compiled `wpewebkit` on the `imx8qxp` and it didn't support etnaviv driver.

I used `imxvideoconvert_g2d` to convert the video file and it was working properly.
```
$ GST_DEBUG_NO_COLOR=1 GST_DEBUG=*:3 GST_DEBUG_FILE=/mnt/data/gst/gst_correct.log GST_DEBUG_DUMP_DOT_DIR=/mnt/data/gst gst-launch-1.0 filesrc location=/mnt/data/shinjuku.mp4 ! video/quicktime ! aiurdemux ! queue ! h264parse ! v4l2h264dec ! imxvideoconvert_g2d ! queue ! waylandsink
```

[image]

BUT I used `videoconvert` to convert the video file and something was wrong.
```
$ GST_DEBUG_NO_COLOR=1 GST_DEBUG=*:3 GST_DEBUG_FILE=/mnt/data/gst/gst_wrong.log GST_DEBUG_DUMP_DOT_DIR=/mnt/data/gst gst-launch-1.0 filesrc location=/mnt/data/shinjuku.mp4 ! video/quicktime ! aiurdemux ! queue ! h264parse ! v4l2h264dec ! videoconvert ! queue ! waylandsink
```

[image]

[image]

I thought `cog` get the same problem.
```
WPE_INIT_VIEW_WIDTH=800 WPE_INIT_VIEW_HEIGHT=480 cog https://www.youtube.com
```

Do you have any idea to figure out this problem?

philn, 2020-06-04:
I doubt you can mix aiurdemux and v4l2h264dec in the same pipeline...

pcjustin, 2020-06-11:
HI, philn,
What would you recommend?
I followed the document(p.35) to video playback.
https://www.nxp.com/docs/en/release-note/IMX_LINUX_RELEASE_NOTES.pdf

gvalcaza, 2020-10-08:
Good afternoon,

I'm currently trying to find a way to fix this issue. As soon as I saw the glitches, I knew it was related to imxvideoconvert_g2d because I've seen those same glitches in the past and it was directly related to that NXP-specific plugin. Once that plugin is in the gstreamer pipeline, the glitches disappear, so I'm certain the solution here is to do the same.

So far I've been looking into the wpewebkit-2.28.1 source code trying to find a place where I could add that plugin into the pipeline, but I'm unfamiliar with both the gstreamer API  and the WPE WebKit source code, so I don't really know what the change should look like or when it should be used to begin with (as I'm sure it isn't needed for absolutely every gstreamer use case). Maybe it's a matter of taking the time to learn how the code works, but does anybody who is more knowledgeable have an idea on where this change could be implemented?

philn, 2020-10-08:
If this works and playbin plugs the vivante plugin, it work the same way in WPE...
`gst-play-1.0 --videosink glimagesink /path/to/video.mp4`
By setting this var `GST_DEBUG_DUMP_DOT_DIR=/tmp` you will get graphviz pipeline dumps that you can inspect to check the pipeline topology includes the decoder you want. or not.

gvalcaza, 2020-10-09:
Hi @philn ,

Thank you for the quick reply. I'm currently building images with the WebKit to be able to debug this with your instructions. I will compare the pipelines of gstreamer on its own with the WebKit one and get back to you ASAP

gvalcaza, 2020-10-20:
Hi @philn ,

Apologies for the delayed response, my last couple of weeks have been quite busy. Using this pipeline on an i.MX8X:
`gst-play-1.0 --videosink glimagesink /path/to/video.mp4`
Results in the same graphical glitches described in the OP of this issue. However, these glitches are fixed by manually adding the i.MX videoconvert plugin to the videosink parameter:
`gst-play-1.0 --videosink "imxvideoconvert_g2d ! queue ! glimagesink" /path/to/video.mp4`
I verified that both pipelines are identical save for the inclusion of the i.MX plugin using `GST_DEBUG_DUMP_DOT_DIR=/tmp`.

The gstreamer application provided by NXP in their BSP, **gplay**, also uses this plugin by default. Looking through the app's code, I can see they hardcode the pipeline (_imxvideoconvert_g2d ! queue ! waylandsink_) using the `gst_parse_bin_from_description()` function. I can't see that function being used anywhere in the WPE code, but I will try to find an appropriate place to add the i.MX plugin to the pipeline.

philn, 2020-10-20:
Might fit behind ifdefs in WebKit's GL video sink bin.

gvalcaza, 2020-10-22:
Hi everyone,

I managed to patch WPE webkit 2.28.1 so it adds imxvideoconvert_g2d to the default gstreamer pipeline. So far, every video I've played renders correctly:

`diff --git a/Source/WebCore/platform/graphics/gstreamer/GLVideoSinkGStreamer.cpp b/Source/WebCore/platform/graphics/gstreamer/GLVideoSinkGStreamer.cpp
index 8d8f3822..3e047ebb 100644
--- a/Source/WebCore/platform/graphics/gstreamer/GLVideoSinkGStreamer.cpp
+++ b/Source/WebCore/platform/graphics/gstreamer/GLVideoSinkGStreamer.cpp
@@ -93,11 +93,13 @@ static void webKitGLVideoSinkConstructed(GObject* object)
     ASSERT(sink->priv->appSink);
     g_object_set(sink->priv->appSink.get(), "enable-last-sample", FALSE, "emit-signals", TRUE, "max-buffers", 1, nullptr);
 
+    GstElement* imxconvert = gst_element_factory_make("imxvideoconvert_g2d", nullptr);
+    GstElement* queue = gst_element_factory_make("queue", nullptr);
     GstElement* upload = gst_element_factory_make("glupload", nullptr);
     GstElement* colorconvert = gst_element_factory_make("glcolorconvert", nullptr);
     ASSERT(upload);
     ASSERT(colorconvert);
-    gst_bin_add_many(GST_BIN_CAST(sink), upload, colorconvert, sink->priv->appSink.get(), nullptr);
+    gst_bin_add_many(GST_BIN_CAST(sink), imxconvert, queue, upload, colorconvert, sink->priv->appSink.get(), nullptr);
 
     // Workaround until we can depend on GStreamer 1.16.2.
     // https://gitlab.freedesktop.org/gstreamer/gst-plugins-base/commit/8d32de090554cf29fe359f83aa46000ba658a693
@@ -121,9 +123,9 @@ static void webKitGLVideoSinkConstructed(GObject* object)
     gst_caps_set_features(caps.get(), 0, gst_caps_features_new(GST_CAPS_FEATURE_MEMORY_GL_MEMORY, nullptr));
     g_object_set(sink->priv->appSink.get(), "caps", caps.get(), nullptr);
 
-    gst_element_link_many(upload, colorconvert, sink->priv->appSink.get(), nullptr);
+    gst_element_link_many(imxconvert, queue, upload, colorconvert, sink->priv->appSink.get(), nullptr);
 
-    GRefPtr<GstPad> pad = adoptGRef(gst_element_get_static_pad(upload, "sink"));
+    GRefPtr<GstPad> pad = adoptGRef(gst_element_get_static_pad(imxconvert, "sink"));
     gst_element_add_pad(GST_ELEMENT_CAST(sink), gst_ghost_pad_new("sink", pad.get()));
 }`

Note that I haven't added any ifdefs, for now I'm simply adding the patch to the wpewebkit Yocto recipe when compiling for imx8qxp platforms.

philn, 2020-10-22:
There might be a way to upstream this without ifdef. The player has some code already detecting imxvpudec at runtime (look for `ImxVPU` in the player), so the sink could perhaps dynamically add those elements... It might require a pad probe, I think...

pcjustin, 2020-12-30:
Hi, gvalcaza,

Thanks for your patch. It's working properly.

philn, 2020-12-30:
Leaving this open until no downstream hack is required.

philn, 2021-03-24:
Should work now. Make sure to have this patch applied: https://bugs.webkit.org/show_bug.cgi?id=223693

closed 2021-03-24 by philn