Igalia/meta-webkit #184: qt-wpe-simple-browser cannot open website

parheliamm, 2020-05-20:
Hi:
When I am trying to port wpewebkit to riscv64 with QT.
I have a trouble on verify the webkit via browser on QEMU.
Here is my confiuration:
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
IMAGE_INSTALL_append = " qtbase-plugins qtwayland-plugins"
IMAGE_INSTALL_append = " wpewebkit-qtwpe-qml-plugin qt-wpe-simple-browser"
qtwpe enabled as well.

Threre is no websit displyed after lunch as below
qt-wpe-simple-browser -platform wayland http://www.google.com
There is only "HELLO WPE" in screen.
The debug message as below:
(process:314): GLib-DEBUG: 07:18:10.504: posix_spawn avoided (child_setup specified) 
(process:314): GLib-DEBUG: 07:18:10.553: posix_spawn avoided (child_setup specified) 
(process:324): GLib-GIO-DEBUG: 07:18:11.073: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(process:324): GLib-GIO-DEBUG: 07:18:11.198: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: NetworkManager not running
(process:324): GLib-GIO-DEBUG: 07:18:11.215: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ?gio-network-monitor?

I've already enable glib-2.0 and QTwebkit works fine.
Would you please share some information with debugging?

Thanks in advance.

philn, 2020-05-20:
Which QPA are you using? QtWPE works only with EGLFS and Wayland-EGL currently.

parheliamm, 2020-05-20:
Currently I am using QT5.15-RC2.
If I set to wayland-egl, it has the same behavior with same error

philn, 2020-05-21:
If you use Wayland-EGL you need to run the browser in a wayland compositor. Is this case?

parheliamm, 2020-05-21:
> If you use Wayland-EGL you need to run the browser in a wayland compositor. Is it the case?

Yes, I leverage weston in my qemu.

philn, 2020-05-22:
QtWPE hasn't been tested with Qt 5.15 yet. Try with 5.14. I don't know how this qemu setup works anyway... do GL apps works there?

parheliamm, 2020-05-25:
> QtWPE hasn't been tested with Qt 5.15 yet. Try with 5.14. I don't know how this qemu setup works anyway... do GL apps works there?

Did you have the GL test apps name? I can try it locally.

medude, 2020-05-26:
Hi,
I have the same issue here too on a imx6 based device. My setup is:
- QT 5.12.6
- WebKit 2.28.2
- wpebackend-fdo 1.6.0
- libwpe 1.6.0

I've tested cog, and the MiniBrowser from WebKit which are running on Wayland. Those Browsers don't have this issue. I also tested the qt-wpe-simple-browser on a EGLFS and Wayland (with Weston compositor) backend, both tests showing the same result: "no webpage". I debugged the code and saw that the `IPC::handleMessage<Messages::WebPageProxy::DecidePolicyForResponse>(decoder, this, &ProvisionalPageProxy::decidePolicyForResponse);` ran into an indefinite alloc/dealloc loop.

medude, 2020-05-26:
A few more details: the `ResourceResponseBase::ResourceResponseBase()` is hanging or looping around.. @parheliamm as workaround you can try to disable bmalloc by `DUSE_SYSTEM_MALLOC=ON`. This works in my use-case. Currently I don't know what change within bmalloc breaks the qtwebview.

parheliamm, 2020-05-27:
> A few more details: the `ResourceResponseBase::ResourceResponseBase()` is hanging or looping around.. @parheliamm as workaround you can try to disable bmalloc by `-DUSE_SYSTEM_MALLOC=ON`. This works in my use-case. Currently I don't know what change within bmalloc breaks the qtwebview.

I will try it and feedback to you later.

medude, 2020-05-27:
Another quick test solution could be the `Malloc=1` environment variable listed here:
https://trac.webkit.org/wiki/EnvironmentVariables

parheliamm, 2020-05-27:
Still didn't work.
If I use cog as below, there is a segment fault

root@qemuriscv64:/# cog -P fdo https://www.baidu.com
[  269.592672] cog[346]: unhandled signal 11 code 0x1 at 0x0000000000000000 in cog[2ab7502000+4000]
[  269.593521] CPU: 1 PID: 346 Comm: cog Not tainted 5.6.14 #1
[  269.593979] epc: 0000000000000000 ra : 0000003fc7e56e82 sp : 0000003fffad52a0
[  269.594705]  gp : 0000002ab7507800 tp : 0000003fc5f06150 t0 : 0000000000000001
[  269.595310]  t1 : 0000003fc800413c t2 : 0000000000000000 s0 : 0000003fffad5330
[  269.595873]  s1 : 0000000000000002 a0 : 0000000000000000 a1 : 0000003fac0052c0
[  269.596416]  a2 : 0000000000000006 a3 : 0000002ab7508010 a4 : 0000003fac0052c0
[  269.596916]  a5 : 0000000000000000 a6 : 0000000000001000 a7 : 0000000000000000
[  269.597447]  s2 : 0000003fffad52a0 s3 : 0000003fc46e8dd8 s4 : 0000000000000000
[  269.597890]  s5 : 0000003fcb688888 s6 : 0000000000000000 s7 : 0000003fffad5350
[  269.598411]  s8 : 0000003fffad5400 s9 : 0000003fffad54e8 s10: 0000000000000002
[  269.598817]  s11: 0000003fcb688888 t3 : 0000003fc8f2d79a t4 : 0000003fc8f9ba90
[  269.599206]  t5 : 0000000000000005 t6 : ffffffffffffffff
[  269.599418] status: 8000000000006020 badaddr: 0000000000000000 cause: 000000000000000c

If I use qt browser, it still a hello wpe in title, the website still cannot displayed.

Chenxi

micdini, 2021-06-08:
Hello, (I think) I'm facing a the same issue.

Cog works fine, qt-wpe-simple-browser starts but no webpage is displayed.

I also tried as suggested above with Malloc=1 without any change.

Console output (no errors):
~~~
root@system:~# qt-wpe-simple-browser -platform wayland-egl www.google.com
Using Wayland-EGL
QStandardPaths: wrong ownership on runtime directory /run/user/1000, 1000 instead of 0
Using the 'xdg-shell' shell integration

root@system:~#
~~~

Poky: hardknott branch
Board: Raspberry Pi CM4 32bit
meta-webkit: master (2.32.1)
weston: 9.0.0
Qt: 5.12.10

Any idea?

micdini, 2021-06-10:
I added some logging to WPEQtView and WPEQtViewBackend.

The backend seems to be loaded fine (runs to the end of *create()*).
Also the view runs to the end of *createWebView()*, however the only signals emitted are:
* *notifyUrlChangedCallback()* - one time, as the url is set
* *notifyLoadProgressCallback()* - one time, after *notifyUrlChangedCallback*

Then *createWebView()* ends and nothing more.

This happens with `-DUSE_SYSTEM_MALLOC=ON` also. Cog (0.10.0) is fine in both cases.

jran-dasa, 2022-02-18:
I'm experiencing the exact same result as above.
Running on i.MX8X with 
- Yocto Hardknott
- Qt 5.15.2
- wpebackend-fdo
- meta-webkit main branch

I can run Cog with no problems, and the performance of that is showing very promising performance compared to Webengine currently used in my Qt application, so I want to try switching over to WPEWebkit. 
But when running qt-wpe-simple-browser, all I get is a blank screen with "Hello WPE" in the top bar.

Also previously tried with Yocto Dunfell, Qt 5.14.2, exactly the same result regarding qt-wpe-simple-browser.
I have also tried `-DUSE_SYSTEM_MALLOC=ON`, which made no improvement.

Are there any more suggestions what I could try?

jran-dasa, 2023-01-23:
I'm still interested in a solution.

closed 2023-05-01 by github-actions[bot]