Igalia/meta-webkit #185: Issue Raspberry 4-64 + Mesa VC4 driver + Gstreamer = red Label on video

embeddedio, 2020-05-20:
I Build an image with Mesa VC4 driver, weston and wpebackend-fdo for Raspberry pi 4-64 (test Zeus and Master Branche) i have a issue with playing video  i have a red layer on video with Gstreamer or With Webkit ( exp : cog-P fdo simpson.mp4 ).

when i play the video with Gstreamer the red label on video exist
==>exp : gst-play-1.0 --videosink glimagesink simpson.mp4 

[image]

But when i play the video with Gstreamer and change "glimagesink" with "waylandsink" it work perfectly. 
==>exp :gst-play-1.0 --videosink waylandsink simpson.mp4

[image]

##local.conf
MACHINE = "raspberrypi4-64"
MACHINE_FEATURES_append = " vc4graphics"
GPU_MEM = "256"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
IMAGE_FEATURES_append = " ssh-server-dropbear hwcodecs"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
IMAGE_INSTALL_append = " wpewebkit cog"

[image]

# Cog Version 0.7.1 (WPE WebKit 2.28.2)
# Weston Version 8.0.0

#gst-inspect-1.0 --version : 
   gst-inspect-1.0 version 1.16.2
   GStreamer 1.16.2
   Unknown package origin

#DEBUG GSTREAMER 
GST_DEBUG="3,webkit*:6" GST_DEBUG_FILE=/tmp/gst.log GST_DEBUG_DUMP_DOT_DIR=/tmp cog -P fdo simpson.mp4
 
[gst_log.zip](https://github.com/Igalia/meta-webkit/files/4658164/gst_log.zip)

philn, 2020-05-20:
Looks like this is a GStreamer bug, please report it there: https://gitlab.freedesktop.org/gstreamer/gst-plugins-base

mickdev, 2020-06-18:
@embeddedio hey ! there is any news with that ? new mesa version solve the issue ? thx

embeddedio, 2020-09-22:
@mickdev no, the problem still exists.

embeddedio, 2020-09-22:
@philn i have already repose the issue on gstramer, they told me that the problem is rather in mesa

micdini, 2021-04-07:
The problem still exists, with both GStreamer 1.16.x and 1.18.x. Upgrading the version does not affect the result.

32-bit build on a Raspberry Pi CM4.

```
root@raspberrypi4:~# cog --version
0.8.0 (WPE WebKit 2.32.0)
root@raspberrypi4:~# weston --version
weston 9.0.0
root@raspberrypi4:~# uname -a
Linux raspberrypi4 5.10.25-v7l #1 SMP Wed Mar 31 19:51:36 UTC 2021 armv7l GNU/Linux

GL_VERSION: OpenGL ES 3.1 Mesa 21.0.1
GL_SHADING_LANGUAGE_VERSION: OpenGL ES GLSL ES 3.10
GL_VENDOR: Broadcom
GL_RENDERER: V3D 4.2
```

From debug logs the video seems to in I420 format.
```
0:00:10.524333790  2900  0x1f29d88 DEBUG             gleglimage gsteglimage.c:628:_drm_direct_fourcc_from_info: Getting DRM fourcc for I420
```

Yocto build log
```
Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "poky"
DISTRO_VERSION       = "3.2+snapshot-e8e10f27b20f331622b7fdb3eb4e60f0850ea876"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "master:e8e10f27b20f331622b7fdb3eb4e60f0850ea876"
meta-raspberrypi     = "master:f9a13f3b9a90f4f033afa9f34b66ca7432a7742b"
meta-networking
meta-oe
meta-python          = "master:454644c578c84062f49b443626b7bb7dfff06060"
meta-webkit          = "main:2ec4f41d8f774bd716aebcdcd2057af3946f6323"
```

philn, 2021-04-07:
> @philn i have already repose the issue on gstramer, they told me that the problem is rather in mesa

Was a mesa bug reported then?

bertogg, 2021-04-28:
I'm having what looks like the same problem with an iMX53, Adreno A200 GPU using DRM.

ceyusa, 2021-04-28:
These are the related gstreamer issues I could find:

* https://gitlab.freedesktop.org/gstreamer/gst-plugins-base/-/issues/755
* https://gitlab.freedesktop.org/gstreamer/gst-plugins-base/-/issues/519
* https://gitlab.freedesktop.org/gstreamer/gst-plugins-bad/-/issues/1204

I'm using buildroot, but let me share my experience:

I still observe the same problem with `glimagesink` , with gbm as windowing system. But, playing videos inside wpe/cog, with drm backend, the rendering is fine (using gl for video rendering in wpe and v4l2 for decoding).

I guess there's still a bug in Mesa for certain code path, but it's fine if windowless (shared gl context).

These are the versions I'm using:

* Mesa: 21.0.3 
* GStreamer: 1.18.4
* WPEWebkit: 2.32.0
* Cog: 0.9.90
* kernel : 5.10.y and latest firmware (with vc4-kms-v3d-pi4 overlay)

txenoo, 2021-04-29:
Merge request https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/10524 with the fix has already been merged at mesa master. 

Commits had been nominated for stable 21.1 and the fix has already been backported to 21.0 at https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/10534

bertogg, 2021-05-05:
> I'm having what looks like the same problem with an iMX53, Adreno A200 GPU using DRM.

Ok, I confirm that this also fixes the issue in my case:

```
--- a/src/gallium/drivers/freedreno/freedreno_screen.c
+++ b/src/gallium/drivers/freedreno/freedreno_screen.c
@@ -840,7 +840,7 @@ fd_screen_query_dmabuf_modifiers(struct pipe_screen *pscreen,
          modifiers[num] = screen->supported_modifiers[i];
 
       if (external_only)
-         external_only[num] = 0;
+         external_only[num] = util_format_is_yuv(format);
 
       num++;
    }
```

psaavedra, 2021-09-23:
> Ok, I confirm that this also fixes the issue in my case:
>
> ```
> --- a/src/gallium/drivers/freedreno/freedreno_screen.c
> +++ b/src/gallium/drivers/freedreno/freedreno_screen.c
> @@ -840,7 +840,7 @@ [...]

Any reason why this park has not been reported in mesa?

psaavedra, 2021-09-23:
> Any reason why this patch (freedreno) has not been reported in mesa?

Reported in https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/12991

psaavedra, 2021-09-23:
I've created this PR https://github.com/Igalia/meta-webkit/pull/312 to add those fixes for mesa 21.0.3 in hardknott. Also I've submitted those changes to poky as backport for hardknott (https://lists.openembedded.org/g/openembedded-core/message/156259).

psaavedra, 2021-09-27:
> > Any reason why this patch (freedreno) has not been reported in mesa?
> 
> Reported in https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/12991

I got answer from Mesa with a new proposed fix in gallium for this problem:
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/13038

I just tested it  over Mesa 21.0.3 (hardknott - without https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/10524/commits) in RPI4 and apparently this fix the issue for VC4 . Probably for Freedreno too, @bertogg  could you verify this?