Igalia/meta-webkit #189: libwpe cannot find version 1.6.0 to build

Valks, 2020-06-17:


philn, 2020-06-17:
Which branch of meta-webkit are you using?

Valks, 2020-06-17:
I'm pulling in master

philn, 2020-06-18:
Are you building wpewebkit or webkitgtk? Which version?

Valks, 2020-06-18:
Here is the configuration I'm using:

```
MACHINE = "raspberrypi4"

DISTRO_FEATURES_remove = "x11"
DISTRO_FEATURES = "wayland opengl"
MACHINE_FEATURES = "screen touchscreen vc4graphics"

IMAGE_INSTALL = " \
weston \
weston-init \
webkitgtk-bin \
wpebackend-fdo \
"
PACKAGECONFIG_pn-cog = "webkitgtk"

PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_PROVIDER_virtual/libwpe = "wpebackend"

DISPMANX_OFFLINE = "1"
```
I'm pretty sure that's all the relevant config (taken from different files).

sources:
```
git.yoctoproject.org/git/poky                            -b dunfell
git.openembedded.org/meta-openembedded -b dunfell
git.yoctoproject.org/git/meta-raspberrypi        -b dunfell
github.com/OSSystems/meta-security              -b dunfell
github.com/Igalia/meta-webkit                         -b master
```

Anything else I pull and use isn't related. Hopefully that's enough to help you reproduce the same issue.

Valks, 2020-06-19:
Thanks for that, it's compiling again. I could have sworn that I successfully compiled with that setting in place but apparently not. I added it to try and get cog to recognize the backend because it fails saying there's no default defined. When I try and add a symbolic link to the -fdo lib I get errors but if I used --platform=fdo it worked.

While using --platform=fdo won't affect my final requirements it'd be nice to have it correctly configured during the build.

Thanks again for your help.

closed 2020-06-19 by Valks