Igalia/meta-webkit #191: Compilation error with Beaglebone

MagnaboscoL, 2020-06-27:
Hi,

I would like to try WPE in a QT application for a Beaglebone with and/or without GPU support enabled.
The first step would be to give it a try to qt-wpe-simple-browser, but I am facing a compilation error while compiling wpewebkit with backend "wpebackend-fdo".

As far as I understand the issue is due to an error in a template instantiation:
```
   /home/qem/Yocto/bbg/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/wpewebkit/2.28.1-r0/wpewebkit-2.28.1/Source/JavaScriptCore/jit/JITPropertyAccess.cpp:1571:69: error: explicit instantiation of 'void JSC::JIT::emit_op_put_by_val(const JSC::Instruction*) [with Op = JSC::OpPutByVal]' but no definition available [-fpermissive]
 1571 | template void JIT::emit_op_put_by_val<OpPutByVal>(const Instruction*);
      |                                                                     ^
```

My configuration is:

- branch zeus
- in my local.conf:
  ```
  IMAGE_INSTALL_append = " qt-wpe-simple-browser wpewebkit cog"
  # Set the backed that supports Qt integration (see qtwpe option)
  PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
  PACKAGECONFIG_pn-wpewebkit_append = " qtwpe 2dcanvas"
  # Since for the moment I am not interested in video stuff
  PACKAGECONFIG_pn-wpewebkit_remove = " gst_gl mediasource video webaudio"
  ```

From issue https://github.com/Igalia/meta-webkit/issues/120 I understand that it was possible to compile WPE for the Beaglebone. That GPU support through a Wayland compositor was possible too and that direct GPU support without a Wayland compositor was the issue at that time. So I guess that my configuration is missing something.

Regards
Luca

philn, 2020-06-28:
Can you try with master branch? Zeus branch doesn't have 2.28.2.

MagnaboscoL, 2020-06-28:
Hi @philn,

thanks fro your prompt reply.

> Can you try with master branch? Zeus branch doesn't have 2.28.2.

Sure, I have just tried. Unfortunately it seems to me that the result is almost the same.

```
...
/home/qem/Yocto/bbg/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/wpewebkit/2.28.2-r0/wpewebkit-2.28.2/Source/JavaScriptCore/bytecompiler/BytecodeGeneratorBase.h:50:5: error: invalid use of incomplete type 'struct JSC::JSGeneratorTraits'
...
/home/qem/Yocto/bbg/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/wpewebkit/2.28.2-r0/wpewebkit-2.28.2/Source/JavaScriptCore/jit/JITPropertyAccess.cpp:1571:69: error: explicit instantiation of 'void JSC::JIT::emit_op_put_by_val(const JSC::Instruction*) [with Op = JSC::OpPutByVal]' but no definition available [-fpermissive]
...
```

The complete log is [compilation_error_log_master.zip](https://github.com/Igalia/meta-webkit/files/4842611/compilation_error_log_master.zip)

Any suggestion?

Regards
Luca

philn, 2020-06-28:
File a bug at bugs.webkit.org.

MagnaboscoL, 2020-06-29:
Bug filed,

[Bug 213713](https://bugs.webkit.org/show_bug.cgi?id=213713)

zdobersek, 2020-06-30:
2.28.{1,2} building fine here for Beaglebone. Believing this is something to do with how Yocto is configured on your side, I'd like to investigate in that direction first, here.

Here's my Yocto build configuration:

```
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "beaglebone"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6.4"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
```

Can you paste the complete section from your configuration? Also, please attach the wpewebkit CMakeCache.txt file so that it can be reviewed.

zdobersek, 2020-06-30:
A different build configuration (using GCC 9.3.0) that also compiles just fine:

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "beaglebone"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1.1"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
```

MagnaboscoL, 2020-06-30:
Hi @zdobersek,

thanks for looking at the issue.

> Can you paste the complete section from your configuration

```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "beaglebone"
DISTRO               = "qmove"
DISTRO_VERSION       = "1.0.0"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "zeus:40daaa0c0504f9dc4bf121ab18c541185954b706"
meta-oe              
meta-networking      
meta-perl            
meta-python          = "zeus:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
meta-python2         = "zeus:4400f9155ec193d028208cf0c66aeed2ba2b00ab"
meta-qt5             = "zeus:852e279a45da2b68719fecdaad3573b2ada9558c"
meta-security        = "zeus:ecd8c30af6b1b957b4fa814d3f25c40f060b3ac1"
meta-ti              = "zeus-ti2020.00:26265d5957eac9fd52f879b79677a044e953e2dc"
meta-webkit          = "master:f940cc145e5c1c25b39c42afa8e02ef55ef56e50"
meta-qem-bbg         = "zeus-ti-bb:c5a7e0502e4cc163b60093fff67994f24a4c18e5"
```

Note that "qmove" distro is very similar to poky, let me know if more info is needed.

>  Also, please attach the wpewebkit CMakeCache.txt file so that it can be reviewed.

Here you have it [CMakeCache_2.28.2.txt](https://github.com/Igalia/meta-webkit/files/4853897/CMakeCache_2.28.2.txt)

closed 2022-03-26 by github-actions[bot]