Igalia/meta-webkit #193: Using RDK/imx6 backend on an iMX8 with mouse/touchscreen support

gvalcaza, 2020-07-08:
Hi all, I'm currently testing the meta-webkit layer on the Yocto 3.0 (zeus) branch by compiling framebuffer images for a i.MX8-based board. I'm using the cog launcher and the wpebackend-rdk/imx6 backend, and I can browse pages perfectly fine out of the box without tweaking anything.

However, I'm unable to use the mouse or touchscreen to interact with the webpages. I have seen that the RDK/rpi backend code has additional code to support the touch and/or cursor, but this code isn't present in the imx6 flavor of the backend.

Is touch/mouse support available in any way in the imx6 flavor of the RDK backend? Or is it something that needs to be added by hand?

philn, 2020-07-09:
We don't maintain the RDK backend here, can you file an issue in this tracker? https://github.com/WebPlatformForEmbedded/WPEBackend-rdk

gvalcaza, 2020-07-09:
Will do, I will close this issue then

closed 2020-07-09 by gvalcaza