Igalia/meta-webkit #194: WPE Webkit usb camera opening problem and development-mode is not working

DomenicoAcri, 2020-07-23:
Hi,
I'm using an NXP iMX8M Mini processor, with version of Yocto Sumo and kernel 4.14.98. We are having problems opening a video stream from a USB camera. It seems that the WpeWebkit does not support this function. Also I can't enable development-mode.
This is my local.conf for configuration wpewebkit and gstreamer packages:
IMAGE_INSTALL_append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PACKAGECONFIG_append_pn-cairo = " glesv2"
PACKAGECONFIG_append_pn-wpewebkit = " deviceorientation encryptedmedia fetchapi gamepad gst_gl indexeddb mediasource video webaudio webcrypto woff2 2dcanvas remote-inspector webrtc"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-ugly = " mpeg2dec x264"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-good = " cairo jpeg libpng libv4l2 soup v4l2"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-bad = " webrtc egl gles2 opengl openjpeg"
PREFERRED_VERSION_cog = "0.2.0"
PREFERRED_VERSION_wpewebkit = "2.22.5"
PREFERRED_VERSION_wpebackend = "0.2.0"

What am I missing?

Best Regards
Domenico Acri

philn, 2020-07-24:
I'm not sure how well this was supported in 2.22. Any chance you could upgrade to a more recent version?
Also what's the output of `gst-device-monitor-1.0 Video/Source` ?

DomenicoAcri, 2020-08-10:
Hi Philn,
Thanks a lot for the answer. I apologize for the delay but I was on vacation.
Yes, I also used version 2.28.1 with the same result.
This is the output of the requested command:

gst-device-monitor-1.0 Video/Source            
Probing devices...

Device found:

        name  : HD Pro Webcam C920
        class : Video/Source
        caps  : video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1536, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
                video/x-raw, format=(string)YUY2, width=(int)2304, height=(int)1296, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)2/1;
                video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
                video/x-raw, format=(string)YUY2, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)432, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                video/x-raw, format=(string)YUY2, width=(int)160, height=(int)90, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)1600, height=(int)896, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)960, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)1024, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)864, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)800, height=(int)448, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
                image/jpeg, width=(int)432, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 24/1, 20/1, 15/1, 10/1, 15/2, 5/1 };
[... cut ...]
                device.capabilities = :capture:
                device.api = v4l2
                device.path = /dev/video0
                v4l2.device.driver = uvcvideo
                v4l2.device.card = "HD\ Pro\ Webcam\ C920"
                v4l2.device.bus_info = usb-ci_hdrc.1-1.1
                v4l2.device.version = 265826 (0x00040e62)
                v4l2.device.capabilities = 2216689665 (0x84200001)
                v4l2.device.device_caps = 69206017 (0x04200001)
        gst-launch-1.0 v4l2src ! ...

Best Regards
Domenico Acri

philn, 2020-08-10:
Please share the html test case for this. Also, did you enable the media-stream websetting?

DomenicoAcri, 2020-08-10:
Hi,
this is the link  https://it.webcamtests.com/ to test the usb camera.
Once the web page is opened, the browser returns the message: this browser does not support functionality for accessing multimedia devices.
By media-stream do you mean the mediasource flag?
Below are the flags enabled in the configuration phase:

-- Enabled features:

--  ENABLE_ACCELERATED_2D_CANVAS .......... ON

--  ENABLE_ENCRYPTED_MEDIA                  ON

--  ENABLE_MEDIA_SOURCE ................... ON

--  ENABLE_VIDEO                            ON

--  ENABLE_WEBDRIVER ...................... ON

--  ENABLE_WEB_AUDIO                        ON

--  ENABLE_WEB_CRYPTO ..................... ON

--  ENABLE_XSLT                             ON

--  USE_WOFF2 ............................. ON

Best Regards
Domenico Acri

philn, 2020-08-10:
mediasource != mediastream. Check the `cog --help-websettings` output.

DomenicoAcri, 2020-08-10:
Yes, now I understand, I had already done this test.

cog --help-websettings
Usage:
  cog [OPTION?] [URL]

WebKitSettings options can be used to configure features exposed to the loaded Web content.

  BOOL values are either 'true', '1', 'false', or '0'. Ommitting the value implies '1'.
  INTEGER values can be decimal, octal (prefix '0'), or hexadecimal (prefix '0x').
  UNSIGNED values behave like INTEGER, but negative values are not accepted.
  FLOAT values may optionally use decimal separators and scientific notation.
  STRING values may need quoting when passed from the shell.

  --enable-javascript=BOOL                             Enable JavaScript.
  --auto-load-images=BOOL                              Load images automatically.
  --load-icons-ignoring-image-load-setting=BOOL        Whether to load site icons ignoring image load setting.
  --enable-offline-web-application-cache=BOOL          Whether to enable offline web application cache.
  --enable-html5-local-storage=BOOL                    Whether to enable HTML5 Local Storage support.
  --enable-html5-database=BOOL                         Whether to enable HTML5 database support.
  --enable-xss-auditor=BOOL                            Whether to enable the XSS auditor.
  --enable-frame-flattening=BOOL                       Whether to enable frame flattening.
  --enable-plugins=BOOL                                Enable embedded plugin objects.
  --enable-java=BOOL                                   Whether Java support should be enabled.
  --javascript-can-open-windows-automatically=BOOL     Whether JavaScript can open windows automatically.
  --enable-hyperlink-auditing=BOOL                     Whether <a ping> should be able to send pings.
  --default-font-family=STRING                         The font family to use as the default for content that does not specify a font.
  --monospace-font-family=STRING                       The font family used as the default for content using monospace font.
  --serif-font-family=STRING                           The font family used as the default for content using serif font.
  --sans-serif-font-family=STRING                      The font family used as the default for content using sans-serif font.
  --cursive-font-family=STRING                         The font family used as the default for content using cursive font.
  --fantasy-font-family=STRING                         The font family used as the default for content using fantasy font.
  --pictograph-font-family=STRING                      The font family used as the default for content using pictograph font.
  --default-font-size=UNSIGNED                         The default font size used to display text.
  --default-monospace-font-size=UNSIGNED               The default font size used to display monospace text.
  --minimum-font-size=UNSIGNED                         The minimum font size used to display text.
  --default-charset=STRING                             The default text charset used when interpreting content with unspecified charset.
  --enable-developer-extras=BOOL                       Whether to enable developer extras
  --enable-resizable-text-areas=BOOL                   Whether to enable resizable text areas
  --enable-tabs-to-links=BOOL                          Whether to enable tabs to links
  --enable-dns-prefetching=BOOL                        Whether to enable DNS prefetching
  --enable-caret-browsing=BOOL                         Whether to enable accessibility enhanced keyboard navigation
  --enable-fullscreen=BOOL                             Whether to enable the Javascript Fullscreen API
  --print-backgrounds=BOOL                             Whether background images should be drawn during printing
  --enable-webaudio=BOOL                               Whether WebAudio content should be handled
  --enable-webgl=BOOL                                  Whether WebGL content should be rendered
  --allow-modal-dialogs=BOOL                           Whether it is possible to create modal dialogs
  --zoom-text-only=BOOL                                Whether zoom level of web view changes only the text size
  --javascript-can-access-clipboard=BOOL               Whether JavaScript can access Clipboard
  --media-playback-requires-user-gesture=BOOL          Whether media playback requires user gesture
  --media-playback-allows-inline=BOOL                  Whether media playback allows inline
  --draw-compositing-indicators=BOOL                   Whether to draw compositing borders and repaint counters
  --enable-site-specific-quirks=BOOL                   Enables the site-specific compatibility workarounds
  --enable-page-cache=BOOL                             Whether the page cache should be used
  --user-agent=STRING                                  The user agent string
  --enable-smooth-scrolling=BOOL                       Whether to enable smooth scrolling
  --enable-accelerated-2d-canvas=BOOL                  Whether to enable accelerated 2D canvas
  --enable-write-console-messages-to-stdout=BOOL       Whether to write console messages on stdout
  --enable-media-stream=BOOL                           Whether MediaStream content should be handled
  --enable-spatial-navigation=BOOL                     Whether to enable Spatial Navigation support.
  --enable-mediasource=BOOL                            Whether MediaSource should be enabled.
[... cut ...]
  --enable-media-capabilities=BOOL                     Whether MediaCapabilities should be enabled.
  --allow-file-access-from-file-urls=BOOL              Whether file access is allowed from file URLs.
  --allow-universal-access-from-file-urls=BOOL         Whether or not universal access is allowed from the context of file scheme URLs

If I don't put any value on websettings they should already be enabled. 
But when in doubt, I enabled enable-media-stream with the same result unfortunately:
cog --enable-media-stream=1 -P fdo  https://it.webcamtests.com/

Best Regards
Domenico Acri

philn, 2020-08-10:
add `--set-permission=all` as well, this might unlock the webcam permission request.

philn, 2020-08-10:
Ah I see what's going on... `-DENABLE_EXPERIMENTAL_FEATURES=ON` should also be passed to cmake at build time... can you try to add this into the webrtc packageconfig?

DomenicoAcri, 2020-08-12:
Hi,
even with "--set_permissions = all" the webcam does not work. 
I don't understand your last tip, where should I enable "-DENABLE_EXPERIMENTAL_FEATURES = ON"?

Best Regards
Domenico Acri

philn, 2020-08-12:
https://github.com/Igalia/meta-webkit/blob/master/recipes-browser/wpewebkit/wpewebkit.inc#L45 make it look like:
`PACKAGECONFIG[webrtc] = "-DENABLE_WEB_RTC=ON -DENABLE_EXPERIMENTAL_FEATURES=ON,-DENABLE_WEB_RTC=OFF -DENABLE_EXPERIMENTAL_FEATURES=OFF,libvpx libevent libopus"`

DomenicoAcri, 2020-08-17:
Hi  Philippe,
I added "" -ENABLE EXPERIMENTAL FEATURES = ON " to webrtc packageconfig:
PACKAGECONFIG[webrtc] = "-DENABLE_WEB_RTC=ON -DENABLE_EXPERIMENTAL_FEATURES=ON,-DENABLE_WEB_RTC=OFF -DENABLE_EXPERIMENTAL_FEATURES=OFF,libvpx libevent libopus"

compiling the wpewebkit stops with this error:
| CMake Error at Source/CMakeLists.txt:29 (add_subdirectory):
|   add_subdirectory given source "ThirdParty/libwebrtc" which is not an
|   existing directory.

The libwebrtc directory seems to be missing.

Do you have any other ideas?

Best Regards
Domenico Acri

philn, 2020-08-17:
libwebrtc is not shipped in tarballs. You'd need to switch to the git flavor of the wpewebkit recipe.

DomenicoAcri, 2020-08-17:
Hi Philippe,
I do not understand what you mean.
Are you telling me that with this version of wpewebkit https://github.com/Igalia/meta-webkit/tree/zeus, I can't open the usb camera?

Best Regards
Domenico Acri

philn, 2020-08-17:
In local.conf.... `PREFERRED_VERSION_wpewebkit = "2.28+git"`

DomenicoAcri, 2020-08-20:
Hi Philippe,
unfortunately it doesn't work. 
I also included wpewebkit 2.28.4 on the master branch with cog version 0.7.1 but still no results.
There seems to be no way to open a webcam with the wpewebkit.

Best Regards
Domenico Acri

DomenicoAcri, 2020-08-20:
I don't know, I was hoping for help but still no one has given me a solution, which I don't even know if it exists at this point.

philn, 2020-08-20:
I gave the solutions in the previous comments, and I verified this on desktop WPE builds. If you don't report the remaining issues after applying the suggested changes I won't be able to help further.

DomenicoAcri, 2020-08-20:
Hi,
for my system the wpewebkit does not access the webcam. 
The test should be done on an embedded system not a desktop system to see if the changes you suggest really work.

Best Regards
Domenico Acri

clopez, 2020-08-20:
If test 3 fails then try this test: https://people.igalia.com/clopez/wkbug/webrtc/enumerate.html it should find at least one device.

clopez, 2020-08-21:
And there is yet an additional issue to make the test 3 work. The request from media-stream to use the Webcam/mic generates a permission request signal that has to be handled and approved. Cog currrently doesn't implement code to handle that. You can apply this test patch to cog that will simply always accept the request.

```
diff --git a/cog.c b/cog.c
index 49fc576..32f5dc5 100644
--- a/cog.c
+++ b/cog.c
@@ -353,6 +353,17 @@ on_web_view_create (WebKitWebView          *web_view,
     return NULL;
 }
 
+
+static gboolean
+decide_permission_request(WebKitWebView *web_view, WebKitPermissionRequest *request, gpointer *user_data)
+{
+    g_print("Accepting %s request\n", G_OBJECT_TYPE_NAME(request));
+    webkit_permission_request_allow(request);
+
+    return TRUE;
+}
+
+
 static WebKitWebView*
 on_create_view (CogShell *shell, void *user_data G_GNUC_UNUSED)
 {
@@ -405,6 +416,7 @@ on_create_view (CogShell *shell, void *user_data G_GNUC_UNUSED)
 #endif
                                                       NULL);
 
+    g_signal_connect (web_view, "permission-request", G_CALLBACK(decide_permission_request), NULL);
     g_signal_connect (web_view, "create", G_CALLBACK (on_web_view_create), NULL);
 
 #if !COG_USE_WEBKITGTK

```

I can confirm now that your test https://it.webcamtests.com/ works in cog for me.

clopez, 2020-08-21:
mmmm, next time I should re-read the whole bug with more care (twice) before commenting.
It seems that @philn previous suggested solution of passing to cog the parameters ``` --set-permissions=all --enable-media-stream=true``` already achieved everything i posted above (without needing for config file, and no need for extra patch).
:shrug: 

At least I hope the simple test-cases I posted help to narrow the problem, there are 3 simple tests that should pass:
1. media-stream has been enabled correctly: https://people.igalia.com/clopez/wkbug/webrtc/mediastream-detect.html
2. wpe can detect the webcam: https://people.igalia.com/clopez/wkbug/webrtc/enumerate.html
3. the webcam is actually working: https://people.igalia.com/clopez/wkbug/webrtc/webcam-microphone.html or https://people.igalia.com/clopez/wkbug/webrtc/webcam.html

DomenicoAcri, 2020-08-24:
Hi Carlos,
still unable to compile the wpewebkit package.
It seems the same mistake as before.

tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/WebKit2Prefix.h -Wno-unused-parameter -DLINUX -DEGL_API_FB -DWL_EGL_PLATFORM -std=c++17 -MD -MT Source/WebKit/CMakeFiles/WebKit.dir/UIProcess/API/glib/WebKitWebsiteData.cpp.o -MF Source/WebKit/CMakeFiles/WebKit.dir/UIProcess/API/glib/WebKitWebsiteData.cpp.o.d -o Source/WebKit/CMakeFiles/WebKit.dir/UIProcess/API/glib/WebKitWebsiteData.cpp.o -c /home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp
/home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp: In function 'bool recordContainsSupportedDataTypes(const WebKit::WebsiteDataRecord&)':
/home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp:80:26: error: 'ServiceWorkerRegistrations' is not a member of 'WebKit::WebsiteDataType'
         WebsiteDataType::ServiceWorkerRegistrations,
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp:82:6: error: no matching function for call to 'WTF::OptionSet<WebKit::WebsiteDataType>::containsAny(<brace-enclosed initializer list>) const'
     });
      ^
In file included from /home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/WebsiteData/WebsiteDataRecord.h:34,
                 from /home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteDataPrivate.h:23,
                 from /home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp:24:
DerivedSources/ForwardingHeaders/wtf/OptionSet.h:176:20: note: candidate: 'constexpr bool WTF::OptionSet< <template-parameter-1-1> >::containsAny(WTF::OptionSet< <template-parameter-1-1> >) const [with E = WebKit::WebsiteDataType]'
     constexpr bool containsAny(OptionSet optionSet) const
                    ^~~~~~~~~~~
DerivedSources/ForwardingHeaders/wtf/OptionSet.h:176:20: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'WTF::OptionSet<WebKit::WebsiteDataType>'
/home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp: In function 'WebKitWebsiteDataTypes toWebKitWebsiteDataTypes(WTF::OptionSet<WebKit::WebsiteDataType>)':
/home/domenico/yocto/warrior/build_imx8mmevk/tmp/work/aarch64-mx8mm-poky-linux/wpewebkit/2.28+gitAUTOINC+7f0791ac47-r0/git/Source/WebKit/UIProcess/API/glib/WebKitWebsiteData.cpp:110:41: error: 'ServiceWorkerRegistrations' is not a member of 'WebKit::WebsiteDataType'
     if (types.contains(WebsiteDataType::ServiceWorkerRegistrations))

clopez, 2020-08-24:
which packageconfig options are you pasing to the wpewebkit build?

DomenicoAcri, 2020-08-25:
Hi Carlos,
yes, now with this packageconfig the compilation is successful:

#PACKAGECONFIG_append_pn-wpewebkit = " service-worker webrtc 2dcanvas encryptedmedia webgl2"

But, I can't enable the MEDIA_STREAM.
Following Philippe's advice, I added the:

# PACKAGECONFIG[webrtc] = "-DENABLE_WEB_RTC=ON -DENABLE_EXPERIMENTAL_FEATURES=ON,-DENABLE_WEB_RTC=OFF -DENABLE_EXPERIMENTAL_FEATURES=OFF,libvpx libevent libopus"

compilation stops with the error attached.

What is missing?

Best Regards
Domenico Acri

clopez, 2020-08-25:
In that log I can see the configure step is failing because it can't find gstreamer-codecparsers-1.0 which is provided by the package gstreamer1.0-plugins-bad. Can you try rebuilding after applying this patch?
```
diff --git a/recipes-browser/wpewebkit/wpewebkit.inc b/recipes-browser/wpewebkit/wpewebkit.inc
index 496b9f5..a41ab7f 100644
--- a/recipes-browser/wpewebkit/wpewebkit.inc
+++ b/recipes-browser/wpewebkit/wpewebkit.inc
@@ -34,7 +34,7 @@ PACKAGECONFIG[gst_gl] = "-DUSE_GSTREAMER_GL=ON,-DUSE_GSTREAMER_GL=OFF,gstreamer1
 PACKAGECONFIG[indexeddb] = "-DENABLE_INDEXED_DATABASE=ON,-DENABLE_INDEXED_DATABASE=OFF,"
 PACKAGECONFIG[mediasource] = "-DENABLE_MEDIA_SOURCE=ON,-DENABLE_MEDIA_SOURCE=OFF,gstreamer1.0 gstreamer1.0-plugins-good"
 PACKAGECONFIG[service-worker] = "-DENABLE_SERVICE_WORKER=ON,-DENABLE_SERVICE_WORKER=OFF,"
-PACKAGECONFIG[video] = "-DENABLE_VIDEO=ON -DENABLE_VIDEO_TRACK=ON,-DENABLE_VIDEO=OFF -DENABLE_VIDEO_TRACK=OFF,gstreamer1.0 gstreamer1.0-plugins-base"
+PACKAGECONFIG[video] = "-DENABLE_VIDEO=ON -DENABLE_VIDEO_TRACK=ON,-DENABLE_VIDEO=OFF -DENABLE_VIDEO_TRACK=OFF,gstreamer1.0 gstreamer1.0-plugins-base gstreamer1.0-plugins-bad"
 PACKAGECONFIG[webaudio] = "-DENABLE_WEB_AUDIO=ON,-DENABLE_WEB_AUDIO=OFF,gstreamer1.0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good"
 PACKAGECONFIG[webcrypto] = "-DENABLE_WEB_CRYPTO=ON,-DENABLE_WEB_CRYPTO=OFF,libgcrypt libtasn1"
 PACKAGECONFIG[webgl2] = "-DENABLE_WEBGL2=ON,-DENABLE_WEBGL2=OFF,"
```

DomenicoAcri, 2020-08-26:
Thanks Carlos,
now the configuration seems to be fine, 
but the compilation fails. I am attaching the log file.

[log.do_compile.txt](https://github.com/Igalia/meta-webkit/files/5128954/log.do_compile.txt)

clopez, 2020-08-26:
There are many build errors in this last log you attached, and those errors are from basic types not being detected properly. I don't know what causes that. It shouldn't happen. Something is wrong in your build environment. I suggest to wipe everything (even yocto sstate-cache) and start a clean build from a scratch.

DomenicoAcri, 2020-08-28:
Hi,
unfortunately I always have compilation errors when I insert "-DENABLE_EXPERIMENTAL_FEATURES" in the webrtc packageconfig.

Attached the log file.
[log.do_compile.txt](https://github.com/Igalia/meta-webkit/files/5140794/log.do_compile.txt)

closed 2021-09-05 by github-actions[bot]