Igalia/meta-webkit #217: No touchscreen (evdev) support

fardei, 2021-01-27:
I've built meta-webkit with yocto warrior for a imx6 board.
I am able to display a java based webpage on the touchscreen display (using cog). 
Unfortunately the touch events are not forwarded to the application and the app doesn't react.

Can this touchscreen support be enabled ? Am I missing any configuration parameters ?

psaavedra, 2021-02-04:
From your description I understand your are using the framebuffer to draw the WPE output in the screen. In that mode, WPE doesn't reacts to any kind of input method it is more attempted to work as a kiosk mode.

fardei, 2021-02-04:
Yes, indeed. The app is displayed in framebuffer and the user interacts with it using the touchscreen.
Is there any way to enable the touch support or is there anything else besides QT that supports this ?

psaavedra, 2021-02-05:
Right now the only method  that I know to support input methods is through _wayland_ using the **wpebackend-fdo** with _weston/libinput_. That is the natural choice if you want to deal with input methods.

If wayland is not a choice I think you need to patch [https://github.com/Igalia/cog](cog) (the WPE-based browser) to read the `/dev/input/eventX` device  and inject events directly to the WPE backend using the `wpe_view_backend_dispatch_*_event()` calls:

```
    wpe_input_touch_event_raw touchpoint =
    {
        pressed ? wpe_input_touch_event_type_down : wpe_input_touch_event_type_up,
        timestamp ? timestamp : static_cast<uint32_t>(g_get_monotonic_time() / 1000),
        0,  x,  y,
    };

    wpe_input_touch_event event =
    {
        &touchpoint,  // touchpoints[]
        1,            // touchpoints_length
        touchpoint.type,
        touchpoint.id,
        touchpoint.time,
        0,
    };
    wpe_view_backend_dispatch_touch_event(getViewBackend(), &event);
```
```
    wpe_input_pointer_event event =
    {
        wpe_input_pointer_event_type_motion,  // type
        timestamp ? timestamp : static_cast<uint32_t>(g_get_monotonic_time() / 1000),
        x, y,
        0,  // button
        0,  // state
        0,  // modifiers
    };
    wpe_view_backend_dispatch_pointer_event(getViewBackend(), &event);
```

Link: https://stackoverflow.com/questions/28841139/how-to-get-coordinates-of-touchscreen-rawdata-using-linux

gust1n, 2021-03-23:
Hello! We're actually running this very setup you're mentioning: Apalis iMX6, using wpewebkit over framebuffer and have successfully (and quite easily) enabled touch event support. I'm not a C/C++ developer at all, but successfully found this solution while reading the source (felt like pure dumb luck).

If any of the (real) webkit developers thinks this is actually a feasable solution I can open a PR to the rdk repo. This is my patch:

```
diff --git a/src/viv-imx6/view-backend.cpp b/src/viv-imx6/view-backend.cpp
index 2759c6f..d8152be 100644
--- a/src/viv-imx6/view-backend.cpp
+++ b/src/viv-imx6/view-backend.cpp
@@ -94,6 +94,9 @@ void ViewBackend::initialize()
 
     wpe_view_backend_dispatch_set_size(backend, width, height);
 
+    WPE::LibinputServer::singleton().setHandleTouchEvents(true);
+    WPE::LibinputServer::singleton().setPointerBounds(width, height);
+
     WPE::LibinputServer::singleton().setClient(this);
 }
 ```

kraj, 2021-04-07:
@gust1n can you post more information on how you are building wpe for fbdev setup ? or any other quirks needed in addition to one you mentioned here ?

psaavedra, 2021-04-07:
> Hello! We're actually running this very setup you're mentioning: Apalis iMX6, using wpewebkit over framebuffer (no X11 or wayland at all in the image) and have successfully (and quite easily) enabled touch event support. I'm not a C/C++ developer at all, but successfully found this solution while reading the source (felt like pure dumb luck).
> 
> If any of the (real) webkit developers thinks this is actually a feasable solution I can open a PR to the rdk repo. This is my patch:
> 
> ```
> diff --git a/src/viv-imx6/view-backend.cpp b/src/viv-imx6/view-backend.cpp
> index 2759c6f..d8152be 100644
> --- a/src/viv-imx6/view-backend.cpp
> +++ b/src/viv-imx6/view-backend.cpp
> @@ -94,6 +94,9 @@ void ViewBackend::initialize()
>  
>      wpe_view_backend_dispatch_set_size(backend, width, height);
>  
> +    WPE::LibinputServer::singleton().setHandleTouchEvents(true);
> +    WPE::LibinputServer::singleton().setPointerBounds(width, height);
> +
>      WPE::LibinputServer::singleton().setClient(this);
>  }
> ```

Nice.

Sounds similar to what is implemented for the bcm-rpi in https://github.com/WebPlatformForEmbedded/WPEBackend-rdk/blob/master/src/bcm-rpi/view-backend.cpp#L190 and https://github.com/WebPlatformForEmbedded/WPEBackend-rdk/blob/master/src/bcm-rpi/view-backend.cpp#L195. 

If works I think it could be something useful to contributed directly in the backend (probably safeguarded with envvars like in the rpi-bcm case).

gust1n, 2021-04-07:
It definately works, we've been using it with various Toradex BSP versions over the past year, together with a touchscreen using the EXC3000 driver. And yes, I think I got it from reading the source of the rpi implementation :)

Let me know if you want any further help/work from me

psaavedra, 2021-04-07:
> It definately works, we've been using it with various Toradex BSP versions over the past year, together with a touchscreen using the EXC3000 driver. And yes, I think I got it from reading the source of the rpi implementation :)
> 
> Let me know if you want any further help/work from me

The only thing is if you want contribute this directly, as author, in https://github.com/WebPlatformForEmbedded/WPEBackend-rdk/ instead of here as a patch. I believe this change fits better there and it can be welcome there as well.

closed 2022-04-15 by github-actions[bot]