Igalia/meta-webkit #219: Default branch is main (instead of master)

psaavedra, 2021-02-09:
_Widely accepted terms “master” and “slave” as an oppressive metaphor for ownership/importance._ 

Lately there is an increasing movement in the software community that supports this idea. Moreover, we can verify that this change is done in many relevant projects in the different areas around the world: 

https://github.com/github/renaming
https://sfconservancy.org/news/2020/jun/23/gitbranchname/
https://github.com/WebKit/WebKit/tree/main

psaavedra, 2021-02-09:
Done in https://github.com/Igalia/meta-webkit/commit/573bd1da254d00276cc423cc286ba1536485022f 

The old master keeps referenciated in the branch `old-master` (https://github.com/Igalia/meta-webkit/tree/old-master)

otavio, 2021-02-09:
@psaavedra, you broke all platforms which refer to the master branch :-( This kind of change done without ANY notice is ridiculous.

psaavedra, 2021-02-10:
> @psaavedra, you broke all platforms which refer to the master branch :-( This kind of change done without ANY notice is ridiculous.

Sorry to read about this inconvenience. I've restored the master branch pointing to the same hash. Also,  I'm open to discuss this topic. How do you think it should be the right mechanism to announce this change?

otavio, 2021-02-10:
Well, any change like this should have a transition plan. Honestly, I don't care about the name of the branch but:
1. a sync period where old and new branch are kept in sync (notified in README)
2. leave master there for some time with a huge warning in README
3. drop master.

psaavedra, 2021-02-10:
what time do you think it is the appropriated for that?

otavio, 2021-02-10:
I think 2 or 3 months are good enough. My concern was forcing the change without a graceful warning and migration period.

psaavedra, 2021-02-10:
> I think 2 or 3 months are good enough. My concern was forcing the change without a graceful warning and migration period.

OK. Thanks for your clarification. I understand your complains. I've added an unreviewed change to keep synced the  `master` branch with `main` (05a0629) with a Github Action. That should be enough to activate this transition period. 

In addition, I've created https://github.com/Igalia/meta-webkit/issues/220 what it takes cover about the leftovers at the end of the period time.

Last but not least, I've accepted your suggestion to add a warning note in the README file notifying this change: https://github.com/Igalia/meta-webkit/pull/221.

Let me know if I miss anything. Thanks for devote time reporting in this issue.

otavio, 2021-02-10:
It looks good. Thanks for taking immediate action on it.

woutervanh, 2021-03-15:
This is just ridiculous wokeness... Never in the history of computing has there been any link with human slavery... It's just the way it is. There's a master and multiple slaves. How would you call it otherwise? We are talking about machines! What's next? Don't use black computers anymore? Let the computer think for itself before obeying a command? It's like otavio said, it broke and will break many build systems of people who are not aware... (woke) (Sorry for the rant :-p). BTW, I have a master degree, now what?

psaavedra, 2021-03-15:
> This is just ridiculous wokeness... Never in the history of computing has there been any link with human slavery... It's just the way it is. There's a master and multiple slaves. How would you call [...]

The change is aligned with other changes widely did already by other open-source projects. Moreover, this type of renaming actions has also the commitment of the platform who actually is hosting the repository ([Github renaming policy](https://github.com/github/renaming)).

Thanks to @otavio's report we understood the side-effects and we took the right actions to make this change compatible with other 3rd party projects that are using the meta-webkit layers. That was one month ago. We appreciate this kind of constructive reports.

In order to keep providing help to the community, I'd appreciate if you could reformulate your frustration in a kind of technical issue derived of this change. In any case, the change was already implemented and there are no plans to revert it back.

woutervanh, 2021-03-15:
Hey @psaavedra, this was not meant to address you in person. No need to reopen the issue too. I'll look for a place to ventilate some frustration in a more proper discussion. I get it that it's a more global policy thing.

closed 2021-03-15 by philn