Igalia/meta-webkit #227: Build Issue

navesta, 2021-02-15:
Hi, 
Following http://blog.neutrino.es/2015/meta-webkit-yocto-layer/ to include meta-webkit into an image, for Xilinx Zynq processor, and cmake is complaining that it cannot find KHR/khrplatform.h: `work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot/usr/include/EGL/eglplatform.h:37:10: fatal error: KHR/khrplatform.h: No such file or directory` . The full error log is as follows:

`NOTE: Running task 4098 of 4135 (/home/nastoohav/work/vcmos/meta-webkit/recipes-browser/libwpe/libwpe_1.8.0.bb:do_compile)
NOTE: recipe libwpe-1.8.0-r0: task do_compile: Started
ERROR: libwpe-1.8.0-r0 do_compile: Function failed: do_compile (log file is located at /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/temp/log.do_compile.9360)
ERROR: Logfile of failure stored in: /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/temp/log.do_compile.9360
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/build --target all -- -j 128
| [1/12] /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native/usr/bin/aarch64-vcm-linux/aarch64-vcm-linux-gcc -DWPE_COMPILATION -Dwpe_EXPORTS -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/include -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src -IDerivedSources/wpe --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native=    --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot -Wall -fPIC -fvisibility=hidden   -std=gnu99 -MD -MT CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o -MF CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o.d -o CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o   -c /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src/renderer-backend-egl.c
| FAILED: CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o
| /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native/usr/bin/aarch64-vcm-linux/aarch64-vcm-linux-gcc -DWPE_COMPILATION -Dwpe_EXPORTS -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/include -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src -IDerivedSources/wpe --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native=    --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot -Wall -fPIC -fvisibility=hidden   -std=gnu99 -MD -MT CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o -MF CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o.d -o CMakeFiles/wpe.dir/src/renderer-backend-egl.c.o   -c /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src/renderer-backend-egl.c
| In file included from /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/include/wpe/renderer-backend-egl.h:44,
|                  from /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src/renderer-backend-egl.c:27:
| /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot/usr/include/EGL/eglplatform.h:37:10: fatal error: KHR/khrplatform.h: No such file or directory
|  #include <KHR/khrplatform.h>
|           ^~~~~~~~~~~~~~~~~~~
| compilation terminated.
| [2/12] /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native/usr/bin/aarch64-vcm-linux/aarch64-vcm-linux-gcc -DWPE_COMPILATION -Dwpe_EXPORTS -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/include -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src -IDerivedSources/wpe --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native=    --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot -Wall -fPIC -fvisibility=hidden   -std=gnu99 -MD -MT CMakeFiles/wpe.dir/src/version.c.o -MF CMakeFiles/wpe.dir/src/version.c.o.d -o CMakeFiles/wpe.dir/src/version.c.o   -c /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src/version.c
| [3/12] /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native/usr/bin/aarch64-vcm-linux/aarch64-vcm-linux-g++  -DWPE_COMPILATION -Dwpe_EXPORTS -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/include -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src -IDerivedSources/wpe --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden   --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot -fno-exceptions -fno-rtti -Wall -fPIC -fvisibility=hidden   -std=gnu++11 -MD -MT CMakeFiles/wpe.dir/src/pasteboard-noop.cpp.o -MF CMakeFiles/wpe.dir/src/pasteboard-noop.cpp.o.d -o CMakeFiles/wpe.dir/src/pasteboard-noop.cpp.o -c /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src/pasteboard-noop.cpp
| [4/12] /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot-native/usr/bin/aarch64-vcm-linux/aarch64-vcm-linux-gcc -DWPE_COMPILATION -Dwpe_EXPORTS -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/include -I/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/libwpe-1.8.0/src -IDerivedSources/wpe --sysroot=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0=/usr/src/debug/libwpe/1.8.0-r0                      -fdeb [... cut ...]
| ninja: build stopped: subcommand failed.
| WARNING: /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/temp/run.do_compile.9360:1 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} }VERBOSE=1 cmake --build '/home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/build' "$@" -- -j 128'
| ERROR: Function failed: do_compile (log file is located at /home/nastoohav/saltspring-build/tmp/work/aarch64-vcm-linux/libwpe/1.8.0-r0/temp/log.do_compile.9360)`
Any thoughts?

clopez, 2021-02-16:
> Following http://blog.neutrino.es/2015/meta-webkit-yocto-layer/ to include meta-webkit into an image

That instructions are old. There are newer instructions on the wiki here: https://github.com/Igalia/meta-webkit/wiki/WPE

> cmake is complaining that it cannot find KHR/khrplatform.h

For some reason that header (KHR/khrplatform.h) is missing on your build. You have to check why is missing and fix it.

Perhaps is related with how Mesa is configured in your build? Here is an old bug that suggests that EGL support has to be enabled on Mesa: https://bugs.freedesktop.org/show_bug.cgi?id=77240

navesta, 2021-02-17:
Thank you for your reply. In my case, what solved it was adding mesa-gl to meta-webkit/recipes-browser/libwpe/libwpe.inc:: 
`DEPENDS = "virtual/egl libxkbcommon mesa-gl"`

clopez, 2021-02-19:
Great. Closing.

closed 2021-02-19 by clopez

clopez, 2021-02-19:
To add some context on why I'm closing this.
 * libwpe itself only includes this external headers: `EGL/eglplatform.h xkbcommon/xkbcommon-compose.h xkbcommon/xkbcommon.h`
 * And it already declares dependencies on `virtual/egl` and `libxkbcommon`

The bug is on some recipe of your BSP. When the `virtual/egl` provider of your BSP is built it should be building mesa-gl as well. For other BSPs mesa-gl may not be needed as they may provide this header by themselves without relying on mesa