Igalia/meta-webkit #230: I.MX6 performance limited by running on single core?

jfinnie, 2021-03-04:
Running cog/webkit on Yocto Zeus/Wayland on a BoundaryDevices SabreLite board, it seems that for the interactive user interfaces we have in HTML5/Javascript the performance by default is limited to whatever cycles can be used in a single CPU core - ie on a quad core device CPU idle hovers around 75% as only one thread is in use.  

Is there any way to improve on this so the load is spread across multiple cores?  Or is it inherent limitation in the underlying webkit?

clopez, 2021-03-04:
If the bottleneck is the speed to execute the JS code then I'm afraid its not possible to use multiple cores to speed that up. As you might know, JavaScript is single-threaded, so it expected that the performance of your JS application will be limited to one core. Note this is not a limitation of WebKit, but happens the same on all browsers. Is just how JS is designed

It may affect also the speed at what WebKit can execute JS how yo build it. For example, enabling JIT (which should be the default) is recommended.

But maybe the bottleneck is not in JS, so perhaps something can be optimized there. Check https://github.com/Igalia/meta-webkit/wiki/PerformanceTips just in case that helps

jfinnie, 2021-03-04:
> If the bottleneck is the speed to execute the JS code then I'm afraid its not possible to use multiple cores to speed that up. As you might know, JavaScript is single-threaded, so it expected that [...]

Thank you for your reply, much appreciated.  

This was my initial thought, that it could be Javascript related.  I'll freely admit to not knowing a ton about how it works at the implementation level so will have to dig into some background reading.  I guess aside from the threading which may be a red herring, is it possible that the Javascript engine is less performant in cog / webkit than Chromium?

We're looking to move away from Chromium on Wayland as the drawing performance is pretty awful, compile time hideous and it is just waaaaay too much software for what we want to do, just we note that page startup is generally faster on Chromium vs cog, sometimes very significantly so.  Cog seems at the moment to be a better option in pretty much all other respects (still getting to grips with all the configuration, cache, etc)

One website I sometimes use as a test is this one:
https://www.soundcraft.com/ui-demo/mixer.html

It loads 3x faster on Chromium than it does on cog.  

I assume the JIT setting is as you mention as we compiled using the defaults in the recipe, though I'll ask the guy who is now building the target HW images if the options for this change the behaviour at all.

Thanks for any insight.

clopez, 2021-03-04:
> Thank you for your reply, much appreciated.
>
> This was my initial thought, that it could be Javascript related. I'll freely admit to not knowing a ton about how it works at the implementation level [...]

If you want to compare the speed of the JSC engine you can run some JS benchmark like https://browserbench.org/JetStream/

But there are lot of things that can contribute to a page rendering slow other than the speed of the JS engine.

Using the web inspector to take a performance profile capture can you give you hints were the time is mostly spent and where are the main bottlenecks.

Optimizations can be done on the browser-side (perhaps there is something on WebKit that make it handle some specific of that site slowly that can be improved) or on the application-side. Perhaps you can try to use less JavaScript on the main thread and delegate background jobs to service workers, or perhaps using canvas (off-screen canvas) can be an idea. Just wild guesses here.

> We're looking to move away from Chromium on Wayland as mostly the performance is pretty awful, compile time hideous and it is just waaaaay too much software for what we want to do, just we note that page startup is generally faster on Chromium vs cog, sometimes very significantly so. Cog seems at the moment to be a better option in pretty much all other respects (still getting to grips with all the configuration, cache, etc)
> 
> One website I sometimes use as a test is this one:
> https://www.soundcraft.com/ui-demo/mixer.html
> 
> It loads 3x faster on Chromium than it does on cog.
>

Another thing to consider is the network.
WPE currently doesn't support HTTP/2, so if Chrome is loading the site via HTTP/2 then it maybe can load it much faster than WPE that needs to use traditional HTTP

jfinnie, 2021-03-04:
Thanks for the pointers, very much appreciated.  Will dig into it a bit.

aperezdc, 2021-03-05:
Lots of good advice in this thread :smile_cat:, so here's one more option...

If the issue is due to JS blocking the main thread, another option is moving some of the work to [Web Workers](https://developer.mozilla.org/en-US/docs/Web/API/Web_Workers_API), which are lighter and easier to program than full-fledged Service Workers.

Lastly, I would add that it is always a good idea to try and understand while the bottlenecks are coming from to then decide what could be the best way to optimize your content.

closed 2022-03-18 by github-actions[bot]