Igalia/meta-webkit #235: QML WPEView dynamic loading

gregor26, 2021-03-15:
Hi,

I'm evaluating WPEQt a bit and have an issue with WPEView control. When using WPEView component in root QML file, all works as expected. But in case QML component which includes WPEView is dynamically created, WPEView is not loaded and nothing is shown.

Example 1 - WPEView is used in the root QML file, all is ok.
```
/// main.qml ///
import QtQuick 2.15
import QtQuick.Window 2.15
import QtQuick.Controls 2.15
import org.wpewebkit.qtwpe 1.0

Window {
    width: 640
    height: 480
    title: qsTr("WPEQt Demo")
    visible: true

    Rectangle {
        id: rect1
        anchors.top: button1.bottom
        anchors.right: parent.right
        anchors.left: parent.left
        anchors.bottom: parent.bottom
        color: "grey"

        Rectangle {
            x: 10
            y: 10
            width: 620
            height: 410

            color:"lightblue"

            WPEView {
                anchors.fill: parent
                url: "https://www.google.com/"
            }
        }
    }
}
```

Example 2 - loaded dynamically (WPEView is part of MyRect which is loaded on button press), WPEView is not loaded/shown.
```
/// main.qml ///
import QtQuick 2.15
import QtQuick.Window 2.15
import QtQuick.Controls 2.15

Window {
    width: 640
    height: 480
    title: qsTr("WPEQt Demo")
    visible: true

    property var myrect

    Button {
        id: button1
        text: qsTr("Button 1")
        width: parent.width / 2
        height: 0.1 * parent.height
        anchors.top: parent.top
        anchors.left: parent.left
        onClicked: {
            console.log("button1 clicked")

            var component = Qt.createComponent("MyRect.qml");
            myrect = component.createObject(rect1);
        }
    }

    Button {
        id: button2
        text: qsTr("Button 2")
        width: parent.width / 2
        height: 0.1 * parent.height
        anchors.top: parent.top
        anchors.right: parent.right
        onClicked: {
            console.log("button2 clicked")

            myrect.destroy(10);
        }
    }

    Rectangle {
        id: rect1
        anchors.top: button1.bottom
        anchors.right: parent.right
        anchors.left: parent.left
        anchors.bottom: parent.bottom
        color: "grey"
    }
}

/// MyRect.qml ///
import QtQuick 2.15
import org.wpewebkit.qtwpe 1.0

Item {
    id: myrectitem

    Rectangle {
        x: 10
        y: 10
        width: 620
        height: 410

        color:"lightblue"

        WPEView {
            anchors.fill: parent
            url: "https://www.google.com/"
        }
    }
}
```

Appreciate any answers or ideas on how to continue. Thanks.

Gregor

philn, 2021-03-15:
Can you check if this method returns too early? https://github.com/WebKit/WebKit/blob/main/Source/WebKit/UIProcess/API/wpe/qt/WPEQtView.cpp#L175

gregor26, 2021-03-17:
Thanks for the answer. Don't have a means at the moment to check/verify this.

Did a small further check and tried to update the url, change enable/visible status on some delay after dynamic control gets loaded and results for the wpeview are the same .. nothing visible and no status/loading changes.

closed 2021-03-17 by philn