Igalia/meta-webkit #284: Is something bad with release naming?

GobGeeklord, 2021-06-25:
Hi there, 

I'm not sure if it is something with my environment, but i'm fiding something strange:
When i'm trying to fetch cog, i'm always getting 404 error: 

> ERROR: cog-0.8.1-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH="/opt/verdin/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/opt/verdin/layers/openembedded-core/scripts:/opt/verdin/build/tmp/work/aarch64-tdx-linux/cog/0.8.1-r0/recipe-sysroot-native/usr/bin/aarch64-tdx-linux:/opt/verdin/build/tmp/work/aarch64-tdx-linux/cog/0.8.1-r0/recipe-sysroot/usr/bin/crossscripts:/opt/verdin/build/tmp/work/aarch64-tdx-linux/cog/0.8.1-r0/recipe-sysroot-native/usr/sbin:/opt/verdin/build/tmp/work/aarch64-tdx-linux/cog/0.8.1-r0/recipe-sysroot-native/usr/bin:/opt/verdin/build/tmp/work/aarch64-tdx-linux/cog/0.8.1-r0/recipe-sysroot-native/sbin:/opt/verdin/build/tmp/work/aarch64-tdx-linux/cog/0.8.1-r0/recipe-sysroot-native/bin:/opt/verdin/layers/openembedded-core/bitbake/bin:/opt/verdin/build/tmp/hosttools"; export HOME="/home/calhotas"; /usr/bin/env wget -t 2 -T 30 --passive-ftp --no-check-certificate -P /opt/verdin/build/../downloads 'https://github.com/Igalia/cog/releases/download/0.8.1/cog-0.8.1.tar.xz' --progress=dot -v failed with exit code 8, output:
> --2021-06-25 18:22:05--  https://github.com/Igalia/cog/releases/download/0.8.1/cog-0.8.1.tar.xz
> Resolving github.com (github.com)... 140.82.121.3
> Connecting to github.com (github.com)|140.82.121.3|:443... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2021-06-25 18:22:05 ERROR 404: Not Found.
> 
> 
> ERROR: cog-0.8.1-r0 do_fetch: Fetcher failure for URL: 'https://github.com/Igalia/cog/releases/download/0.8.1/cog-0.8.1.tar.xz'. > Unable to fetch URL from any source.

I've checked and the URL is not ok, because bitbake is searching for **https://github.com/Igalia/cog/releases/download/0.8.1/cog-0.8.1.tar.xz** but the correct tarball is **https://github.com/Igalia/cog/releases/download/v0.8.1/cog-0.8.1.tar.xz** (is missing a v).

Is this some problem in my environment?

Regards

philn, 2021-06-25:
When you report a bug please tell us which branch you use. Is this with dunfell? Can you cherry-pick this? https://github.com/Igalia/meta-webkit/commit/f98f5c85e77d912c65501f3077ba17ce4e47d1f3

GobGeeklord, 2021-06-30:
> When you report a bug please tell us which branch you use. Is this with dunfell? Can you cherry-pick this? [f98f5c8](https://github.com/Igalia/meta-webkit/commit/f98f5c85e77d912c65501f3077ba17ce4e47d1f3)

Hi @philn ,

Yup i'm using dunfell branch, and this worked.

Regards

philn, 2021-07-01:
Keeping this open until the dunfell branch is updated accordingly.

closed 2021-07-14 by philn