Igalia/meta-webkit #293: Something changed regarding usage of playbin between 2.30.4 and 2.32.0?

woutervanh, 2021-07-07:
Hello,
I'm already running wpewebkit for a while on our imx6 platform. Using backend-rdk/vivante. To play rtp streams directly, we've added a rtp:// uri handler in gstreamer. Uptill now, it ran smooth. But now we've updated wpewebkit, it stopped working via the videotag. It still runs via playbin. What could explain this? Http in videotag is still working, only our custom uri isn't.

woutervanh, 2021-07-08:
I generated dot files, and only null->ready and ready->paused are generated. And it seems nothing gets connected to the uridecodebin

philn, 2021-07-08:
Can you provide logs please? `GST_DEBUG="3,webkit*:6 GST_DEBUG_FILE=gst.log`

woutervanh, 2021-07-09:
Full log as requested:
[gst.log](https://github.com/Igalia/meta-webkit/files/6789917/gst.log)

philn, 2021-07-09:
Maybe this is related with the glvideoflip missing. Try to update to 2.32.1 or at least cherry-pick this https://trac.webkit.org/changeset/277088/webkit

woutervanh, 2021-07-09:
Well, I come from 2.32.1, where it wasn't working either.... 
Bear in mind that a plain http:// uri in the videotag does work.
Also if I provide the rtp:// uri to gst-launch-1.0 playbin uri=rtp:// video-sink=glimagesink, it works also.

woutervanh, 2021-07-09:
> Maybe this is related with the glvideoflip missing. Try to update to 2.32.1 or at least cherry-pick this https://trac.webkit.org/changeset/277088/webkit

Is 2.32.1 having this patch? I just compiled it, and still see the:

> 0:00:04.055939667   575 0xace040a0 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:458:gst_element_factory_make: no such element factory "glvideoflip"!

woutervanh, 2021-07-09:
Trace gives continuously (after Time duration query failed):

> 0:05:18.118838037   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:18.469678704   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:18.820962038   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:19.172162704   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:19.523323371   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:19.873946371   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:20.224744371   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:20.575012371   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}
0:05:20.925541371   576 0xace040a0 TRACE      webkitmediaplayer MediaPlayerPrivateGStreamer.cpp:569:durationMediaTime:<media-player-0> Cached duration: {0/1 = Infinity}

woutervanh, 2021-07-09:
Another thing I note now.
If I say it works via gst-launch, I mean it works via playbin. I tried via playbin3 and it's also not working. Has this been changed recently maybe? That webkit took playbin before, and taking playbin3 now?

philn, 2021-07-09:
> To simulate how it works in wpe, should i try gst-launch with playbin or playbin3?

No. By default WPE uses playbin, unless this env var is set: `WEBKIT_GST_USE_PLAYBIN3=1`

woutervanh, 2021-07-09:
> > To simulate how it works in wpe, should i try gst-launch with playbin or playbin3?
> 
> No. By default WPE uses playbin, unless this env var is set: `WEBKIT_GST_USE_PLAYBIN3=1`

Ok, than it should work :-) Any clue on my other findings? I see that recently there were huge reworks of the mediaplayer

philn, 2021-07-09:
I don't have access to any rtp server, so can't really debug this. Sorry :)
Anyway, a page with a live video source works fine here (using trunk though): `<video autoplay src="testbin://video"/>`

woutervanh, 2021-07-10:
Hi @philn , I quickly set up a rtsp server:
rtsp://wonda.be/test2
Although I was unable to test if the issue persist with this stream also (switches to tcp in stead of udp in my usecase) as I don't have the setup at hand.

woutervanh, 2021-07-12:
Just did some further tests
http:// -> works
testbin://video -> works
rtsp:// with link provided does not work, although gst-launch-1.0 playbin uri=rtsp:// does
rtp:// (with own plugin) also does not work

philn, 2021-07-13:
https://bugs.webkit.org/show_bug.cgi?id=227904

closed 2021-07-13 by philn