Igalia/meta-webkit #294: add a getting started for beginners

georgfaust, 2021-07-08:
the wiki assumes some knowledge of yocto that I do not have, but I like to give WPE a try.

here is what I did:

1. follow the yocto quick-build https://docs.yoctoproject.org/brief-yoctoprojectqs/index.html (but did not checkout a release branch, because if I understand correctly WPE wants to work with master as of readme.md)
2. in `poky` clone the repos mentioned in the readme
3. `source oe-init-build-env`
4. edit `build/conf/bblayers.conf` and `build/conf/local.conf` according to wiki
5. `poky$ bitbake core-image-weston`

here is the error I get:

```
poky$ bitbake core-image-weston
Loading cache: 100% |#####################################################################| Time: 0:00:00
Loaded 1494 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
NOTE: Runtime target 'wpewebkit' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['wpewebkit', 'openjpeg']
ERROR: Required build target 'core-image-weston' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-weston', 'wpewebkit', 'openjpeg']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```
```
poky$ bitbake -e wpewebkit > output.log
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'brotli' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'brotli' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'brotli' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'brotli' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'brotli' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'openjpeg' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-browser/wpewebkit/wpewebkit_2.32.1.bb DEPENDS on or otherwise requires it)
ERROR: Nothing PROVIDES 'brotli' (but /home/sf/ws/yocto/poky/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
```

psaavedra, 2021-09-10:
I'm a bit surprised about that error because `openjpeg` is part of the  `/meta-openembedded/meta-oe/` layer (in `recipes-graphics/openjpeg`) and I suppose that layer is one of the one suggested in the minimal scenario of the quick start guide in Yocto. 

However, I sympathize with the idea of to provide a quick start guide. We thought that was addressed by the following documents:

* https://github.com/Igalia/meta-webkit/wiki/RPi
* https://github.com/Igalia/meta-webkit/wiki/Wandboard

Was you aware of them? If not, could you give them a try? if they are not clear enough. Do you have any suggestion to improve them? Ideas are welcome.

georgfaust, 2021-10-01:
I will, as soon I've got the time and be back.

psaavedra, 2021-11-09:
Those documents pointed by the previous comment were recently refreshed with clean-ups and updated information.

closed 2025-02-18 by github-actions[bot]