Igalia/meta-webkit #300: About the benchmark result for cog/wpewebkit and chromium-ozone-wayland on iMX8M Mini

black53421, 2021-09-08:
> Test Environment
- CPU run as 1.8GHz
- 2G RAM
- 1920x1080@60 MIPI-DSI to HDMI Display Output
- Yocto zeus (NXP's 5.4.70_2.3.0 Yocto BSP)
- iMX8MM only support wayland, so I test on wayland
- cog 0.10.0 + wpewebkit 2.32.3 + libwpe 0.10.0 + wepbackend-fdo 0.10 + proprietary Vivante driver
- chromium-ozone-wayland 83.0.4103.116

> Benchmark Result
- [JetStream1.1](https://browserbench.org/JetStream1.1/)
  - cog: **27.533**
  - chromium: 21.734
- [JetStream2](https://browserbench.org/JetStream/)
  - cog: **15.740**
  - chromium: 14.161
- [MotionMark1.2](https://browserbench.org/MotionMark1.2/)
  - cog: 1
  - chromium: **5.52**
- [bmark](https://www.wirple.com/bmark/)
  - cog: 119
  - chromium: **156**
- [Unity3D Benchmark2015](https://files.unity3d.com/jonas/benchmark2015/)
  - cog: 4381
  - chromium: **7825**
- [Antutu HTML5](https://www.antutu.com/html5/)
  - cog: 8701
  - chromium: **9293**
- [WebGL Aquarium](https://webglsamples.org/aquarium/aquarium.html)
  - cog: 5 fps (500 fishes)
  - chromium: **13 fps (500 fishes)**

> Q&A
- Cog got better result on JavaScript, but it had poor performance for 3D rendering. Is there any way I can do to improve this?

Thank you

clopez, 2021-09-08:
Here are some things you can try:

- Tune [Weston repaint window](https://github.com/Igalia/meta-webkit/wiki/PerformanceTips#tunning-weston-repaint-window)
- Try with the open source graphics drivers (Mesa/Etnaviv)
- Try running Cog without Weston, so Cog runs directly over the framebuffer. For that enable the packageconfig option ```drm``` on `cog` and then start it with the ```-P drm``` parameter. Example:
```cog -P drm https://igalia.com```
Note: is likely this `drm` mode needs the open source drivers (Mesa/Etnaviv) to work.

black53421, 2021-09-09:
Thanks for your information!

- About "Weston repaint window": I tried to set different value, but the FPS in [WebGL Aquarium](https://webglsamples.org/aquarium/aquarium.html) is not changed. I also tested in [igalia FPGA meter](https://people.igalia.com/clopez/wkbug/fps-counter/raf_onload.html), the FPS is 60.
- I will take a look on Etnaviv open source driver :)

black53421, 2021-09-14:
I already compiled the open source graphic driver (mesa 21.0.3) in my Yocto environment and Cog could run on Weston successfully.
However, I has some trouble in making Cog runs over the framebuffer. It show the following warning message when I use `-P drm`:
`(cog:1003): Cog-DRM-WARNING **: 03:33:52.139: failed to create framebuffer: No such file or directory`

The following is the message while I set `G_MESSAGES_DEBUG=all` and `LIBGL_DEBUG=verbose`:
(Weston is disabled while testing with `drm`.)
```
root@imx8mm-ddr4-evk:~# cog -P drm https://webglsamples.org/aquarium/aquarium.html
(cog:1003): Cog-DEBUG: 03:33:50.471: platform_setup: Platform name: drm
(cog:1003): Cog-DEBUG: 03:33:50.473: platform_setup: Platform plugin: libcogplatform-drm.so
(cog:1003): Cog-DRM-DEBUG: 03:33:50.479: init_config: overriding device_scale value, using 1.00 from shell
(cog:1003): Cog-DRM-DEBUG: 03:33:50.481: init_drm: enumerated device 0xaaaac3538660, available_nodes 5
(cog:1003): Cog-DRM-DEBUG: 03:33:50.481: init_drm:   DRM_NODE_PRIMARY: /dev/dri/card1
(cog:1003): Cog-DRM-DEBUG: 03:33:50.481: init_drm:   DRM_NODE_RENDER: /dev/dri/renderD128
(cog:1003): Cog-DRM-DEBUG: 03:33:50.481: init_drm: enumerated device 0xaaaac3538bc0, available_nodes 1
(cog:1003): Cog-DRM-DEBUG: 03:33:50.481: init_drm:   DRM_NODE_PRIMARY: /dev/dri/card0
(cog:1003): Cog-DRM-DEBUG: 03:33:50.482: init_drm: using device 0xaaaac3538bc0, DRM_NODE_PRIMARY /dev/dri/card0
(cog:1003): Cog-DRM-DEBUG: 03:33:50.482: init_drm: 1 connectors available
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:  connector id 35, type 11, connected, 9 usable modes
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [0]: '1920x1080', 1920x1080@60, flags 5, type 72 (preferred)
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [1]: '1920x1080', 1920x1080@60, flags 1048581, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [2]: '1920x1080', 1920x1080@50, flags 1048581, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [3]: '1920x1080', 1920x1080@30, flags 1048581, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [4]: '1280x720', 1280x720@60, flags 5, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [5]: '1280x720', 1280x720@60, flags 1048581, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [6]: '1280x720', 1280x720@50, flags 5, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [7]: '1280x720', 1280x720@50, flags 1048581, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.488: init_drm:    [8]: '800x600', 800x600@75, flags 5, type 64
(cog:1003): Cog-DRM-DEBUG: 03:33:50.494: init_drm: using connector id 35, type 11
(cog:1003): Cog-DRM-DEBUG: 03:33:50.494: init_drm: using mode [0] '1920x1080'
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
drmOpenDevice: node name is /dev/dri/renderD128
drmOpenDevice: open result is 15, (OK)
drmGetBusid returned 'NULL'
(cog:1003): Cog-DEBUG: 03:33:50.547: platform_setup: Platform = 0xaaaac34d4930

(cog:1003): GLib-GIO-WARNING **: 03:33:50.559: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
(process:1009): GLib-GIO-DEBUG: 03:33:50.560: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(process:1009): GLib-GIO-DEBUG: 03:33:50.577: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: NetworkManager not running
(process:1009): GLib-GIO-DEBUG: 03:33:50.578: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ?gio-network-monitor?
(process:1009): GLib-GIO-DEBUG: 03:33:50.582: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /home/root/.drirc: No such file or directory.
drmOpenDevice: node name is /dev/dri/renderD128
drmOpenDevice: open result is 18, (OK)
drmGetBusid returned 'NULL'
Cog-Core-Message: 03:33:50.755: <https://webglsamples.org/aquarium/aquarium.html> Load started.
[... cut ...]
(cog:1003): GLib-GIO-DEBUG: 03:33:51.679: _g_io_module_get_default: Found default implementation gnutls (GTlsBackendGnutls) for ?gio-tls-backend?
Cog-Core-Message: 03:33:51.689: <https://webglsamples.org/aquarium/aquarium.html> Loading...

(cog:1003): Cog-DRM-WARNING **: 03:33:52.139: failed to create framebuffer: No such file or directory
Cog-Core-Message: 03:33:54.756: <https://webglsamples.org/aquarium/aquarium.html> Loaded successfully.
```

NOTE: I add `"PACKAGECONFIG_append_pn-cog = " drm"` in my `local.conf` to enable `drm` support.

Do I need modify the configuration for other recipes (such as `mesa`)?

Thanks.

clopez, 2021-09-14:
Not sure what is happening and what leads to that error, but checking the log, it seems to me that it fails to open `/dev/dri/card1` which seems to be the device with the render node associated (`/dev/dri/renderD128`) in this case. Can you double-check that this devices are with `rw` permissions ? Run: ```chmod 660 /dev/dri/*```

black53421, 2021-09-15:
I found someone had mentioned a similar issue [#144](https://github.com/Igalia/cog/issues/144) in `Igalia/cog`. After checking the mesa in GitLab, I found there is a [patch](https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/12018/diffs) for this issue. I could run Cog with `-P drm` parameter after apply this patch. Unfortunately, the screen only show garbage data, so I still investigate it. Do you have any idea for this?

NOTE: `kmscube` and `glmark2` could work.

gnulux, 2022-08-17:
@black53421 : I'm trying to build a Yocto image for imx8mp board . 
Would you please to share your local.conf which include mesa driver , cog + wayland ? 

Thanks

closed 2022-11-24 by github-actions[bot]