Igalia/meta-webkit #302: Yocto/Kas Build Error: No package 'wpe-webkit-1.0' found

chrisvollorenew, 2021-09-10:
Hello, I have tried following the instructions to add meta-webkit to my kas build, and got the following error:
```
ERROR: cog-0.10.1-r0 do_configure: Execution of '/home/chris/renew/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/cog/0.10.1-r0/temp/run.do_configure.743723' failed with exit code 1:
-- The C compiler identification is GNU 11.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/chris/renew/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/cog/0.10.1-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PkgConfig: /home/chris/renew/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/cog/0.10.1-r0/recipe-sysroot-native/usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'gio-2.0>=2.44'
--   Found gio-2.0, version 2.68.3
-- Checking for module 'libsoup-2.4'
--   Found libsoup-2.4, version 2.72.0
-- Checking for module 'wpe-webkit-1.0>=2.23.91'
--   No package 'wpe-webkit-1.0' found
CMake Error at /home/chris/renew/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/cog/0.10.1-r0/recipe-sysroot-native/usr/share/cmake-3.20/Modules/FindPkgConfig.cmake:556 (message):
  A required package was not found
Call Stack (most recent call first):
  /home/chris/renew/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/cog/0.10.1-r0/recipe-sysroot-native/usr/share/cmake-3.20/Modules/FindPkgConfig.cmake:778 (_pkg_check_modules_internal)
  CMakeLists.txt:122 (pkg_check_modules)
```

Part of my config is as follows:

```
repos:
  meta-kas:

  poky:
    url: "https://git.yoctoproject.org/git/poky"
    refspec: d49e7dde842de0840cc445282d2017451278a604
    layers:
      meta:
      meta-poky:
      meta-yocto-bsp:

  meta-openembedded:
    url: "https://github.com/openembedded/meta-openembedded.git"
    refspec: 8fbcfb9f023e0eab7f122eb19607a1c757b0d245
    layers:
      meta-oe:
      meta-python:
      meta-networking:
      meta-filesystems:
      meta-webserver:
      meta-perl:

  meta-security:
    url: "https://git.yoctoproject.org/git/meta-security"
    refspec: 5050d1267ad41288c903086030594f8702bfa039

  meta-webkit:
    url: "https://github.com/Igalia/meta-webkit.git"
    refspec: 6afa4fe73105a18b579c23221234cc4bd1d39755
```

```
    # gtk webkit
    DISTRO_FEATURES_append = " opengl x11"
    IMAGE_INSTALL_append = " webkitgtk-bin cog"
    PACKAGECONFIG_pn-cog = "webkitgtk"
```

I am using branch main for meta-webkit. I have seen issue #88 but I am not sure how it applies or if it applies to this issue. Any help is appreciated! Thank you!

philn, 2021-09-12:
WebKitGTK support was removed from cog. If you want a WebKitGTK browser you can build MiniBrowser maybe.
If you want to keep using Cog, you need to add wpewebkit to your image.

philn, 2021-09-13:
Done!

closed 2021-09-13 by philn