Igalia/meta-webkit #331: webkitgtk: Build issue on i.MX6 with wayland target

dharanlinux, 2021-11-02:
Hi Team,

While building with wayland target, facing below build issue.

The default yocto build passing the '-DWL_EGL_PLATFORM' options, still we facing the issue.

Hardware: imx6qsabresd

`
                  -fdebug-prefix-map=/home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/recipe-sysroot -fno-strict-aliasing -fno-exceptions -fno-rtti -DNDEBUG -fPIC -DLINUX -DWL_EGL_PLATFORM -pthread -std=c++17 -MD -MT Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/egl/GLContextEGL.cpp.o -MF Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/egl/GLContextEGL.cpp.o.d -o Source/WebCore/CMakeFiles/WebCore.dir/platform/graphics/egl/GLContextEGL.cpp.o -c /home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/webkitgtk-2.34.1/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp
| /home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/webkitgtk-2.34.1/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp: In static member function 'static std::unique_ptr<WebCore::GLContextEGL> WebCore::GLContextEGL::createWindowContext(GLNativeWindowType, WebCore::PlatformDisplay&, EGLContext)':
| /home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/webkitgtk-2.34.1/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp:198:59: error: invalid 'static_cast' from type 'GLNativeWindowType' {aka 'long long unsigned int'} to type 'EGLNativeWindowType' {aka 'wl_egl_window*'}
|   198 |         surface = eglCreateWindowSurface(display, config, static_cast<EGLNativeWindowType>(window), nullptr);
|       |                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| In file included from /home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/webkitgtk-2.34.1/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp:20:
| /home/linux/Download/imx-yocto-bsp/build-weston/tmp/work/cortexa9t2hf-neon-mx6qdl-poky-linux-gnueabi/webkitgtk/2.34.1-r0/webkitgtk-2.34.1/Source/WebCore/platform/graphics/egl/GLContextEGL.h:32:8: note: class type 'wl_egl_window' is incomplete
|    32 | struct wl_egl_window;
`

psaavedra, 2021-11-03:
Can you share here the Build Configuration Bitbake output and also de local.conf?

dharanlinux, 2021-11-15:
Hi Team,

Do we have fixup for this issue?

psaavedra, 2021-12-09:
Sorry I didn't tested that yet. Let me first try to reproduce the issue in local.

psaavedra, 2021-12-10:
It seems that the `EGLNativeWindowType` changes depending on the EGL implementation. In https://bugs.webkit.org/show_bug.cgi?id=178090 @aperezdc  uses a plain C cast expression. The reason is because a plain C cast expression works in all cases. On the contrary, `reinterpret_cast` for pointers (only if they are 64-bit wide and not for other cases) and `static_cast` for numeric types (and when needed they get extended to 64-bit) but not for pointers. This error should be addressed directly in WebKit.

aperezdc, 2021-12-15:
In this case the incomplete definition of `wl_egl_window` is intended, note how in `GLContextEGL.h` there is a forward-declaration:

```h
#if PLATFORM(WAYLAND)
#include "WlUniquePtr.h"
struct wl_egl_window;
#endif
```

Indeed if one tries to compile the following reduced case with `g++ -c` it fails with a similar error:

```cpp
struct wl_egl_window;            // Forward declaration
extern long long unsigned value; // The type of GLNativeWindowType

// Type definition from the EGL khrplatform.h header
typedef struct wl_egl_window* EGLNativeWindowType;

int main() {                                                                                
    return static_cast<EGLNativeWindowType>(value) == 0;                         
}
```

aperezdc, 2021-12-16:
This is where `GLNativeWindowType` gets defined, in `GLContext.h` (slightly reformatted for clarity):

```cpp
#if USE(EGL) && !PLATFORM(GTK)
    #if PLATFORM(WPE)
        // FIXME: For now default to the GBM EGL platform, but this should really be
        // somehow deducible from the build configuration.
        #define __GBM__ 1
    #endif // PLATFORM(WPE)

    #include <EGL/eglplatform.h>
    typedef EGLNativeWindowType GLNativeWindowType;

#else // !USE(EGL) || PLATFORM(GTK)
    typedef uint64_t GLNativeWindowType;
#endif
```

Which means we have a case here where `GLNativeWindowType` is defined to be an integer instead of a pointer for the GTK port. While that is okay on X11/GLX where windows are represented by their XID (an integer), for builds where we have GTK with Wayland support but no X11 then I think this might be a problem.

aperezdc, 2021-12-16:
@dharanlinux Could you share the CMake command from the `tmp/work/*/wpewebkit/2.34.*/temp/run.do_configure` file? That way we would know exactly how the build is being configured.

dharanlinux, 2021-12-16:
@aperezdc Please check the below details. If you need further info please let me know.

```
-DPORT=GTK -DENABLE_INTROSPECTION=OFF -DENABLE_GTKDOC=OFF -DENABLE_MINIBROWSER=ON -DUSE_SOUP2=ON -G Ninja -DENABLE_BUBBLEWRAP_SANDBOX=OFF -DENABLE_SPELLCHECK=ON -DENABLE_GAMEPAD=OFF -DENABLE_GEOLOCATION=OFF -DENABLE_GLES2=ON -DUSE_LD_GOLD=ON -DENABLE_JIT=ON -DENABLE_C_LOOP=OFF -DUSE_LIBHYPHEN=OFF -DUSE_LIBNOTIFY=OFF -DUSE_LIBSECRET=ON -DENABLE_OPENGL=ON -DUSE_OPENJPEG=ON -DCMAKE_BUILD_TYPE=Release -DUSE_SYSTEMD=ON -DENABLE_VIDEO=ON -DENABLE_WAYLAND_TARGET=ON -DENABLE_WEB_CRYPTO=ON -DENABLE_WEBGL=ON -DUSE_WOFF2=ON -DUSE_WPE_RENDERER=ON -DENABLE_X11_TARGET=ON \
```

dharanlinux, 2022-01-10:
Hi @aperezdc , any update on this issue. Do we need further information from my build setup?

adahl5, 2022-04-12:
I've also just run into this exact issue. I don't know enough about C(++) to grasp what's going on. I've tried just de-referencing the 'window' variable from the wl_egl_window* type (likely a bad idea) but that hasn't worked. Is there a missing dependency I'm overlooking? I've seen plenty of examples of people building webkitgtk for the iMX6, so it'd be odd for it to suddenly break.

I'm also getting dozens of -Wcast-align warnings but I'm not sure if that's related.

matteoscordino, 2023-01-13:
I am having a very similar issue on webkit 2.38.3, building for i.MX8, so this shouldn't be marked as stale.

```
| /mnt//work/compulab-nxp-bsp/build-ucm-imx8m-plus/tmp/work/armv8a-mx8mp-poky-linux/webkitgtk/2.38.3-r0/webkitgtk-2.38.3/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp: In static member fu
nction 'static std::unique_ptr<WebCore::GLContextEGL> WebCore::GLContextEGL::createWindowContext(GLNativeWindowType, WebCore::PlatformDisplay&, EGLContext)':                                                        
| /mnt//work/compulab-nxp-bsp/build-ucm-imx8m-plus/tmp/work/armv8a-mx8mp-poky-linux/webkitgtk/2.38.3-r0/webkitgtk-2.38.3/Source/WebCore/platform/graphics/egl/GLContextEGL.cpp:198:59: error: inval
id 'static_cast' from type 'GLNativeWindowType' {aka 'long unsigned int'} to type 'EGLNativeWindowType' {aka 'wl_egl_window*'}                                                                                       
|   198 |         surface = eglCreateWindowSurface(display, config, static_cast<EGLNativeWindowType>(window), nullptr);                                                                                              
|       |                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                         
| In file included from /mnt//work/compulab-nxp-bsp/build-ucm-imx8m-plus/tmp/work/armv8a-mx8mp-poky-linux/webkitgtk/2.38.3-r0/webkitgtk-2.38.3/Source/WebCore/platform/graphics/egl/GLContextEGL.cp
p:20:                                                                                                                                                                                                                
| /mnt//work/compulab-nxp-bsp/build-ucm-imx8m-plus/tmp/work/armv8a-mx8mp-poky-linux/webkitgtk/2.38.3-r0/webkitgtk-2.38.3/Source/WebCore/platform/graphics/egl/GLContextEGL.h:32:8: note: class type
 'wl_egl_window' is incomplete                                                                                                                                                                                       
|    32 | struct wl_egl_window;                                                                                                                                                                                      
|       |        ^~~~~~~~~~~~~ 
```

aperezdc, 2023-04-23:
I've hit this while trying to build with the Mali driver (`sunxi-mali-utgard` in Buildroot, [patched](https://o.perezdecastro.org/buildroot-sunxi-mali-utgard-wayland.diff) to allow installing the Wayland variant). So I double-checked things again: this happens when GTK3 itself is built only with Wayland support but no X11; the fix in [bug #178090](https://bugs.webkit.org/show_bug.cgi?id=178090) fixed a location where `EGLNativeWindowType` (a.k.a. `struct wl_egl_window*`) is cast to a 64-bit integer, but in `GLContextEGL.cpp` (or `GLContext.cpp` in newer WebKit versions) there is a cast for the opposite conversion which needs a similar treatment.

I have reported this upstream, and I will make a version of the patch that people can apply to the releases from the 2.38 branch. There is also one more issue with 2.38.x and drivers which are strictly OpenGL ES 2, without ES 3 features:

- https://bugs.webkit.org/show_bug.cgi?id=255846
- https://bugs.webkit.org/show_bug.cgi?id=255847

aperezdc, 2023-04-23:
Combined patch for 2.38.x here: https://github.com/WebKit/WebKit/pull/13084

closed 2023-07-31 by github-actions[bot]