Igalia/meta-webkit #334: Video Encoding Issue on iMX8M Mini

miltechniks, 2021-11-12:
Hi,

currently I have an Issue with Video-Playback where with some VP8 encoded videos there is an Issue with the picture.
That's how it should look like:
[image]
That's how it look like:
[image]

So the Issue is this bar on the top of the Image. This bar is there, depending on the source video.

I already had a look to this issue https://github.com/Igalia/meta-webkit/issues/183 but it seems to be another problem, as it hasn't solved the issue.
When I play the same Video with gst-play-1.0 from the Web-Source there is no Issue.
So, to play it bad:  cog --platform=fdo https://dl8.webmfiles.org/big-buck-bunny_trailer.webm
To play it good: gst-play-1.0 https://dl8.webmfiles.org/big-buck-bunny_trailer.webm

I'm using the Vivante OpenGL Driver due to the fact, that the Etnaviv OpenGL isn't possible to compile for the iMX8M Mini, or at least I haven't found a solution for that.

What am I missing to get this Issue solved?
Thanks for your help.

philn, 2021-11-15:
Please provide logs: https://github.com/Igalia/meta-webkit/wiki/Providing-useful-GStreamer-Zero-copy-issue-reports
Which video-sink is used by gst-play-1.0 ?

philn, 2021-11-16:
I suspect this might the same as https://github.com/Igalia/cog/issues/364

miltechniks, 2021-11-16:
Thanks for your answer. I used the waylandsink (gst-play-1.0 --videosink=waylandsink https://dl8.webmfiles.org/big-buck-bunny_trailer.webm). With the glimagesink it's similar, except that it's not fullscreen and somewhere on the window.

### Informations and logs

Version: GStreamer 1.18.0
Plugins: gstreamer1.0-plugins-imx, gstreamer1.0-plugins-good, gstreamer1.0-plugins-bad, gstreamer1.0-plugins-base and a few others included by wpewebkit.inc
Yocto Release: Hardknott
Layers:
BBLAYERS = " \
  ${BSPDIR}/sources/poky/meta \
  ${BSPDIR}/sources/poky/meta-poky \
  \
  ${BSPDIR}/sources/meta-openembedded/meta-oe \
  ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
  ${BSPDIR}/sources/meta-openembedded/meta-python \
  \
  ${BSPDIR}/sources/meta-freescale \
  ${BSPDIR}/sources/meta-freescale-3rdparty \
  ${BSPDIR}/sources/meta-freescale-distro \
"

# i.MX Yocto Project Release layers
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-bsp"
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-sdk"
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-ml"
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-v2x"
BBLAYERS += "${BSPDIR}/sources/meta-nxp-demo-experience"

BBLAYERS += "${BSPDIR}/sources/meta-browser/meta-chromium"
BBLAYERS += "${BSPDIR}/sources/meta-clang"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-gnome"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-networking"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-filesystems"
BBLAYERS += "${BSPDIR}/sources/meta-qt5"
BBLAYERS += "${BSPDIR}/sources/meta-python2"
BBLAYERS += "${BSPDIR}/sources/meta-webkit"

BBLAYERS ?= " \
    ${BSPDIR}/sources/meta-openembedded/oe-core/meta \
    ${BSPDIR}/sources/meta-openembedded/meta-openembedded/meta-oe \
    ${BSPDIR}/sources/meta-openembedded/meta-openembedded/meta-perl \
    ${BSPDIR}/sources/meta-openembedded/meta-openembedded/meta-python \
    ${BSPDIR}/sources/meta-openembedded/meta-openembedded/meta-networking \
    ${BSPDIR}/sources/meta-security"

Here are the logs: [Trailer_Log.zip](https://github.com/Igalia/meta-webkit/files/7545562/Trailer_Log.zip)

Regarding the other issue I'm not sure if it is the same.  In my case the video is playing and loading.
.

philn, 2021-11-16:
Is rendering OK with this too? `gst-play-1.0 --videosink "imxvideoconvert_g2d ! glimagesink"` ?

miltechniks, 2021-11-16:
The Issue seems to be in the imxvideoconvert_g2d, as with this command the rendering is wrong as well. When I remove it and only use  gst-play-1.0 --videosink "glimagesink" everything is fine.

philn, 2021-11-16:
Can you report this to Freescale/NXP then? Or whoever maintains this element ;)

miltechniks, 2021-11-16:
Yeah, I'm going to report this. Thanks for your help.

I removed the imxvideoconvert_g2d within the webkit for testing purposes and then everything runs as expected. If anyone is interested, that's the patch for it:
[0003-Remove-imxvideoconvert_g2d-plugin-in-gstreamer-pipeline.txt](https://github.com/Igalia/meta-webkit/files/7549511/0003-Remove-imxvideoconvert_g2d-plugin-in-gstreamer-pipeline.txt)

closed 2021-11-16 by miltechniks