Igalia/meta-webkit #335: OOM when streaming webcam video via webrtc on iMX8M mini

miltechniks, 2021-11-18:
Hi,

I have an Issue when Streaming a video from a camera, connected via CSI to the iMX8M Mini, via webrtc to somewhere else. The Memory-Usage of the WPEWebProcess is increasing continously and after some time the system crashes. I currently can't find the cause of this increasing memory and already tested the gstreamer-pipeline which I extracted from the Log, but the pipeline runs smoothly.

With this webpage is reproduceable: cog --platform=fdo --set-permissions=all --enable-media-stream=true  https://janus.conf.meetecho.com/echotest.html

That's the Pipeline I tested: gst-launch-1.0 v4l2src ! videoscale ! videoconvert ! videorate ! capsfilter ! valve ! tee ! queue ! autovideosink

Another Issue, which seems to be related to this, is that after stopping the video and restarting it again, there is no video stream anymore and the memory usage doesn't increase.

Here are the gstreamer logs: [GStreamer.zip](https://github.com/Igalia/meta-webkit/files/7563657/GStreamer.zip)

miltechniks, 2021-11-18:
Is there any way, to get logs from the WPEWebProcess? I already tried an strace, but up to now I haven't found anything usefull in it. But here it is attached:
[strace.log](https://github.com/Igalia/meta-webkit/files/7563801/strace.log)

miltechniks, 2021-12-01:
I reproduced the error on a Raspberry Pi 4 as well. So it's not something related to the NXP-Board and the NXP Libraries and even a newer version of Webkit doesn't fix the Issue (I tried it with commit b067fbd3e80e53b9359f99367a6c2c342cdd75ce).

philn, 2021-12-10:
This is clearly a WebKit bug, forwarded upstream: https://bugs.webkit.org/show_bug.cgi?id=234134

closed 2021-12-10 by philn