Igalia/meta-webkit #336: framebuffer rpi4

admindotnu, 2021-11-24:
I am a bit confused about what build options we need to use, if it is possible at all.

We are used to use webkit and display directly on the framebuffer of the rpi3.
Is that possible for the rpi4 / cm4 too ?
If yes, what local.conf options do i need to use ?

psaavedra, 2021-12-09:
Please follow the instructions documented in https://github.com/Igalia/meta-webkit/wiki/RPi#image-with-mesa-vc4-driver-drm-and-wpebackend-fdo-heavy_check_mark and change the MACHINE to `raspberry4-64`. Also notice the warning note referring to work around commented in https://github.com/Igalia/meta-webkit/issues/275#issuecomment-963909594

admindotnu, 2021-12-13:
Hi @psaavedra 

Will do a test compile soon !. Will close the issue, if needed will reply in this issue again.

closed 2021-12-13 by admindotnu