Igalia/meta-webkit #363: Memory pressure handler not releasing memory

d21d3q, 2022-04-09:
Continuation of issue [WebPlatformForEmbedded/WPEWebKit#82](https://github.com/WebPlatformForEmbedded/WPEWebKit/issues/826).

I am running `meta-webkit` on rpi with rdk backend. 

I have an SPA (Vue), it receives some data over websocket, and it seems that the more data it receives, the sooner it crashes. 
I am observing memory usage growth which leads to system unresponsiveness.

I've patched wpewebkit:
```
diff --git a/Source/WTF/wtf/MemoryPressureHandler.cpp b/Source/WTF/wtf/MemoryPressureHandler.cpp
--- a/Source/WTF/wtf/MemoryPressureHandler.cpp
+++ b/Source/WTF/wtf/MemoryPressureHandler.cpp
@@ -105,7 +105,7 @@ static size_t thresholdForMemoryKillOfInactiveProcess(unsigned tabCount)
 #else
     size_t baseThreshold = tabCount > 1 ? 3 * GB : 2 * GB;
 #endif
-    return std::min(baseThreshold, static_cast<size_t>(ramSize() * 0.9));
+    return std::min(baseThreshold, static_cast<size_t>(ramSize() * 0.3 * 0.9));
 }
 
 void MemoryPressureHandler::setPageCount(unsigned pageCount)
@@ -128,7 +128,7 @@ Optional<size_t> MemoryPressureHandler::thresholdForMemoryKill()
 
 static size_t thresholdForPolicy(MemoryUsagePolicy policy)
 {
-    const size_t baseThresholdForPolicy = std::min(3 * GB, ramSize());
+    const size_t baseThresholdForPolicy = std::min(3 * GB, static_cast<size_t>(ramSize() * 0.3));
 
 #if PLATFORM(IOS_FAMILY)
     const double conservativeThresholdFraction = 0.5
```

As a result I am observing changing policy to `Strict` (previously it was at most `conservative`). But Memory usage keeps building up. Rpi has ~600Mb of ram, after almost 2 days of uptime, htop is reporting 245Mb of RES memory usage. (threshold should be set to around 180Mb).  
Here is plot from first value from logs: ` res+swap = 89944064/89972736/28672`:  

[image]

Any ideas how I can debug it?

philn, 2022-04-09:
Looks like your app is leaking? Try massif maybe?

d21d3q, 2022-04-12:
I've noticed slight memory use growth under Chrome, but on webkit it seemed higher. I'll investigate it and will update. It can take some time because testing it is a little bit time consuming.

philn, 2022-04-12:
Keep in mind there's so much Chrome-specific web-apps nowadays, having the same code working in Chrome does not necessarily mean it's a WebKit bug.

d21d3q, 2022-10-24:
@github-actions, I'll be working on that issue during following days and I'll close that isse once I confirm that it is not webkit related.

closed 2023-01-30 by github-actions[bot]