Igalia/meta-webkit #379: rpi4: vc4: Memory Error when playing multiple mp4 videos

perrin-1, 2022-06-09:
**Describe the bug**
Yocto 4 Kirkstone or Yocto 3.4 honister, core-image-weston, cog

We have a web-app that is used to play fitness training videos. Videos are played through cog as html5 video elements. Videos are mp4, h264 1080p video files. Each training consists of many steps, each step is an individual video. 
The website has a "Next" button to skip to the next training step. Upon clicking "Next" the current video is stopped and a new video url is added to the html5 video element.
Depending on the amount of video-ram that is allocated on the rpi4, one can play from 3-8 videos. After that, pressing the "Next" button does not show any video. dmesg on the rpi4 shows the following errors

[  993.563815] swiotlb_tbl_map_single: 140 callbacks suppressed
[  993.563832] vc4-drm gpu: swiotlb buffer is full (sz: 3571712 bytes), total 32768 (slots), used 94 (slots)
[  993.598321] vc4-drm gpu: swiotlb buffer is full (sz: 4194304 bytes), total 32768 (slots), used 62 (slots)
[  993.632091] vc4-drm gpu: swiotlb buffer is full (sz: 3571712 bytes), total 32768 (slots), used 94 (slots)
[  993.663334] vc4-drm gpu: swiotlb buffer is full (sz: 4194304 bytes), total 32768 (slots), used 62 (slots)
[  993.696107] vc4-drm gpu: swiotlb buffer is full (sz: 3571712 bytes), total 32768 (slots), used 94 (slots)
[  993.733370] vc4-drm gpu: swiotlb buffer is full (sz: 4194304 bytes), total 32768 (slots), used 62 (slots)
[  993.765738] vc4-drm gpu: swiotlb buffer is full (sz: 3571712 bytes), total 32768 (slots), used 94 (slots)
[  993.799292] vc4-drm gpu: swiotlb buffer is full (sz: 4194304 bytes), total 32768 (slots), used 62 (slots)
[  993.831739] vc4-drm gpu: swiotlb buffer is full (sz: 3571712 bytes), total 32768 (slots), used 94 (slots)
[  993.864497] vc4-drm gpu: swiotlb buffer is full (sz: 4194304 bytes), total 32768 (slots), used 62 (slots)

**Sample HTML Code and MP4 videos to reproduce can be provided**

I dont't know if the error is related to cog/wpewebkit or other layers below.. Please help...

**To Reproduce**
Build config:
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi4"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.1"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:45d7615dfef8093a2467d52ca1130e62db6a1187"
meta-oe              
meta-python          
meta-networking      = "kirkstone:fcc7d7eae82be4c180f2e8fa3db90a8ab3be07b7"
meta-raspberrypi     = "kirkstone:0135a02ea577bd39dd552236ead2c5894d89da1d"
meta-webkit          = "main:f707dbab77cc0f7f5abc2c13c3f3fe6db4e569c4"`

local.conf:
MACHINE = "raspberrypi4" 
MACHINE_FEATURES:append = " vc4graphics"
GPU_MEM_256 = "128"
GPU_MEM_512 = "196"
GPU_MEM_1024 = "396"
DISTRO_FEATURES:append = " opengl egl wayland"
EXTRA_IMAGE_FEATURES = "debug-tweaks"
IMAGE_FEATURES:append = " ssh-server-dropbear hwcodecs"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
IMAGE_INSTALL:append = " wpewebkit cog kernel-modules"

**Expected behavior**
There should not be a limit, when playing multiple mp4 videos

philn, 2022-06-09:
Might be related with your webapp leaking `<video>` elements too. Have you checked in that direction?

perrin-1, 2022-06-10:
Hello philn, 
thank you for your reply. We took some time to analyze if there are any leaking `<video>` tags - however we could not find any of these.

To reproduce the error, we created two plain html/JS pages for you:

Demo 1: Version that replaces the video tag every time a new video gets played: [Demo1 - new Video Tag](https://videotest.dev.omsag.de/demo-new-video-tag.html)  
(edit: we noted that this demo also has a memory leak when using Chrome and Firefox - so this is some edge case)

Demo 2: Version that replaces the `src` element with the URL of the next video [Demo2 - new Video SRC](https://videotest.dev.omsag.de/demo-one-video-tag.html)

Both versions crash cog after 2-8 videos on the RPI4. When crashing cog prints an error message to the console:

0:00:03.612463632 2298 0xa44848 ERROR glbasetexture gstglmemory.c:890:_gl_tex_copy: Cannot copy External OES textures

and dmesg contains the error concerning the vc4-drm gpu: swiotlb buffer

closed 2023-01-29 by github-actions[bot]