Igalia/meta-webkit #382: gstreamer1.0-plugins-good-souphttpsrc not found

mgonzalezg9, 2022-06-14:
I am building an image for IMX8 processor, so I added meta-webkit on top of BSP layer. As I am using Poky sumo (2.5), I also added meta-gstreamer-1.0 because as it is highly recommended. However, when building Yocto it complains about missing `gstreamer1.0-plugins-good-souphttpsrc`:

```
WARNING: wpewebkit-2.22.5-r0 do_package_qa: QA Issue: wpewebkit rdepends on gstreamer1.0-plugins-good-souphttpsrc, but it isn't a build dependency? [build-deps]
ERROR: fsl-image-qt5-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/recipe-sysroot-native/usr/bin/dnf -y -c /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/rootfs/etc/yum.repos.d --repofrompath=oe-repo,/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/oe-rootfs-repo --installroot=/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/rootfs --setopt=logdir=/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/temp --nogpgcheck install locale-base-en-us locale-base-en-gb gstreamer1.0-plugins-good-qt imx-qtapplications packagegroup-core-tools-debug ttf-dejavu-sans-mono ttf-dejavu-serif psplash packagegroup-qt5-demos packagegroup-fsl-tools-audio packagegroup-core-boot packagegroup-core-tools-profile packagegroup-fsl-gstreamer1.0-full ttf-dejavu-common packagegroup-fsl-tools-gpu-external packagegroup-tools-bluetooth weston-init packagegroup-core-tools-testapps packagegroup-core-nfs-server packagegroup-fsl-tools-gpu packagegroup-fsl-gstreamer1.0 packagegroup-fsl-tools-benchmark qtwayland packagegroup-core-full-cmdline run-postinsts packagegroup-base-extended qtwayland-plugins cog packagegroup-fsl-tools-testapps packagegroup-core-ssh-dropbear wpewebkit ttf-dejavu-sans' returned 1:
Added oe-repo repo from /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/imx8mqevk-poky-linux/fsl-image-qt5/1.0-r0/oe-rootfs-repo
Last metadata expiration check: 0:00:01 ago on Mon 13 Jun 2022 10:46:11 AM UTC.
Error: 
 Problem 1: conflicting requests
  - nothing provides gstreamer1.0-plugins-good-souphttpsrc needed by wpewebkit-2.22.5-r0.aarch64_mx8m
 Problem 2: package cog-0.2.0-r0.aarch64 requires libWPEWebKit-0.1.so.2()(64bit), but none of the providers can be installed
  - conflicting requests
  - nothing provides gstreamer1.0-plugins-good-souphttpsrc needed by wpewebkit-2.22.5-r0.aarch64_mx8m
```

**Steps to reproduce the behavior:**
1. Append the following to local.conf:
```
IMAGE_INSTALL_append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
```
2. Add the layers meta-webkit and meta-gstreamer-1.0
3. run bitbake fsl-image-qt5
4. Wait for the error

**Environment:**
 - yocto release: sumo
 - hardware target: IMX8MQ

psaavedra, 2022-06-28:
Could you try to use the branch `dunfell` for meta-webkit. The dunfell branch is also compatible with sumo (https://github.com/Igalia/meta-webkit/blob/dunfell/conf/layer.conf#L19) but more modern.

mgonzalezg9, 2022-07-08:
Now I am receiving the following error, this time with meta-webkit:
```
ubuntu@ip-172-31-12-24:~/ibr210-yocto-bsp-20220516/ibr210$ bitbake fsl-image-qt5 
WARNING: You have included the meta-gnome layer, but 'x11' has not been enabled in your DISTRO_FEATURES. Some bbappend files may not take effect. See the meta-gnome README for details on enabling meta-gnome support.
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:01:11
Parsing of 2635 .bb files complete (0 cached, 2635 parsed). 3605 targets, 453 skipped, 9 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
NOTE: Multiple providers are available for runtime ptpd (ptpd, ptpd-qoriq)
Consider defining a PREFERRED_RPROVIDER entry to match ptpd
ERROR: Nothing PROVIDES 'brotli' (but /home/ubuntu/ibr210-yocto-bsp-20220516/sources/meta-webkit/recipes-graphics/woff2/woff2_1.0.2.bb DEPENDS on or otherwise requires it)
NOTE: Runtime target 'cog' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['cog', 'wpewebkit', 'woff2', 'brotli']
ERROR: Required build target 'fsl-image-qt5' has no buildable providers.
Missing or unbuildable dependency chain was: ['fsl-image-qt5', 'cog', 'wpewebkit', 'woff2', 'brotli']

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code
```

mgonzalezg9, 2022-07-19:
@psaavedra any thoughts on this?

psaavedra, 2022-07-22:
> @psaavedra any thoughts on this?

you could try to revert this commit on your local checkout of meta-webkit:  https://github.com/Igalia/meta-webkit/commit/a4c12eb4392e8b3da1b0fb4e59adbcf0e2fd5a00

mgonzalezg9, 2022-07-25:
It appeared another error, apparently related to gstreamer1.0
```
ERROR: wpewebkit-2.32.1-r0 do_compile: Function failed: do_compile (log file is located at /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/temp/log.do_compile.26527)
ERROR: Logfile of failure stored in: /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/temp/log.do_compile.26527
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: VERBOSE=1 cmake --build /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/build --target all -- -j 4
| [1/2896] cd /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore && /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/recipe-sysroot-native/usr/bin/perl-native/perl /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/bindings/scripts/generate-bindings-all.pl --defines "LANGUAGE_JAVASCRIPT  ENABLE_3D_TRANSFORMS ENABLE_ASYNC_SCROLLING ENABLE_AUTOCAPITALIZE ENABLE_CHANNEL_MESSAGING ENABLE_CONTENT_EXTENSIONS ENABLE_CONTEXT_MENUS ENABLE_CSS_BOX_DECORATION_BREAK ENABLE_CSS_COMPOSITING ENABLE_CSS_CONIC_GRADIENTS ENABLE_CSS_SELECTORS_LEVEL4 ENABLE_DFG_JIT ENABLE_FILTERS_LEVEL_2 ENABLE_FTL_JIT ENABLE_FTPDIR ENABLE_FULLSCREEN_API ENABLE_INDEXED_DATABASE ENABLE_INDEXED_DATABASE_IN_WORKERS ENABLE_INTERSECTION_OBSERVER ENABLE_JIT ENABLE_MATHML ENABLE_MEDIA_CONTROLS_SCRIPT ENABLE_MEDIA_SOURCE ENABLE_MEMORY_SAMPLER ENABLE_MHTML ENABLE_NETWORK_CACHE_STALE_WHILE_REVALIDATE ENABLE_NOTIFICATIONS ENABLE_PERIODIC_MEMORY_MONITOR ENABLE_PUBLIC_SUFFIX_LIST ENABLE_REMOTE_INSPECTOR ENABLE_RESIZE_OBSERVER ENABLE_RESOURCE_LOAD_STATISTICS ENABLE_RESOURCE_USAGE ENABLE_SAMPLING_PROFILER ENABLE_SERVICE_WORKER ENABLE_SHAREABLE_RESOURCE ENABLE_TOUCH_EVENTS ENABLE_UNIFIED_BUILDS ENABLE_USER_MESSAGE_HANDLERS ENABLE_VIDEO ENABLE_VIDEO_USES_ELEMENT_FULLSCREEN ENABLE_WEBASSEMBLY ENABLE_WEBASSEMBLY_B3JIT ENABLE_WEBDRIVER ENABLE_WEBGL ENABLE_WEB_AUDIO ENABLE_WEB_CRYPTO ENABLE_XSLT USE_GSTREAMER_GL USE_GSTREAMER_TEXT_SINK USE_OPENJPEG USE_SOUP2 USE_SYSTEMD USE_WOFF2" --generator JS --outputDir /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/build/DerivedSources/WebCore --idlFilesList /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/build/Source/WebCore/idl_files_WebCoreBindings.tmp --ppIDLFilesList /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/build/Source/WebCore/pp_idl_files_WebCoreBindings.tmp --preprocessor "\"/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++\" -E -P -x c++" --idlAttributesFile /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/bindings/scripts/IDLAttributes.json --supplementalDependencyFile /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/build/DerivedSources/WebCore/supplemental_dependency.tmp --numOfJobs 4 --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/css --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/dom --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/fileapi --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/html --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/inspector --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/page --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/plugins --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/storage --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/svg --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/workers --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/xml --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/Modules/airplay --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/Modules/cache --include /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/Modules/credentialmanag [... cut ...] 0-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/bindings/scripts/preprocessor.pm --generatorDependency /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/build/cmakeconfig.h --generatorDependency /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/wpewebkit-2.32.1/Source/WebCore/bindings/scripts/CodeGeneratorJS.pm
(error continues...)
```

psaavedra, 2022-09-08:
Sorry the error says not too much to me. Maybe you could try to search for `error:` inside of the file `/home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/recipe-sysroot-native/usr/bin/ccache /home/ubuntu/ibr210-yocto-bsp-20220516/ibr210/tmp/work/aarch64-mx8m-poky-linux/wpewebkit/2.32.1-r0/tmp/log.do_compile`

closed 2022-12-15 by github-actions[bot]