Igalia/meta-webkit #383: Allow only hardware supported codecs

Ossanes, 2022-06-24:
In order to get the better performance in some boards, I'm looking for an way to allow only codecs supported by hardware.

In my tests I'm using a Freescale imx6 Sabre to run videos on Youtube TV (www.youtube.com/tv). But for some reason, Youtube is providing a stream using a not hardware compatible plugin. So it tries to run using software (without hardware acceleration). In this case is the VP9 codec that isn't compatible with this board. But the solution need to be generic and should works in any board. So I think that the best way to this solution is to find a way to tell to COG to ask for YouTube to only send streams compatible with the current hardware.

I generated a log running `XDG_RUNTIME_DIR=/run GST_DEBUG="*:9" GST_DEBUG_NO_COLOR=1 GST_DEBUG_DUMP_DOT_DIR=/tmp GST_DEBUG_FILE=/t
mp/gst.log cog -P wl --user-agent="Mozilla/5.0 (SMART-TV; Linux; Tizen 4.0)" https://www.youtube.com/tv`
...and I could note that the followings codecs were registered:

```
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Hardware decoder codec pattern registered: x-vp8
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Hardware decoder codec pattern registered: mp4v*
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Hardware decoder codec pattern registered: x-h264
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: vp9.0
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: x-vp9
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: vp09*
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Hardware decoder codec pattern registered: avc*
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: mp4a*
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Hardware decoder codec pattern registered: vp8
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: vp9
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Hardware decoder codec pattern registered: vp8.0
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: mpeg
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: x-vorbis
webkitregistryscanner GStreamerRegistryScanner.cpp:215:GStreamerRegistryScanner: Software decoder codec pattern registered: vorbis
```
But VP9 (by software) is automatically chosen.

This part of log is generated by GStreamerRegistryScanner.cpp
https://github.com/WebKit/WebKit/blob/b9288e3b242cb8a8830c2ba860e210310047a747/Source/WebCore/platform/graphics/gstreamer/GStreamerRegistryScanner.cpp#L215
There is some methods to verify the codecs available and some properties about them. But I couldn't find a "pretty" way to intervene to allow only hardware codecs.

How I can I change this behaviour to allow only codecs supported by hardware?

philn, 2022-06-26:
Can you try to set the sw decoders rank to 0 with an env var, for instance: `export GST_PLUGIN_FEATURE_RANK=x264dec:0,vp8dec:0,vp9dec:0` and so on.

otavio, 2022-06-27:
@philn does WPEWebKit or Cog offer a way to do it automatically for sw decoders?

philn, 2022-06-27:
AFAIK, nope

closed 2022-10-31 by github-actions[bot]