Igalia/meta-webkit #387: Can't use cog on imx8 yocto(zeus)  build

gnulux, 2022-08-11:
**Describe the issue**

I created a Yocto(zeus) buid for dev board based on imx8mp . 
I used meta-webkit 

Her is my image.bb 

```
require recipes-images/images/headless-image.bb

SUMMARY =  "This image is designed to be a development image for complex \
            multi application scenarios under weston or for desktop use."

IMAGE_FEATURES += "splash package-management ssh-server-openssh hwcodecs"

REQUIRED_DISTRO_FEATURES = "wayland"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PACKAGECONFIG_pn-wpebackend-fdo = "wayland"
PACKAGECONFIG_pn-cog = "fdo"

IMAGE_INSTALL += "\
    weston \
    weston-init \
    wpewebkit \
    libwpe \
    cog \
    qt-wpe-simple-browser \
    browserperfrunner \
    packagegroup-base \
    \
    packagegroup-gstreamer \
```
When i boot on the image i start wayland , i link libWPEBackend-fdo-1.0.so.1.6.1 to libWPEBackend-default.so then from the weston-console when i try to launch cog https://www.google.com

I have this error : 

```
** (cog:1187): CRITICAL3032.888524] audit: type=1701 audit(3032.884:22): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=1187 comm="cog" exe="/usr/bin/cog" sig=5 res=1
[0m **: 00:50:32.882: WebKitWebViewBackend* webkit_web_view_backend_new(wpe_view_backend*, GDestroyNotify, gpointer): assertion 'backend' failed

** (cog:1187): ERROR **: 00:50:32.884: Could not instantiate any WPE backend.
Trace/breakpoint trap
```
Any suggestion ?

**To Reproduce**

Fikes to reproduce the behavior:
1. local.conf
```
MACHINE ?= "phyboard-pollux-imx8mp-2"
DISTRO ?= "yogurt-vendor-xwayland"
DEPLOY_DIR = "${TOPDIR}/deploy"
BB_GIT_SHALLOW = "1"
KCONFIG_CONFIG_COMMAND = "nconfig"
KCONFIG_CONFIG_COMMAND_pn-busybox = "menuconfig"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
SYSTEMD_AUTO_ENABLE_pn-phytec-qtdemo = "disable"
DEBUG_BUILD_pn-linux-mainline = "1"
DEBUG_BUILD_pn-linux-imx = "1"
DEBUG_BUILD_pn-barebox = "1"
TOOLCHAIN_TARGET_TASK_append = " kernel-devsrc"
EXTRA_IMAGE_FEATURES = ""
EXTRA_IMAGE_FEATURES += "debug-tweaks"
CONF_VERSION = "1"
ACCEPT_FSL_EULA = "1"
BB_NUMBER_THREADS="14"
INHERIT+="toaster buildhistory"
```
3. bblayers.conf
```
POKY_BBLAYERS_CONF_VERSION = "2"
BBPATH = "${TOPDIR}"
BBFILES ?= ""
OEROOT := "/data/build/yocto/sources/poky"
BSPDIR := "${OEROOT}/../.."
BBLAYERS ?= ""
BBLAYERS += "\
  ${OEROOT}/../poky/meta \
  ${OEROOT}/../poky/meta-poky \
  ${OEROOT}/../meta-openembedded/meta-oe \
  ${OEROOT}/../meta-openembedded/meta-networking \
  ${OEROOT}/../meta-openembedded/meta-python \
  ${OEROOT}/../meta-openembedded/meta-multimedia \
  ${OEROOT}/../meta-openembedded/meta-filesystems \
  ${OEROOT}/../meta-openembedded/meta-perl \
  ${OEROOT}/../meta-openembedded/meta-gnome \
  ${OEROOT}/../meta-imx/meta-bsp \
  ${OEROOT}/../meta-imx/meta-sdk \
  ${OEROOT}/../meta-imx/meta-ml \
  ${OEROOT}/../meta-browser \
  ${OEROOT}/../meta-clang \
  ${OEROOT}/../meta-freescale \
  ${OEROOT}/../meta-freescale-3rdparty \
  ${OEROOT}/../meta-freescale-distro \
  ${OEROOT}/../meta-phytec \
  ${OEROOT}/../meta-qt5 \
  ${OEROOT}/../meta-rauc \
  ${OEROOT}/../meta-rust \
  ${OEROOT}/../meta-python2 \
  ${OEROOT}/../meta-yogurt \
  ${OEROOT}/../meta-security \
  ${OEROOT}/../meta-security/meta-tpm \
  ${OEROOT}/../meta-webkit \
  "
```

5. hash references of each repository of each used layer
```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-phytec-linux"
MACHINE              = "phyboard-pollux-imx8mp-2"
DISTRO               = "yogurt-vendor-xwayland"
DISTRO_VERSION       = "BSP-Yocto-FSL-i.MX8MP-PD21.1.3"
TUNE_FEATURES        = "aarch64 cortexa53 crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:d88d62c20d7d8da85f02edb170dae0280624ad7e"
meta-oe              
meta-networking      
meta-python          
meta-multimedia      
meta-filesystems     
meta-perl            
meta-gnome           = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-bsp             
meta-sdk             
meta-ml              = "HEAD:98f80226336fe894ecac5f81e4b40a40f442bbd4"
meta-browser         = "HEAD:ee3be3b5986a4aa0e73df2204a625ae1fe5df37e"
meta-clang           = "HEAD:711e593d5984aad3bf35c51b7ac4482982bc16c7"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:dbcc686f52c3c84db8cb86aa8973a4e373651b98"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-phytec          = "HEAD:a76dda7b5757d116abb9ca127e79e849164008f4"
meta-qt5             = "HEAD:21ce4c124d9a972d9122f87c64ac2773bf04c284"
meta-rauc            = "HEAD:cc7d622b2d49c8c7d426bb9965ad1d58806ef52f"
meta-rust            = "HEAD:53bfa324891966a2daf5d36dc13d4a43725aebed"
meta-python2         = "HEAD:4400f9155ec193d028208cf0c66aeed2ba2b00ab"
meta-yogurt          = "HEAD:8cd4c058eecceda49956282ee29f44f5a9daeffe"
meta-security        
meta-tpm             = "HEAD:52e83e6b7e57bf352fd6770466996530dece3da5"
meta-webkit          = "zeus:ab4c99d825ed99901e7bd86870312f6c70865f2e"
```
7. local changes applied in the build environment
``` no change except my image-desktop.bb in recipes-images/images/ ```
9. bitbake command causing the error
``` bitbake  image-desktop ```

11. Error output
``` cog don't work with or without -P fdo ```

**Expected behavior**
cog to  work on weston

**Environment (please complete the following information):**
 - yocto release: [ zeus ]
 - hardware targe: [phyBOARD-i.MX 8MPlus Pollux](https://www.phytec.fr/produits/solutions-oem/sbcs/phyboard-imx-8mplus-pollux)]

**Additional context**
You can recover the project build with this script : 
``` wget https://download.phytec.de/Software/Linux/Yocto/Tools/phyLinux ```
More info to gather the project : https://www.phytec.de/cdocuments/?doc=wYBVEw  - see section Building the BSP

psaavedra, 2022-08-16:
Could I suggest to use the [dunfell branch (what is still compatible with zeus)](https://github.com/Igalia/meta-webkit/blob/dunfell/conf/layer.conf#L19)?

psaavedra, 2022-08-16:
Dont  link libWPEBackend-fdo-1.0.so.1.6.1 as  libWPEBackend-default.so and , instead of that, try to execute cog as: `cog -P wl https://www.google.com/`  (`or cog -P fdo ...` in case of a older version of Cog.)

gnulux, 2022-08-16:
@psaavedra , 
When i remove the link with libWPEBackend-default.so  and execute ```cog -P wl https://www.google.com``` i have the result below :
```
** (cog:833): WARNING **: 14:39:53.336: Could not load: libcogplatform-wl.so (possible cause: Resource temporarily unavailable).

wpe: could not load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory
```

with ``` cog -P fdo https://www.google.com``` 
```
**
Cog-FDO:ERROR:/usr/src/debug/cog/0.7.1-r0/cog-0.7.1/platform/cog-platform-fdo.c:1590:on_export_fdo_egl_image: assertion failed: (s_eglCreateWaylandBufferFromImageWL)
```

psaavedra, 2022-08-17:
cog-0.7.1 is quite a old vCog version. I could suggest to use the Dunfell branch that I've suggested.

 Regarding the assertion failed: (s_eglCreateWaylandBufferFromImageWL)  ... it looks like Cog is not able to find the required Wayland extensons. Can you put the weston.log output here?

gnulux, 2022-08-17:
I'm currently building it with Dunfell branch. 
Keep you in touch in few minutes.

gnulux, 2022-08-17:
Ok , i confirm it works fine with dunfell branch . many thk for your support.

closed 2022-08-17 by gnulux

psaavedra, 2022-08-17:
> Ok , i confirm it works fine with dunfell branch . many thks for your support.

Nice. You are welcome.