Igalia/meta-webkit #388: not able to load url / website which uses WebSockets

vijaynkamble86, 2022-08-16:
Hello,

I have ported wpewebkit on QCOM SoC which uses Yocto Dunfell. I have used  wpewebkit 2.20_2 and cog 0.1.0 versions. I am able to compile wpewebkit(recipes-browser) successfully. And able to launch URL / website with cog command.

But when I launch a particular website which uses websocket, white page / canvas gets displayed. Basically i am using websocket in for reading the data dynamically and plotting it as graphs.

Could you please let me know how shall i use URL / websites which uses websocket..? Do I need to compile additional package..?

Appreciate your inputs.

Cheers,
-Vijay N K

*

psaavedra, 2022-08-16:
First of all I'd suggest to you to use the main(kirkstone) branch (ref: 211482e97530a2e3e632f96526d01e1b7b5b0f39) what is still compatible with Dunfell (https://github.com/Igalia/meta-webkit/blob/main/conf/layer.conf#L20). This will allow to you to use a newer Cog/WebKit WPE convination.

For debugging your webpage, I can also suggest to you to define this envvar: `export WEBKIT_INSPECTOR_SERVER=0.0.0.0:12321` and remotely connect to the browser using a Epiphany browser and the URL `inspector://<address of your board>:12321`.  This will allow to you to see if there is some error being logged in the console.

psaavedra, 2022-08-17:
Good to know that works for you. About flickering, as I said, tey to use the Remote Inspector and use the profiler to identify where the webpage is spending time.

I could say you will have to optimize the webpage for embedded devices.

vijaynkamble86, 2022-08-17:
But i see the flicker with google.com, bing.com as well. Its not specific to my website. 

Just wanted to know if this is coming from wpewebkit..?

psaavedra, 2022-08-17:
nom that is not the expected behavior. What backend (drm / wl)  are you using?

vijaynkamble86, 2022-08-17:
I am using wayland backend.

vijaynkamble86, 2022-08-17:
Apologies, but I don't see the flicker issue with other browsers like Chromium.

psaavedra, 2022-08-17:
Some questions and ideas that comes to my mind to narrow down the issue:

* What versions of wpewebkit, cog, libwpe and wpebackend-fdo are you using right now?
* Could you show me how are you launching the browser?
* Could you check the CPU usage during the browser execution?
* Could you launch cog using this `export WEBKIT_SHOW_FPS=1` envvar?

vijaynkamble86, 2022-08-18:
Hello Pablo,

What versions of wpewebkit, cog, libwpe and wpebackend-fdo are you using right now?
>> wpewebkit_2.36.4, cog_0.12.4, libwpe_1.12.0 wpebackend-rdk_1.20200213 [not using wpebackend-fdo]

Could you show me how are you launching the browser?
>> cog https://www.google.com

Could you check the CPU usage during the browser execution?
>> sure, Let me get the data

Could you launch cog using this export WEBKIT_SHOW_FPS=1 envvar?
>> can i pass the the above variable to cog command..?

Cheers,
-Vijay N. K.

psaavedra, 2022-08-18:
> Hello Pablo,
> 
> What versions of wpewebkit, cog, libwpe and wpebackend-fdo are you using right now?
> 
> > > wpewebkit_2.36.4, cog_0.12.4, libwpe_1.12.0 wpebackend-rdk_1.20200213 [not using wpebackend-fdo]
> 

wpebackend-rdk is deprecated. Actually we are considering remove it soon. 

I strongly suggest to use the wpebackend-fdo. 1) with `drm` platform if you are not interested on Wayland 2)  with `wl` platform if your goal is to use it into a Wayland environment.

> Could you show me how are you launching the browser?
> 
> > > cog https://www.google.com
> 

After activate the wpebackend-fdo in your build. Try to run the launcher as follows:

1) `cog -P drm https://www.google.com`, for DRM
1) `cog -P wl https://www.google.com`, for Wayland.

> Could you check the CPU usage during the browser execution?
> 
> > > cog is just consuming 1.8% cpu at max.
> 
> Could you launch cog using this export WEBKIT_SHOW_FPS=1 envvar?
> 
> > > After setting the envar, I see number getting displayed on top left corner of browser. And those numbers are varying from 2 to 10. And settles to 7/8/9
> 

these are the generated frames per second. Try to get the number using this URL https://webkit.org/blog-files/3d-transforms/poster-circle.html

> Cheers, -Vijay N. K.

psaavedra, 2022-08-18:
Related documentation: https://github.com/Igalia/meta-webkit/wiki/WPE#building-an-image-with-wpe

vijaynkamble86, 2022-08-22:
Hello Pablo,

I tried getting the FPS with this URL https://webkit.org/blog-files/3d-transforms/poster-circle.html
>> the FPS displayed is 59/60. And surprisingly I don't see any flicker with this. 

I tried launching the URL with -P option but looks like its not supported - 

=================================================
/ # cog -P wl http://www.google.com

(cog:1485): Cog-Core-WARNING **: 08:49:35.237: cog_modules_get_preferred: cannot find module 'wl'

(cog:1485): Cog-Core-WARNING **: 08:49:35.238: cog_modules_get_preferred: preferred module 'wl' not supported

(cog:1485): Cog-WARNING **: 08:49:35.238: Cannot create platform: Could not find an usable platform module
Segmentation fault (core dumped)
================================================

I am trying with wpebackend-fdo, build is in progress, will load and update.

Cheers,
-Vijay Navnath Kamble

vijaynkamble86, 2022-08-22:
is the flicker because of slower / small FPS..?

psaavedra, 2022-08-22:
> Hello Pablo,
>
> I tried getting the FPS with this URL https://webkit.org/blog-files/3d-transforms/poster-circle.html
>
>
> I tried launching the URL with -P option but looks like its not supported - [...]

Ensure WL and DRM modules are installed in the system: 

```
# ls /usr/lib/cog/modules/               
libcogplatform-drm.so  libcogplatform-wl.so
```

and then try with this:

```
export $(cat /proc/$(pidof weston-keyboard)/environ  | strings | grep XDG_RUNTIME_DIR)
export $(cat /proc/$(pidof weston-keyboard)/environ  | strings | grep WAYLAND_DISPLAY)
cog -P wl http://www.google.com
```

psaavedra, 2022-08-22:
> Hello Pablo,
> 
> I tried getting the FPS with this URL https://webkit.org/blog-files/3d-transforms/poster-circle.html
> 
> > > the FPS displayed is 59/60. And surprisingly I don't see any flicker with this.

Looks fine, 60 frames per second is the expected

vijaynkamble86, 2022-08-22:
> 

with https://www.google.com the FPS is 1. And I see Google Chrome Flickering.

psaavedra, 2022-08-22:
1 is find if the web content is nit update for a change (like a animation).
As I said before, the RDK backend is deprecated I can't guarantee anything.

The RDK backend paints directly in the framebuffer? Don you have some other
running program painting in the screen? (like Weston...)

vijaynkamble86, 2022-08-22:
Hello Pablo,

I have updated to wpebackend-fdo and the build is in progress. Will check the issue with wpebackedn-fdo.

Cheers,
-Vijay Navnath Kamble

vijaynkamble86, 2022-09-08:
Hello Pablo,

apologies for later response. 

I ported the new recipe of meta-webkit and used the wpebackend as wpebackend-fdo. I did not see the flicker issue on couple of devices. 

Running the test on multiple set ups to confirm thats its working fine.

Thank you so much for you help and guidance to debug the issue.

Cheers,
-Vijay Navnath Kamble

vijaynkamble86, 2022-09-08:
Hello Pablo,

I have one query - 

"Does cog browser on embedded devices support WEB RTC Streaming..?"

Cheers,
-Vijay Navnath Kamble

psaavedra, 2022-09-08:
Replied on https://github.com/Igalia/meta-webkit/issues/401

> Hello Pablo,
> 
> I have one query -
> 
> "Does cog browser on embedded devices support WEB RTC Streaming..?"
> 
> Cheers, -Vijay Navnath Kamble

closed 2022-09-08 by psaavedra