Igalia/meta-webkit #391: *   - nothing provides libwpebackend-fdo-1.0-1 >= 1.12.0 needed by cog-0.12.3-r0.aarch64-mx8mp

Oquirella, 2022-08-30:
**Describe the bug**
When building the rootfs i get the above error

**To Reproduce**

Fikes to reproduce the behavior:
1. local.conf
2. bblayers.conf
3. hash references of each repository of each used layer
4. local changes applied in the build environment
5. bitbake command causing the error
6. Error output

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - yocto release: hardknott
 - hardware targte: imx8mpevk

**Additional context**
Add any other context about the problem here.

clopez, 2022-08-30:
Likely fixed by #392

psaavedra, 2022-08-31:
Like in https://github.com/Igalia/meta-webkit/pull/392#issuecomment-1231492821 I'd suggest to you to use the main branch since this is also compatible with the `hardknott` Yocto release.

Oquirella, 2022-08-31:
I will try the main bracnh