Igalia/meta-webkit #393: Requested 'gio-2.0 >= 2.67.4' but version of GIO is 2.62.6

Oquirella, 2022-08-31:
**Describe the bug**
The main branch of meta-webkit requires glib 2.67.4 while the glib version on Dunfell is 2.62.6. The main branch should be compatible with dunfell, but that is not the case.

**Build output**
ERROR: cog-0.14.0-r0 do_configure: Execution of '/home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/temp/run.do_configure.362' failed with exit code 1
ERROR: Logfile of failure stored in: /home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/temp/log.do_configure.362
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 9.3.0
| -- Check for working C compiler: /home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
| -- Check for working C compiler: /home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Cog module path: /usr/lib/cog/modules
| -- Found PkgConfig: /home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/recipe-sysroot-native/usr/bin/pkg-config (found version "0.29.2")
| -- Checking for module 'gio-2.0>=2.67.4'
| --   Requested 'gio-2.0 >= 2.67.4' but version of GIO is 2.62.6
| CMake Error at /home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/recipe-sysroot-native/usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:467 (message):
|   A required package was not found
| Call Stack (most recent call first):
|   /home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/recipe-sysroot-native/usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:647 (_pkg_check_modules_internal)
|   CMakeLists.txt:155 (pkg_check_modules)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/omar/acos/build/tmp/work/aarch64-mx8mp-poky-linux/cog/0.14.0-r0/temp/run.do_configure.362' failed with exit code 1

**Environment :**
 - yocto release: dunfell
 - hardware target: imx8mpevk

philn, 2022-09-01:
Have you tried the dunfell branch? https://github.com/Igalia/meta-webkit/tree/dunfell

sindelar-fr, 2022-09-01:
As somewhat of a workaround you can use cog 0.12.4 that is still present in the main branch by adding something like
`PREFERRED_VERSION_cog = "0.12.%"`
in your local.conf.

Oquirella, 2022-09-01:
Cog 0.12.4 + imx-gpu-viv_6.4.3.p1.0-aarch64 works perfect on IMX8MP. All versions below these cause the applications to crash.

sindelar-fr, 2022-10-21:
I've just tried and it seems that cog 0.14.1 now builds on dunfell.

closed 2023-01-28 by github-actions[bot]