Igalia/meta-webkit #395: Unable to play videos using cog and Gstreamer

jm-costa, 2022-09-02:
**Describe the bug**

I'm currently unable to play videos with cog, even though the gst-play-1.0 command works fine.

**To Reproduce**

```sh
#gst-play-1.0 --videosink glimagesink https://test-videos.co.uk/vids/bigbuckbunny/mp4/h264/360/Big_Buck_Bunny_360_10s_1MB.mp4
export GST_DEBUG="3,webkit*:6"
export GST_DEBUG_FILE=/tmp/gst.log
export GST_DEBUG_DUMP_DOT_DIR=/tmp
cog -P wl https://test-videos.co.uk/vids/bigbuckbunny/mp4/h264/360/Big_Buck_Bunny_360_10s_1MB.mp4
```

The installed packages:
```
IMAGE_INSTALL += "\
    cute-files \
    ca-certificates \
    cog \
    glib-networking \
    gstreamer1.0 \
    gstreamer1.0-python \
    gstreamer1.0-plugins-imx \
    gstreamer1.0-plugins-imx-meta \
    gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-base-apps \
    gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad \
    gstreamer1.0-meta-base \
    packagegroup-fsl-gstreamer1.0 \
    gstreamer1.0-vaapi \
    gstreamer1.0-rtsp-server \
    ttf-bitstream-vera \
    imx-codec \
    imx-vpu-hantro \
"
```
The versions:
```
gstreamer1.0:
  meta-freescale       1.18.0.imx
gstreamer1.0-meta-base:
  meta                 1.0
gstreamer1.0-plugins-bad:
  meta-freescale       1.18.0.imx
gstreamer1.0-plugins-base:
  meta-freescale       1.18.0.imx
gstreamer1.0-plugins-good:
  meta-freescale       1.18.0.imx
gstreamer1.0-plugins-imx:
  meta-freescale       2.0.0+gitAUTOINC+11e3a555a2
gstreamer1.0-python:
  meta                 1.18.6
gstreamer1.0-rtsp-server:
  meta-freescale       1.18.0
gstreamer1.0-vaapi:
  meta                 1.18.6
packagegroup-fsl-gstreamer1.0:
  meta-freescale-distro 1.0
imx-codec:
  meta-bsp             4.6.3
imx-vpu-hantro:
  meta-bsp             1.24.0
```

Also:
```diff
diff --git a/meta-bsp/recipes-multimedia/imx-codec/imx-codec.inc b/meta-bsp/recipes-multimedia/imx-codec/imx-codec.inc
index ec39530e5..6beb0b200 100644
--- a/meta-bsp/recipes-multimedia/imx-codec/imx-codec.inc
+++ b/meta-bsp/recipes-multimedia/imx-codec/imx-codec.inc
@@ -17,7 +17,7 @@ SRC_URI = "${FSL_MIRROR}/${BP}.bin;fsl-eula=true"
 EXTRA_OECONF = "${@bb.utils.contains('TUNE_FEATURES', 'aarch64', '--enable-armv8', \
                    bb.utils.contains('TUNE_FEATURES', 'callconvention-hard', '--enable-fhw', '', d), d)}"
 
-PACKAGECONFIG ?= ""
+PACKAGECONFIG ?= "vpu"
```

`journalctl`'s output:
```
Sep 02 09:25:29 sm2s-imx8mp runc[2202]: (cog:9): GLib-GIO-WARNING **: 09:25:29.935: Your application 
does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  
It should do one of these.
Sep 02 09:25:30 sm2s-imx8mp runc[2202]: Cog-Core-Message: 09:25:30.180: <https://test-videos.co.uk/vi
ds/bigbuckbunny/mp4/h264/360/Big_Buck_Bunny_360_10s_1MB.mp4> Load started.
Sep 02 09:25:30 sm2s-imx8mp runc[2202]: Cog-Core-Message: 09:25:30.348: <https://test-videos.co.uk/vi
ds/bigbuckbunny/mp4/h264/360/Big_Buck_Bunny_360_10s_1MB.mp4> Loading...
Sep 02 09:25:30 sm2s-imx8mp runc[2202]: WebKit wasn't able to find a WebVTT encoder. Subtitles handli
ng will be degraded unless gst-plugins-bad is installed.
Sep 02 09:25:30 sm2s-imx8mp runc[2202]: Cog-Core-Message: 09:25:30.539: <https://test-videos.co.uk/vi
ds/bigbuckbunny/mp4/h264/360/Big_Buck_Bunny_360_10s_1MB.mp4> Loaded successfully.
```

**Environment (please complete the following information):**
 - yocto release: hardknott-3.3.6
 - hardware targe: iMX8M+ (aarch64)
[bugreport.tar.gz](https://github.com/Igalia/meta-webkit/files/9477027/bugreport.tar.gz)

[image]

Thanks for the support!

Best regards,
João

Oquirella, 2022-09-02:
Which version of cog are you using? I am also use an IMX8MP, only 0.12.4 or above worked for me.

jm-costa, 2022-09-02:
Oh, I forgot to mention this quite important detail...
```
cog:
  meta-webkit          0.12.3
```
I'm gonna try a newer version. Thanks!

jm-costa, 2022-09-02:
The issue is still happening, actually.
I tried version 0.12.4 of cog, and still:
```
Sep 02 14:32:37 sm2s-imx8mp runc[1043]: 0.12.4 (WPE WebKit 2.36.3)                                   
Sep 02 14:32:37 sm2s-imx8mp runc[1043]: (cog:10): GLib-GIO-WARNING **: 14:32:37.104: Your application
 does not implement g_application_activate() and has no handlers connected to the 'activate' signal. 
 It should do one of these.                                                                          
Sep 02 14:32:37 sm2s-imx8mp runc[1043]: Cog-Core-Message: 14:32:37.369: <http://commondatastorage.goo
gleapis.com/gtv-videos-bucket/sample/ElephantsDream.mp4> Load started.                               
Sep 02 14:32:37 sm2s-imx8mp runc[1043]: Cog-Core-Message: 14:32:37.664: <http://commondatastorage.goo
gleapis.com/gtv-videos-bucket/sample/ElephantsDream.mp4> Loading...                                  
Sep 02 14:32:37 sm2s-imx8mp runc[1043]: WebKit wasn't able to find a WebVTT encoder. Subtitles handli
ng will be degraded unless gst-plugins-bad is installed.                                             
Sep 02 14:32:38 sm2s-imx8mp runc[1043]: Cog-Core-Message: 14:32:38.742: <http://commondatastorage.goo
gleapis.com/gtv-videos-bucket/sample/ElephantsDream.mp4> Loaded successfully. 
```
I got the same black screen as in the screenshot.

philn, 2022-09-02:
Thanks for the log, is your `/var/tmp/` writable? If you check the gst.log, the end of the file gives some hints that it's not writable.

philn, 2022-09-02:
Or not even existing. Looks like an OS setup issue.
Also please check that `gst-play-1.0 --videosink glimagesink /path/to/video.mp4` works.

jm-costa, 2022-09-05:
Thanks for pointing that out! Indeed, `/var/tmp` wasn't there. Now, it works just fine, even with v0.12.3.

closed 2022-09-05 by jm-costa