Igalia/meta-webkit #396: 100% CPU usage on core 1 (out of 4) on IMX8MP

Oquirella, 2022-09-06:
I am running cog version 0.12.4 and it looks like the load is not shared over the multiple cores. Any idea how i can fix this? /usr/libexec/wpe-webkit-1.0/WPEWebProcess is causing this.

I am running Yocto (Dunfell) on the IMX8MP using:

* Weston 8.0.0
* Wayland 1.18.0
* imx-gpu-viv 6.4.0.p2.4

psaavedra, 2022-09-08:
The WPEWebProcess process has several threads executing different logics so yes the WPEWebProcess is multi-thread. About the hardware acceleration yes, WPE has it always enabled. If Cog is displaying web content for you, then WPE is using the GPU.

Oquirella, 2022-09-09:
Pages with SVG graphs are really slow, i don't think that they are accelerated using the GPU. I got 0 FPS on them.  Any idea how i can get it to be performant?

closed 2022-12-17 by github-actions[bot]