Igalia/meta-webkit #398: Unable to play videos in a centered position

jm-costa, 2022-09-06:
Hello,

I'm currently using cog v.0.12.4 and the exact same Yocto setup as in https://github.com/Igalia/meta-webkit/issues/395

I run the following commands to open cog and play a video:

```
export COG_PLATFORM_FDO_VIEW_FULLSCREEN=1
cog -P wl http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/ElephantsDream.mp4 &
```

However, it always spawns in a different position. Is there any way to ensure the video widget is always well-centered?
If not, can cog play the video, by default, in full screen mode? Without the need to click into the full screen button.

I'm attaching two printscreens hoping it will be helpful to clarify my current situation. Thanks for the support!

[image]
[image]

philn, 2022-09-06:
Maybe you can do this with a web-extension, that when loaded, looks for the `<video>` element in the page and switches it to fullscreen using JS. Cog can load web-extensions.. I don't think there's anything else we can do...

psaavedra, 2022-09-08:
@jm-costa  please use COG_PLATFORM_**WL**_VIEW_FULLSCREEN instead of COG_PLATFORM_**FDO**_VIEW_FULLSCREEN

jm-costa, 2022-09-21:
Thanks for the suggestion! It works :)

closed 2022-09-21 by philn