Igalia/meta-webkit #400: Server video fileplay in COG downloads video locally on everytime at /var/tmp/XXX - imx8mp

Rutvij-dev, 2022-09-07:
Hi,
I am running COG with below environment setup,

**Hardware**
Toradex verdin - imx8mp 

**Software**
BSP - 5.4.161 (dunfell)
COG - v0.12.3 
WPEwebkit - 2.34.6

I have created one ngnix server on **x86** machine, wrote index.html (attached index.html.txt - [index.html.txt](https://github.com/Igalia/meta-webkit/files/9507624/index.html.txt)) and kept H264 video in the same directory.

On target hardware when I run `cog <http://ip of server>` , it plays video from server with below observations.

**Observation**

- Video on target hardware plays with the help of GStreamer plugin `downloadbuffer`.

- It to cache video from server to `/var/tmp/XXX` directory.

- Now After stopping and starting it again this caching of video happens again (same video, no restart of hardware or server)

Also I observed that in this particulate time of caching ,
user can see high pick in CPU % utilisation in top command for `WPENetworkProce` process until full cached download, this be the separate utilization apart from multimedia processing.

Is there anything I can do to avoid multiple time caching for the same setup on multiple run ?

Appreciate your guidance.

--
Thanks

philn, 2022-09-07:
I suspect that, while not optimal, this is might be the expected behaviour... So you reload the page and download starts over, right?

One workaround, in case download buffering is not so useful for you, is to disable it... `<video preload=none ....>`.

Rutvij-dev, 2022-09-08:
Hi @philn,
Thank you for an option , 

The problem i am facing is in both the case restart and reload the page.

Also I tried this in `index.html` as 

`<body>
    <video **preload=none** muted autoplay controls>
    <source src="BBB-720p-30fps.mp4" type="video/mp4" **preload="none"**>
`
However No difference , the caching still happens every time in restart and in re-load of page in cog.

philn, 2022-09-08:
Can you set this env var and try again? `WPE_SHELL_DISABLE_MEDIA_DISK_CACHE=1` ?

Rutvij-dev, 2022-09-09:
Still no luck , `WPENetworkProce` does the downloading again.

philn, 2022-09-09:
And tmp files are still written on disk or not?

closed 2022-12-17 by github-actions[bot]