Igalia/meta-webkit #402: Cog: How to check if the GPU is really being used for hardware acceleration?

Oquirella, 2022-09-08:
I get really poor performance on pages where svg graphs are being used. (on IMX8MP with vivante graphics driver).

nikolaszimmermann, 2022-09-09:
> I have added the option to wpewebkit.inc:
>
> PACKAGECONFIG[svg] = "-DENABLE_LAYER_BASED_SVG_ENGINE=ON, -DENABLE_LAYER_BASED_SVG_ENGINE=OFF,"
>
> I will start the build now, if it is performant, we should enable it as default.

I would call the option „lbse“.

Oquirella, 2022-09-09:
I enable the option, but i didn't see any performance improvements when showing my SVG graphs @nikolaszimmermann. The library i am using is: https://github.com/amcharts/amcharts4

Oquirella, 2022-09-12:
That's too bad @nikolaszimmermann , i really wanted to test this feature. Is there no nightly or experimental branch where this feature is fully enabled?

nikolaszimmermann, 2022-09-13:
No not yet, we're working on it to make it happen soon.

Oquirella, 2022-09-21:
The performance on the svg graphs in chromium is so much better, with cog it is unusable...

nikolaszimmermann, 2022-09-21:
We're aware of the current limitations and actively work on improvments. If you want want to assist and getting your specific case to work faster, a striped down minimal reduced version of your web page, which exhibits the performance problems, would be helpful.

Oquirella, 2022-09-21:
My website uses something like this but moving: https://www.amcharts.com/demos-v4/stacked-area-v4/

closed 2023-01-03 by github-actions[bot]