Igalia/meta-webkit #407: Any way to build meta-webkit for Yocto Kirkstone ?

Rutvij-dev, 2022-09-15:
Hi,
Currently I am able to build meta-webkit for `dunfell`, however need arrived to switch to `krikstone`.

So any port available ?
Any advice to achieve this ?

Appreciate any help.

--
Thanks

marvinfrankeengram, 2022-09-16:
Although there is no kirkstone branch (yet?), using the main branch of this layer works for me. Note that they are still working on it and changes might break things.
In the honister branch, they claim to be compatible with kirkstone, haven't testet that though. But I guess, changes there are probably more unlikely to break things, so it may be worth a try.

aahmed7, 2022-09-16:
The master branch has the compat for kirkstone in conf/layer.conf, so it should build

Rutvij-dev, 2022-09-19:
Yes it did work , Thanks for your inputs

closed 2022-09-19 by Rutvij-dev