Igalia/meta-webkit #416: Ability to easily build ToT wpewebkit

guijemont, 2022-10-17:
**Is your feature request related to a problem? Please describe.**

ToT WPE WebKit requires the unifdef tool at build time (in the case of yocto, that is provided by `unifdef-native`). We don't have a recipe that does that.

**Describe the solution you'd like**

Some way to easily chose in my local.conf to build the latest upstream (or something close, using a local checkout), and to have new dependencies taken into account for me.

**Describe alternatives you've considered**

Currently I've added this to my local.conf:

`DEPENDS:append:pn-wpewebkit = " unifdef-native"`

It works, but I think a better solution would be to be able to express in local.conf that I'm building recent upstream, either through setting a version or depending on a variant of the package (e.g. `wpewebkit-devupstream`), and that in that case the `unifdef-native` dependency is added, so that other webkit developers could easily use meta-webkit to do cross-compiled upstream development tasks.

**Additional context**

My understanding so far from the documentation is that thanks to what we have in devupstream.inc, we should already be able to build `wpewebkit-devupstream`, and that the wpewebkit recipe could have something like `DEPENDS:class-devupstream:append = " unifdef-native"`, but the package `wpewebkit-devupstream` is not automatically made available (`Nothing PROVIDES wpewebkit-devupstream` error)

psaavedra, 2022-10-19:
We could add something like this in the `wpewebkit_2.38.bb` recipe 

```
# unifdef-native: Needed since >2.38.
DEPENDS:append = " unifdef-native"
```

guijemont, 2022-10-19:
> We could add something like this in the `wpewebkit_2.38.bb` recipe
> 
> ```
> # unifdef-native: Needed since >2.38.
> DEPENDS:append = " unifdef-native"
> ```

That would work, though that would mean that we're compiling an unneeded extra dependency when compiling 2.38.

I was wondering if the good solution wouldn't be to just have a separate `wpewebkit_git.bb` recipe or something like that.

psaavedra, 2022-10-22:
> That would work, though that would mean that we're compiling an unneeded extra dependency when compiling 2.38.
>
> I was wondering if the good solution wouldn't be to just have a separate [...]

What do you thing about this solution? -> https://github.com/Igalia/meta-webkit/pull/423/files

guijemont, 2022-10-22:
> What do you thing about this solution? -> https://github.com/Igalia/meta-webkit/pull/423/files

Fine if it works. I still wonder why the `CLASSOVERRIDE` modification [here](https://git.yoctoproject.org/poky/tree/meta/classes/devupstream.bbclass?h=kirkstone) doesn't do that. Looking at it now, I wonder if it shouldn't be adding `class-devupstream` to  `CLASSOVERRIDE` instead of `:class-devupstream`.

guijemont, 2022-10-28:
It looks like ToT WebKit also needs libavif now, unless we deactivate it in the recipe.

guijemont, 2022-11-01:
BTW, I think we'd also want to remove webgl2 from `PACKAGECONFIG` in wpewebkit.inc, and instead have something like `PACKAGECONFIG:class-devupstream:append = " webgl2", probably in the .bb instead of the inc. Rationale:
 - in the future in wpewebkit_2.40.0.bb, we'd want to append webgl2 to PACKAGECONFIG always, not just in devupstream
 -  we do *not* want to have webgl2 in 2.38.1 as it seems to break compilation; also might not make a lot of sense to include it in 2.38.0 (unsure if it's working well or at all before ANGLE)

PR #423 (2022-11-15): wpewebkit: Add unifdef-native dependency for versions >2.38
* wpewebkit: Add unifdef-native dependency for versions >2.38
* devupstream: Honors DEPENDS:class-devupstream

Fixes: #416
closed 2022-11-15 by psaavedra