Igalia/meta-webkit #421: Cog recipe doesn't prove option to use X11?

Oquirella, 2022-10-19:
Cog supports X11, but in the recipe no option is provided to use it.

hiagofranco, 2022-10-20:
I'm currently facing an issue with Wayland that doesn't happen with X11. I would like to use it with X11 as well, but it doesn't seem to be easy to change the recipe to use X11.

marvinfrankeengram, 2022-10-26:
It least it compiles when I add the following lines to the recipe using a cog_%.bbappend:

```
PACKAGECONFIG[x11] = "-DCOG_PLATFORM_X11=ON,-DCOG_PLATFORM_X11=ofF,"
PACKAGECONFIG:append = "${@bb.utils.filter('DISTRO_FEATURES', 'x11', d)}"
```

But if I try to use it on my RPI 400 using the [raspberrypi4](http://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/conf/machine/raspberrypi4.conf?h=master) Machine and the [core-image-x11](https://layers.openembedded.org/layerindex/recipe/351/) Image, I only get a Segmentation fault. But maybe it works on a different Machine?

Oquirella, 2022-10-26:
@marvinfrankeengram I also did that and i also get a segmentation fault on Intel.

psaavedra, 2022-10-27:
Did you generate a backtrace for the crash?

closed 2023-02-02 by github-actions[bot]