Igalia/meta-webkit #422: webgl not working on raspberrypi3 with ANGLE

guijemont, 2022-10-19:
**Describe the bug**

When compiling the wpewebkit with recent upstream WebKit, the ANGLE backend for webgl (or webgl2) is not working on raspberry pi 3, on kirkstone. Since [254947@main](https://commits.webkit.org/254947@main) ANGLE is the only way to have webgl in WebKit.

When running `XDG_RUNTIME_DIR=/tmp cog -P drm http://get.webgl.org/` with the generated image, the spinning cube does not appear. The output of cog is the following:

```
(cog:615): Cog-DRM-WARNING **: 13:35:15.961: Renderer 'modeset' does not support rotation 0 (0 degrees).

(cog:615): GLib-GIO-WARNING **: 13:35:16.089: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
Cog-Core-Message: 13:35:17.230: <http://get.webgl.org/> Load started.
Cog-Core-Message: 13:35:17.252: <http://get.webgl.org/> Loading...
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
ERR: Display.cpp:997 (initialize): ANGLE Display::initialize error 12289: Failed to initialize system egl
Cog-Core-Message: 13:35:17.663: <http://get.webgl.org/> Loaded successfully.
```

This also seems to reproduce with older revisions since ANGLE was made the default, unless compiling with `-DUSE_ANGLE_WEBGL=OFF`, which is done in the attached [local.conf](https://github.com/Igalia/meta-webkit/files/9820832/local.conf.txt).

**To Reproduce**

Files to reproduce the behavior:
1. [local.conf](https://github.com/Igalia/meta-webkit/files/9820832/local.conf.txt)
2. [bblayers.conf](https://github.com/Igalia/meta-webkit/files/9820831/bblayers.conf.txt)
3. hash references of each repository of each used layer:
    - poky: `453be4d258f71855205f45599eea04589eb4a369`
    - meta-openembedded: `0b78362654262145415df8211052442823b9ec9b`
    - meta-raspberrypi: `dacad9302a92b0b7edf8546cdcad1f8ef753e462`
    - meta-webkit: `3b2f69f7cb057eaf9091afb21153533bcfcc8251`
5. local changes applied in the build environment
    - `devtool modify -n wpewebkit /path/to/webkit/source` to easily test different revisions
7. bitbake command causing the error
8. - `bitbake core-image-base` or `bitbake wpewebkit`
9. Error output

See cog error output in bug description.

**Expected behavior**

Spinning wireframe cube when rendering http://get.webgl.org/.

**Screenshots**

This is what http://get.webgl.org/ looks like when the bug happens:

[image]

**Environment (please complete the following information):**
 - yocto release: kirkstone
 - hardware targe: rpi3 32 bits

**Additional context**

I only tested with drm, don't know if it works under wayland.

manuelafm, 2023-01-18:
In principle it was working fine in Wayland for a long time (at least ~6 months), next time that I fire it up I'll try to see if this happens with DRM.

guijemont, 2023-04-19:
I believe this was fixed in WebKit by [r258932](https://commits.webkit.org/258932@main). Sorry I forgot to update this issue.

closed 2023-04-19 by guijemont