Igalia/meta-webkit #427: http is good, https fails to render

flatmax, 2022-11-17:
**Describe the bug**
If I go to an http site, the render is good. If I go to an https site, the site doesn't render. I have glib networking installed. Here is the output :

```
# cog http://www.github.com
xkbcommon: ERROR: couldn't find a Compose file for locale "C" (mapped to "C")

(cog:720): GLib-GIO-WARNING **: 02:00:37.090: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
Cog-Core-Message: 02:00:38.858: <http://www.github.com/> Load started.
Cog-Core-Message: 02:00:38.925: <https://www.github.com/> Redirected.

(cog:720): Cog-Core-WARNING **: 02:00:38.974: <https://www.github.com/> Page load error: WebKit encountered an internal error
Cog-Core-Message: 02:00:38.974: <> Loaded successfully.
Cog-Core-Message: 02:00:39.058: <https://www.github.com/> Load started.
Cog-Core-Message: 02:00:39.060: <https://www.github.com/> Loading...
Cog-Core-Message: 02:00:39.071: <https://www.github.com/> Loaded successfully.

```

**To Reproduce**

Using buildroot on a Radxa rock 3a platforrm. With the following settings :

BR2_PACKAGE_GLIB_NETWORKING=y
BR2_PACKAGE_SHARED_MIME_INFO=y
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_PACKAGE_WPEWEBKIT=y
BR2_PACKAGE_WPEWEBKIT_SANDBOX=y
BR2_PACKAGE_COG=y
BR2_PACKAGE_MESA3D=y
BR2_PACKAGE_MESA3D_GALLIUM_DRIVER_SWRAST=y
BR2_PACKAGE_MESA3D_OPENGL_ES=y
BR2_PACKAGE_MESA3D_OPENGL_EGL=y
BR2_PACKAGE_WESTON=y
BR2_PACKAGE_WESTON_DEFAULT_DRM=y
BR2_PACKAGE_WESTON_WAYLAND=y

6. Error output

See above.

**Environment (please complete the following information):**
 - buildroot [buildroot.rockchip](https://github.com/flatmax/buildroot.rockchip)
 - Radxa rock 3a

flatmax, 2022-11-17:
buildroot version is ec9519831ed5d5698c787b48d2ab952525cacb93

psaavedra, 2022-11-17:
Looks like the issue is not for meta-webkit (Yocto) but for buildroot. Anyway, general speaking, it seems with the time of the system. I'd suggest, to check `datetime` on the system. Probably the date too much on the past or in the future.

flatmax, 2022-11-17:
oops apologies !

closed 2022-11-17 by flatmax