Igalia/meta-webkit #428: Cog with drm frame buffer on Intel not working

gustavqmatic, 2022-11-17:
**Describe the issue**
We build cog with framebuffer on Intel. We use master on meta-webkit and kirkstone on meta-intel etc.

The build compile and run but nothing is seen on the screen. I can see that drm is set in package config using "bitbake -e cog". Other egl applications such as kmscube and glmark works as they should.

I have tried to debug the issue by adding my own logging to both webkit and mesa. I can see that the drm init is done correctly, mode '1920x1080' @ 60Hz is selected. A display with gbm backend is then created and it seems to work. Then /usr/lib/dri/iris_dri.so is loaded and I select gles as renderer, it list lots of configs and selects “config #0 with format 'XR30'”. After that I can see that the eglDisplay is created again, but this time it does not seem to be correct, driver name is empty, gbm name is garbage and fd is -1. This gives error “EGL user error 0x3001 (EGL_NOT_INITIALIZED) in eglInitialize: DRI2: gbm device using incorrect/incompatible backend”.

Cog does not crash/stop when this happens, but nothing is seen on the screen. When I press Ctrl-C to stop it some additional output is written:
^ClibEGL debug: EGL user error 0x3008 (EGL_BAD_DISPLAY) in eglQueryContext
function is no-op
function is no-op
Couldn't open libGL.so.1 or libOpenGL.so.0

Any ideas what can stop it from setting up the display correctly?

**To Reproduce**

Fikes to reproduce the behavior:
1. local.conf
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
2. distro
DISTRO_FEATURES:append = "opengl"
3. Image
IMAGE_INSTALL:append (more than needed?)
wpewebkit cog \
    intel-media-driver libva mesa libva-utils igt-gpu-tools \
    libgbm libdrm \
    kmscube \
4. Starting cog
export G_MESSAGES_DEBUG=all
export LIBGL_DEBUG=verbose
export EGL_PLATFORM=drm
export EGL_LOG_LEVEL=debug
cog --platform-params="renderer=gles" -P drm https://www.google.com

**Expected behavior**
Cog UI and webpage on the screen.

**Environment (please complete the following information):**
 - yocto release: kirkstone
 - hardware targe: intel x86_64

**Additional context**
Full log inc logs added by me (starting with “Gustav”).

```
(process:3412): Cog-Core-DEBUG: 21:35:09.618: ensure_extension_points: Extension points registered.
(process:3412): Cog-Core-DEBUG: 21:35:09.619: cog_modules_add_directory: Scanning '/usr/lib/cog/modules'
Cog-INFO: 21:35:09.621: com.igalia.Cog.Platform:
Cog-INFO: 21:35:09.621:   drm - 200/CogDrmPlatform
(cog:3412): Cog-DEBUG: 21:35:09.624: platform_setup: Platform name: drm
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_config: overriding device_scale value, using 1.00 from shell
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm: enumerated device 0x562ab4297d80, available_nodes 5
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm:   DRM_NODE_PRIMARY: /dev/dri/card0
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm:   DRM_NODE_RENDER: /dev/dri/renderD128
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm: using device 0x562ab4297d80, DRM_NODE_PRIMARY /dev/dri/card0
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm: 5 connectors available
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm:  connector id 239, type 10, not connected, 0 usable modes
(cog:3412): Cog-DRM-DEBUG: 21:35:09.626: init_drm:  connector id 250, type 11, not connected, 0 usable modes
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:  connector id 256, type 11, connected, 36 usable modes
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [0]: '1920x1080', 1920x1080@60, flags 5, type 72 (preferred) 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [1]: '1920x1080', 1920x1080@60, flags 1048581, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [2]: '1920x1080', 1920x1080@60, flags 1048581, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [3]: '1920x1080', 1920x1080@50, flags 1048581, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [4]: '1680x1050', 1680x1050@60, flags 9, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [5]: '1600x900', 1600x900@60, flags 5, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [6]: '1280x1024', 1280x1024@75, flags 5, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [7]: '1280x1024', 1280x1024@60, flags 5, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [8]: '1440x900', 1440x900@60, flags 9, type 64 
(cog:3412): Cog-DRM-DEBUG: 21:35:09.662: init_drm:    [9]: '1280x800', 1280x800@60, flags 9, type 64 
[... 346 lines cut ...]
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: starting to write 9 bytes to TLS connection
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: claiming operation OP_WRITE
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: claiming operation OP_WRITE succeeded
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: yielding operation OP_WRITE
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: successfully write 9 bytes to TLS connection
(process:3421): libsoup-http2-DEBUG: 21:35:10.240: [C3-S0] [-] [SEND] [PING]
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: starting to write 17 bytes to TLS connection
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: claiming operation OP_WRITE
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: claiming operation OP_WRITE succeeded
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: yielding operation OP_WRITE
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: successfully write 17 bytes to TLS connection
(process:3421): libsoup-http2-DEBUG: 21:35:10.240: [C3-S0] [-] [SEND] [RST_STREAM] stream_id=1
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: starting to write 13 bytes to TLS connection
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: claiming operation OP_WRITE
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: claiming operation OP_WRITE succeeded
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: yielding operation OP_WRITE
(process:3421): GLib-Net-DEBUG: 21:35:10.240: CLIENT[0x55bc9a2d05b0]: successfully write 13 bytes to TLS connection
Cog-Core-Message: 21:35:10.248: <https://www.google.com/> Loaded successfully.
^ClibEGL debug: EGL user error 0x3008 (EGL_BAD_DISPLAY) in eglQueryContext

function is no-op
function is no-op
Couldn't open libGL.so.1 or libOpenGL.so.0
./ctest.sh: line 19:  3412 Aborted                 cog --platform-params="renderer=gles" -P drm https://www.google.com
```

clopez, 2022-12-27:
This error in your log seems it can be problematic:
`libEGL debug: EGL user error 0x3001 (EGL_NOT_INITIALIZED) in eglInitialize: DRI2: gbm device using incorrect/incompatible backend`

Cog works fine with DRM/KMS with other Mesa-based drivers like the Mesa VC4 driver for the RPi 3.
If you have one RPi 3 or 4 at hand, then you can try to build Cog for it and see how it works there via KMS/DRM. 

Not sure why it is not working for you on Intel, in theory it should work (AFAIK).

gustavqmatic, 2023-01-10:
Thank you for the response. We have decided to use Wayland instead of DRM, so I will close this issue.

closed 2023-01-10 by gustavqmatic