Igalia/meta-webkit #429: Network freeze observed through websocket client running on page

clorichel, 2022-11-18:
**Describe the bug**
I'm having frequent "network freezes" observed with a websocket client running on the page browsed by WPE Webkit.

Every few minutes, the websocket client running on that webpage (rendered with WPE Webkit) will not answer the ping from the websocket server. Then it will take that webpage anywhere from 30 to 90 seconds to actually reconnect to the server.

Not sure "freeze" is the appropriate word, but it looks like the webpage stopped, or at least blacked out network communications during those periods of time, then it suddenly "reappears", and then business as usual, everything is back to normal, until it eventually freezes again 🙃

Mentioning "network" because during those freezes, there are visual animations going on the webpage with no freeze at all. That webpage also outputs some music, which is continuously played during those events. That leads me to think the freeze is somehow limited to the network, but obviously I might be wrong about that assumption, it's just what I'm observing (graphics and musics are going, network is down).

**To Reproduce**
1. using the latest stable version (2.38.0) of [wpe-webkit-mir-kiosk](https://snapcraft.io/wpe-webkit-mir-kiosk) snap
2. over the latest (90-mir2.10.0) [ubuntu-frame](https://snapcraft.io/ubuntu-frame)
3. on the latest (22) [Ubuntu Core ](https://ubuntu.com/core)
4. running a [Socket.IO](http://socket.io/) server with default 5s `pingTimeout` and 25s `pingInterval`
5. running a Socket.IO client on the webpage rendered via WPE Webkit
6. running the same Socket.IO client on the same webpage but on another computer/browser

**Expected behavior**
The websocket / network connection should just be stable all the time.

**Screenshots**
Implementing a [latency check](https://socket.io/how-to/check-the-latency-of-the-connection) every 5 seconds, through a socket.emit run on the webpage, and a server callback to log the actual latency server-side:

```javascript
// on the webpage
setInterval(() => {
  const start = Date.now();

  socket.emit('latency-test', () => {
    const duration = Date.now() - start;
    socket.emit('latency-test-result', duration);
  })
}, 5000)
```
```javascript
// on the server
socket.on('latency-test', (callback) => {
  logger.debug(`Websocket client id#${socket.id} latency-test`);
  callback();
});
socket.on('latency-test-result', (duration) => {
  logger.info(`Websocket client id#${socket.id} latency-test-result is ${duration}ms`);
});
```

We can see:
- the webpage running on WPE Webkit through the websocket client `id#btJn7HWoV2iHofhRAAAA`, its latency is pretty fast and stable which is expected as the server is running on the same machine
- some other websocket client `id#-PF_fCkJEncteXVBAAAD` with various latency which is expected too as it's on the same network but going through some WiFi
- suddenly `id#btJn7HWoV2iHofhRAAAA` stops answering, ending up on a `pingTimeout + pingInterval` = `5 + 25` = 30ish seconds ping timeout
- then it ultimately comes back online (expected through Socket.IO auto reconnect) as a new websocket client `id#qlfogrmuaE8DPtjVAAAF` which is expected too, but it does come back only after about 60 seconds on that example (12 x 5s interval latency tests on the other client)

```
info: Websocket client id#btJn7HWoV2iHofhRAAAA latency-test-result is 2ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 7ms
info: Websocket client id#btJn7HWoV2iHofhRAAAA latency-test-result is 2ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 112ms
info: Websocket client id#btJn7HWoV2iHofhRAAAA latency-test-result is 2ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 8ms
info: Websocket client id#btJn7HWoV2iHofhRAAAA latency-test-result is 2ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 71ms
info: Websocket client id#btJn7HWoV2iHofhRAAAA latency-test-result is 1ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 11ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 9ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 15ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 9ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 13ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 11ms
info: Websocket client id#btJn7HWoV2iHofhRAAAA disconnected with reason: ping timeout
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 6ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 13ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 9ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 8ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 161ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 10ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 25ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 11ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 9ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 8ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 8ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 9ms
info: Websocket client id#qlfogrmuaE8DPtjVAAAF connected
info: Websocket client id#qlfogrmuaE8DPtjVAAAF latency-test-result is 861ms
info: Websocket client id#qlfogrmuaE8DPtjVAAAF latency-test-result is 4ms
info: Websocket client id#-PF_fCkJEncteXVBAAAD latency-test-result is 11ms
```

Again, during this network freeze, both the animations and the music on page just keep going normally.

**Environment (please complete the following information):**
 - ubuntu-core release: 22
 - hardware target: some random intel amd64 platform

**Additional context**
I'm at a loss when thinking about where to start digging in to solve this issue. Any pointers that could help me help myself? 🤗 🤓

Thank you ❤️

clorichel, 2022-11-18:
Addendum to the **Additional context**: as this has been driving me crazy 😜 I've been closely monitoring the machine `htop` during the whole time, and I haven't seen anything funky going on before, when and after the WPE Webkit ran webpage stopped responding.

woodreamz, 2023-02-17:
@clorichel I am experiencing a similar issue, it's starting to make me crazy :D

By any chance, did you find any solution or anything which help you?

closed 2023-05-27 by github-actions[bot]