Igalia/meta-webkit #431: cog not working on hardknott

gnulux, 2022-11-24:
**Describe the issue**
Wich branch is working with Yocto Hardknott release ? 
I built fsl-imx-wayland for imx8mp and i have the folowing error in terminal from  Weston. 
```
cog -P fdo https://www.google.com
(cog:1979): Cog-Core-WARNING **: 16:51:34.882: Platform module name 'fdo' is deprecated, please use 'wl' instead.
(cog:1979): Cog-Core-WARNING **: 16:51:34.884: cog_modules_get_preferred: cannot find module 'wl'
(cog:1979): Cog-Core-WARNING **: 16:51:34.884: cog_modules_get_preferred: preferred module 'wl' not supported
(cog:1979): Cog-WARNING **: 16:51:34.884: Cannot create platform: Could not find an usable platform module
wpe: could not load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory
Aborted
``` 
I suppose that cog was compiled with drm instead. it launch something when i run cog -P drm  from local console but can't read nothing as drm not supported for Vivante driver . 
I already build and run cog with Yocto (Zeus) before but with meta-webkit from Dunfelll. Perhaps I need to change branch. 
Can you help me with this

**To Reproduce**

Fikes to reproduce the behavior:
Local.conf
```
MACHINE ??= 'lec-imx8mp'
DISTRO ?= 'fsl-imx-wayland'
PACKAGE_CLASSES ?= 'package_rpm'
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"
PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
CONF_VERSION = "1"

DL_DIR ?= "${BSPDIR}/downloads/"
ACCEPT_FSL_EULA = "1"

# Switch to Debian packaging and include package-management in the image
PACKAGE_CLASSES = "package_deb"
EXTRA_IMAGE_FEATURES += "package-management"

CONNECTIVITY_CHECK_URIS = "https://www.google.com"

CORE_IMAGE_EXTRA_INSTALL += "packagegroup-adlink"

PACKAGECONFIG_append_pn-opencv_mx8 += " test"
TOOLCHAIN_TARGET_TASK_append += " onnxruntime-dev"

DISTRO_FEATURES_append = " sensors benchmarks adlink utils"
MACHINE_FEATURES_append = " wifi bluetooth"
IMAGE_FEATURES_append = " ssh-server-openssh"

OVERRIDES_append = ":wifibt"

INHERIT += "image-buildinfo"
IMAGE_BUILDINFO_VARS = "DISTRO DISTRO_VERSION MACHINE DEVICE_MODEL DEVICE_TYPE IMX_GPU_VERSION RELEASE_VERSION"

#BUILDTAG ??= "LEC-iMX8MP-1v2"
RELEASE_VERSION ?= "LEC-iMX8MP-${@'PREBUILD-%s' % (lambda dt: (dt.date.today().__str__()))(__import__('datetime')) if (d.getVar('BUILDTAG', True) is None) else '%s' % d.getVar('BUILDTAG')}"

KERNEL_MODULE_AUTOLOAD += "mlan moal"
PA_USER ?= "adlink-guest"
PA_TOKEN ?= ""

MACHINE_EXTRA_RDEPENDS += "kernel-module-lon"
BB_NUMBER_THREADS = "12"
INHERIT += "ccache"
```
2. bblayers.conf
```
LCONF_VERSION = "7"

BBPATH = "${TOPDIR}"
BSPDIR := "${@os.path.abspath(os.path.dirname(d.getVar('FILE', True)) + '/../..')}"

BBFILES ?= ""
BBLAYERS = " \
  ${BSPDIR}/sources/poky/meta \
  ${BSPDIR}/sources/poky/meta-poky \
  \
  ${BSPDIR}/sources/meta-openembedded/meta-oe \
  ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
  ${BSPDIR}/sources/meta-openembedded/meta-python \
  \
  ${BSPDIR}/sources/meta-freescale \
  ${BSPDIR}/sources/meta-freescale-3rdparty \
  ${BSPDIR}/sources/meta-freescale-distro \
"

# i.MX Yocto Project Release layers
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-bsp"
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-sdk"
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-ml"
BBLAYERS += "${BSPDIR}/sources/meta-imx/meta-v2x"
BBLAYERS += "${BSPDIR}/sources/meta-nxp-demo-experience"

BBLAYERS += "${BSPDIR}/sources/meta-browser/meta-chromium"
BBLAYERS += "${BSPDIR}/sources/meta-clang"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-gnome"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-networking"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-filesystems"
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-webserver"
BBLAYERS += "${BSPDIR}/sources/meta-qt5"
BBLAYERS += "${BSPDIR}/sources/meta-python2"

# ADLINK Yocto Project Release layers
BBLAYERS += "${BSPDIR}/sources/meta-adlink-nxp"
BBLAYERS += "${BSPDIR}/sources/meta-adlink-sema"
BBLAYERS += "${BSPDIR}/sources/meta-adlink-demo"

# Additional from open-embedded for meta-adlink-demo xfce recipes
BBLAYERS += "${BSPDIR}/sources/meta-openembedded/meta-xfce"
BBLAYERS += "${BSPDIR}/sources/meta-fnd"
BBLAYERS += "${BSPDIR}/sources/meta-webkit"
```
4. hash references of each repository of each used layer
```
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "debian-10"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "lec-imx8mp"
DISTRO               = "fsl-imx-xwayland"
DISTRO_VERSION       = "5.10-hardknott"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:58cbdaecf75b0248f96780b6882e8d4f232d038a"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:c3a36263f91e42302ad7c347e051cf1cd83e39f6"
meta-freescale       = "HEAD:80dbe4bd63bd537fc9cfda2e009f8543464b4698"
meta-freescale-3rdparty = "HEAD:7f23af99cb97a12134a46b5b9d497f05b758bf0c"
meta-freescale-distro = "HEAD:916df6d24c0a33a3b1533bde70b6a2724ec77af4"
meta-bsp             
meta-sdk             
meta-ml              
meta-v2x             = "HEAD:15fddd76b460fb56d9de121eb2d16a1e22cbe900"
meta-nxp-demo-experience = "HEAD:46107357abd2d2da9ffd702c87fce3984a422435"
meta-chromium        = "HEAD:cb3278e31340c7f081e8deb0683df2145da515c9"
meta-clang           = "HEAD:d797409435d3b0e9f2859992439989ff1d81e66d"
meta-gnome           
meta-networking      
meta-filesystems     
meta-webserver       = "HEAD:c3a36263f91e42302ad7c347e051cf1cd83e39f6"
meta-qt5             = "HEAD:a00af3eae082b772469d9dd21b2371dd4d237684"
meta-python2         = "HEAD:810d6d842f103eb59f18b06426106462b15de7e2"
meta-adlink-nxp      = "HEAD:6b9083d42e5f1a26f3a75b4157b56d49c2ed6c94"
meta-adlink-sema     = "HEAD:19a9da10bc0846597fd1df1319807133f7267296"
meta-adlink-demo     = "HEAD:618ce30ebfc7756ba50277560a497ef292c5426d"
meta-xfce            = "HEAD:c3a36263f91e42302ad7c347e051cf1cd83e39f6"
meta-fnd             = "<unknown>:<unknown>"
meta-webkit          = "hardknott:93adadbd06fa802e480ae72f834cb27f5ed850dd"
```
6. local changes applied in the build environment
```
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_RPROVIDER_virtual/wpebackend = "wpebackend-fdo"
PACKAGECONFIG_pn-wpebackend-fdo = "wayland"
PACKAGECONFIG_pn-cog = "fdo"
PACKAGECONFIG_append_pn-wpewebkit = " deviceorientation encryptedmedia fetchapi gamepad gst_gl indexeddb mediasource video webaudio webcrypto woff2 2dcanvas remote-inspector webrtc"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-ugly = " mpeg2dec x264"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-good = " cairo jpeg libpng libv4l2 soup v4l2"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-bad = " webrtc egl gles2 opengl openjpeg"
PACKAGECONFIG[webrtc] = "-DENABLE_WEB_RTC=ON -DENABLE_EXPERIMENTAL_FEATURES=ON,-DENABLE_WEB_RTC=OFF -DENABLE_EXPERIMENTAL_FEATURES=OFF,libvpx libevent libopus"
```

8. bitbake command causing the error
``` 
bitbake build works well
```
10. Error output
```
cog -P fdo https://www.google.com
(cog:1979): Cog-Core-WARNING **: 16:51:34.882: Platform module name 'fdo' is deprecated, please use 'wl' instead.
(cog:1979): Cog-Core-WARNING **: 16:51:34.884: cog_modules_get_preferred: cannot find module 'wl'
(cog:1979): Cog-Core-WARNING **: 16:51:34.884: cog_modules_get_preferred: preferred module 'wl' not supported
(cog:1979): Cog-WARNING **: 16:51:34.884: Cannot create platform: Could not find an usable platform module
wpe: could not load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory
Aborted
``` 
**Expected behavior**
expect to have cog browser on Weston compositor

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - yocto release: harknott
 - hardware targe: IMX8MP / Wayland

clopez, 2022-11-30:
This can be related to the environment.

You should start cog from inside a weston shell, this is the safe-proof method.

But if you are trying to start it from ssh then you have to at least set first the environment variables `WAYLAND_DISPLAY` and `XDG_RUNTIME_DIR` to the right values.

Example:

1. Open a weston-terminal and get the pid of it
```
# pidof weston-terminal
14881

```
2. Examine the environment variables of that pid
```
# strings /proc/14881/environ 
LANG=C
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
NOTIFY_SOCKET=/run/systemd/notify
HOME=/home/weston
LOGNAME=weston
USER=weston
SHELL=/bin/sh
INVOCATION_ID=b04467b8cdf149db8c44741a928227fe
TERM=linux
JOURNAL_STREAM=8:193511
SYSTEMD_EXEC_PID=14862
XDG_SESSION_ID=c7
XDG_RUNTIME_DIR=/run/user/1000
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
XDG_SESSION_TYPE=wayland
XDG_SESSION_CLASS=user
XDG_SESSION_DESKTOP=weston
XDG_SEAT=seat0
XDG_VTNR=7
WESTON_CONFIG_FILE=/etc/xdg/weston/weston.ini
WAYLAND_DISPLAY=wayland-1
```
3. Export the relevant ones on the ssh session
``` 
# export WAYLAND_DISPLAY=wayland-1
# export XDG_RUNTIME_DIR=/run/user/1000
```
4. Finally execute cog
```
# cog -P fdo https://google.it

(cog:15036): Cog-Core-WARNING **: 03:27:59.678: Platform module name 'fdo' is deprecated, please use 'wl' instead.

(cog:15036): GLib-GIO-WARNING **: 03:27:59.836: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
Cog-Core-Message: 03:28:00.750: <https://google.it/> Load started.
Cog-Core-Message: 03:28:00.833: <https://www.google.it/> Redirected.
Cog-Core-Message: 03:28:01.233: <https://www.google.it/> Loading...
Cog-Core-Message: 03:28:02.548: <https://www.google.it/> Loaded successfully.
```

It works to me now, but it was giving the error you get before exporting the environment variables

gnulux, 2022-11-30:
Hi @clopez

```
root@lec-imx8mp:~# pidof weston-terminal
1183

root@lec-imx8mp:~# strings /proc/1183/environ 
LANG=C
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
NOTIFY_SOCKET=/run/systemd/notify
WATCHDOG_PID=1177
WATCHDOG_USEC=40000000
HOME=/root
LOGNAME=root
USER=root
SHELL=/bin/sh
INVOCATION_ID=0403115648c140a2be019f16238336c5
TERM=linux
JOURNAL_STREAM=8:11664
XDG_SESSION_ID=c3
XDG_RUNTIME_DIR=/run/user/0
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/0/bus
XDG_SESSION_TYPE=wayland
XDG_SESSION_CLASS=user
XDG_SESSION_DESKTOP=weston
XDG_SEAT=seat0
XDG_VTNR=7
WESTON_CONFIG_FILE=/etc/xdg/weston/weston.ini
XWAYLAND_NO_GLAMOR=1
WAYLAND_DISPLAY=wayland-0

```
The 2 variables are already set but i defined thenm again. 
cog -P fdo http://google.it  still gives me the prevoius error: 
```
(cog:1275): Cog-Core-WARNING **: 16:05:25.302: Platform module name 'fdo' is deprecated, please use 'wl' instead.

(cog:1275): Cog-Core-WARNING **: 16:05:25.304: cog_modules_get_preferred: cannot find module 'wl'

(cog:1275): Cog-Core-WARNING **: 16:05:25.304: cog_modules_get_preferred: preferred module 'wl' not supported

(cog:1275): Cog-WARNING **: 16:05:25.304: Cannot create platform: Could not find an usable platform module
wpe: could not load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory

```

clopez, 2022-12-01:
Can you try exporting this environment variable :
```
export G_MESSAGES_DEBUG=all
```

... and run cog again (inside the weston environment) ?

It will print lot of debug messages, paste them here.

gnulux, 2022-12-01:
Hi 

I have only this with G_MESSAGES_DEBUG=all set .

```
(cog:1207): Cog-Core-DEBUG: 09:39:02.269: ensure_extension_points: Extension points registered.
(cog:1207): Cog-Core-DEBUG: 09:39:02.276: cog_modules_add_directory: Scanning '/usr/lib/cog/modules'
Cog-INFO: 09:39:02.285: com.igalia.Cog.Platform:
Cog-INFO: 09:39:02.285:   drm - 200/CogDrmPlatform
(cog:1207): Cog-DEBUG: 09:39:02.314: platform_setup: Platform name: fdo

(cog:1207): Cog-Core-WARNING **: 09:39:02.314: Platform module name 'fdo' is deprecated, please use 'wl' instead.

(cog:1207): Cog-Core-WARNING **: 09:39:02.314: cog_modules_get_preferred: cannot find module 'wl'

(cog:1207): Cog-Core-WARNING **: 09:39:02.314: cog_modules_get_preferred: preferred module 'wl' not supported

(cog:1207): Cog-WARNING **: 09:39:02.314: Cannot create platform: Could not find an usable platform module
(cog:1207): Cog-DEBUG: 09:39:02.314: Instantiating default WPE backend as fall-back.
wpe: could not load the impl library. Is there any backend installed?: libWPEBackend-default.so: cannot open shared object file: No such file or directory
(process:1213): GLib-GIO-DEBUG: 09:39:02.425: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(process:1213): GLib-GIO-DEBUG: 09:39:02.442: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: NetworkManager not running
(process:1213): GLib-GIO-DEBUG: 09:39:02.443: _g_io_module_get_default: Found default implementation netlink (GNetworkMonitorNetlink) for ?gio-network-monitor?
```

gnulux, 2022-12-01:
I added waylandeglinfo , it gives : 

```
EGL_VERSION = 1.5
EGL_VENDOR = Vivante Corporation
EGL_EXTENSIONS = EGL_KHR_fence_sync EGL_KHR_reusable_sync EGL_KHR_wait_sync EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_gl_renderbuffer_image EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_lock_surface EGL_KHR_create_context EGL_KHR_no_config_context EGL_KHR_surfaceless_context EGL_KHR_get_all_proc_addresses EGL_EXT_create_context_robustness EGL_EXT_protected_surface EGL_EXT_protected_content EGL_EXT_buffer_age EGL_ANDROID_native_fence_sync EGL_WL_bind_wayland_display EGL_WL_create_wayland_buffer_from_image EGL_KHR_partial_update EGL_EXT_swap_buffers_with_damage EGL_KHR_swap_buffers_with_damage EGL_EXT_pixel_format_float
EGL_CLIENT_APIS = OpenGL_ES OpenVG
GL_VERSION: OpenGL ES 1.1
GL_RENDERER:  
GL_EXTENSIONS:
```

clopez, 2022-12-01:
What do you have inside the directory `/usr/lib/cog/modules` ? 
Do you have a file named `libcogplatform-wl.so` ?

gnulux, 2022-12-01:
Yes i do

```
root@lec-imx8mp:/usr/lib/cog/modules# ls
libcogplatform-drm.so
root@lec-imx8mp:/usr/lib/cog/modules# 
```

clopez, 2022-12-02:
Ok.. there is the problem.
You are missing `libcogplatform-wl.so` in that directory 

Can you try to rebuild after setting this in your local.conf:

```
PACKAGECONFIG_pn-cog = "wl"
```
(so use `wl` instead of `fdo`)

Questions:
1. Do you have a `libcogplatform-wl.so` after that in `/usr/lib/cog/modules` ?
2. Does that make any difference? Does cog now work?

gnulux, 2022-12-02:
Ok , I rebuilt only cog and installed the deb package and it works now . 
Thanks a lot.

clopez, 2022-12-02:
Great! :+1: 
Happy to know that it works! :tada:

closed 2022-12-02 by clopez