Igalia/meta-webkit #443: no pointer mouse in kirkstone?

ihermida, 2023-01-24:
Hi guys,

I do not have the mouse pointer on a building for kirsktone.

\# cog --version
0.14.1 (WPE WebKit 2.38.2)

My relevant sha1 layers are:

meta-yocto-bsp = "HEAD:d3e378397395542179a8497fbff4b610f6f7e802"
meta-oe
meta-python
meta-networking
meta-webserver = "HEAD:1f31570d0795da90083d1dbf28127c90908e30ee"
eta-freescale = "HEAD:3f5fc5810cfb7be61b487ade0a7ecfaeafa3d2aa"
meta-webkit = "HEAD:eb31f54387bfd309ec3e936b5f19febaac8045bb"

Does anyone have the same problem or is just me? Maybe I need to export some environment variable to see it?
I checked on a touch display and in the HDMI monitor without luck. I am running it on top of wayland, where the mouse works.

psaavedra, 2023-01-26:
Assuming you are talking about the Cog FDO platform for Wayland (e.g. Weston), I think is matter of the  Weston itself. You could try with this attached file (backported recipes of wayland and weston from Poky master).

```
recipes-backport/
└── wayland
    ├── libinput
    │   ├── determinism.patch
    │   └── run-ptest
    ├── libinput_1.19.4.bb
    ├── mtdev_1.1.6.bb
    ├── required-distro-features.inc
    ├── wayland
    │   ├── 0001-build-Fix-strndup-detection-on-MinGW.patch
    │   ├── 0002-Do-not-hardcode-the-path-to-wayland-scanner.patch
    │   └── run-ptest
    ├── wayland_1.20.0.bb
    ├── wayland-protocols_1.27.bb
    ├── wayland-utils_1.0.0.bb
    ├── weston
    │   ├── systemd-notify.weston-start
    │   ├── weston.desktop
    │   ├── weston.png
    │   └── xwayland.weston-start
    ├── weston_11.0.0.bb
    ├── weston-init
    │   ├── init
    │   ├── weston-autologin
    │   ├── weston.env
    │   ├── weston.ini
    │   ├── weston.service
    │   ├── weston.socket
    │   └── weston-start
    └── weston-init.bb
```
You could add these recipes in one of your project layers:

[wayland.zip](https://github.com/Igalia/meta-webkit/files/10508498/wayland.zip)

ihermida, 2023-01-27:
Hi Pablo,

Thanks for your replay... 
The attached zip is empty, could you reattach it

"""
$ unzip -l wayland.zip 
Archive:  wayland.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
        0  2023-01-25 16:26   wayland/
---------                     -------
        0                     1 file
"""

Regardless to that, I get all all the poky fixes you mentioned from poky master, but despite of that, the mouse still does not work on my side.

"""
project sources/poky/                           (*** NO BRANCH ***)
 --     meta/recipes-graphics/wayland/files/0001-wayland-info-Fix-build-without-libdrm.patch
 --     meta/recipes-graphics/wayland/libinput_1.22.0.bb
 --     meta/recipes-graphics/wayland/wayland-protocols_1.31.bb
 --     meta/recipes-graphics/wayland/wayland-utils_1.1.0.bb
 --     meta/recipes-graphics/wayland/wayland_1.21.0.bb
 -m     meta/recipes-graphics/wayland/weston-init.bb
 -m     meta/recipes-graphics/wayland/weston/xwayland.weston-start
 --     meta/recipes-graphics/wayland/weston_11.0.1.bb
"""

I removed all my sstate and cache files, so I did a build from scratch, but with no luck.

I still has the issue no resolved. I was looking for a solution, and in some threads people mention something about "COG_USE_WAYLAND_CURSOR".
Maybe should I try with that option?

Using xwayland I can see the cursor, but not when launching "cog" on top of it.

BR / Isaac

psaavedra, 2023-01-27:
Here: [wayland.zip](https://github.com/Igalia/meta-webkit/files/10517431/wayland.zip)

I experimented the same issue and finally I fixed by backporting the wayland and weston recipes.

To check what version of the recipes is using your *bitbake* environment:  `bitbake-layers  show weston wayland*

ihermida, 2023-01-27:
Thanks!

Finally I have it working. For that, I have applied the next patch on my recipe (https://github.com/Igalia/cog/commit/fe83f9b721f67ffa4d53c5261fc447355ebe06da)

webkit/recipes-browser/cog/cog/0002-wl-Fix-wrong-wl_shm-for-cursor.patch
webkit/recipes-browser/cog/cog_%.bbappend

I did not need to apply your attached zip file

closed 2023-01-27 by ihermida