Igalia/meta-webkit #446: get_drm_fd error on stm32

nandra, 2023-02-03:
Compiling cog for stm32 platform kirkstone branch getting following :

```
cog/0.14.1-r0/cog-0.14.1/platform/drm/cog-drm-modeset-renderer.c:110:1: error: conflicting types for 'get_drm_fd'; have 'int(CogDrmModesetRenderer *)'
  110 | get_drm_fd(CogDrmModesetRenderer *self)
      | ^~~~~~~~~~
In file included from /data/projects/test/yocto/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/cog/0.14.1-r0/cog-0.14.1/platform/drm/cog-drm-modeset-renderer.c:11:
/data/projects/test/yocto/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/cog/0.14.1-r0/recipe-sysroot/usr/include/gbm.h:246:1: note: previous declaration of 'get_drm_fd' with type 'int(void)'
  246 | get_drm_fd(void);
      | ^~~~~~~~~~
```

Any ideas? Seem local static function clash with one defined in drm headers?

clopez, 2023-02-03:
It seems something in your environment (likely non-mesa graphics drivers) is defining a type different than expected.

Can you paste here the content (or link to it) of this file?  
`/data/projects/test/yocto/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/cog/0.14.1-r0/recipe-sysroot/usr/include/gbm.h`

nandra, 2023-02-06:
@clopez this is file content:
```/*
 * Copyright © 2011 Intel Corporation
 *
 * Permission is hereby granted, free of charge, to any person obtaining a
 * copy of this software and associated documentation files (the "Software"),
 * to deal in the Software without restriction, including without limitation
 * the rights to use, copy, modify, merge, publish, distribute, sublicense,
 * and/or sell copies of the Software, and to permit persons to whom the
 * Software is furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice (including the next
 * paragraph) shall be included in all copies or substantial portions of the
 * Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 * EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
 * MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 * NONINFRINGEMENT.  IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 * HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 * WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
 * DEALINGS IN THE SOFTWARE.
 *
 * Authors:
 *    Benjamin Franzke <benjaminfranzke@googlemail.com>
 */

#ifndef _GBM_H_
#define _GBM_H_

#define __GBM__ 1

#include <stddef.h>
#include <stdint.h>

#ifdef __cplusplus
extern "C" {
#endif

/**
 * \file gbm.h
 * \brief Generic Buffer Manager
 */

struct gbm_device;
struct gbm_bo;
struct gbm_surface;

/**
 * \mainpage The Generic Buffer Manager
 *
 * This module provides an abstraction that the caller can use to request a
 * buffer from the underlying memory management system for the platform.
 *
 * This allows the creation of portable code whilst still allowing access to
 * the underlying memory manager.
 */

/**
 * Abstraction representing the handle to a buffer allocated by the
 * manager
 */
union gbm_bo_handle {
   void *ptr;
   int32_t s32;
   uint32_t u32;
   int64_t s64;
   uint64_t u64;
};
[... 325 lines cut ...]

void
gbm_surface_release_buffer(struct gbm_surface *surface, struct gbm_bo *bo);

int
gbm_surface_has_free_buffers(struct gbm_surface *surface);

int
gbm_surface_get_in_fence_fd(struct gbm_surface *surface);

void
gbm_surface_set_in_fence_fd(struct gbm_surface *surface, int fd);

void
gbm_surface_set_sync_post(struct gbm_surface *surface, int sync_post);

void
gbm_surface_destroy(struct gbm_surface *surface);

#ifdef __cplusplus
}
#endif

#endif
```

nandra, 2023-02-06:
@clopez seems it's coming from this recipe: https://github.com/STMicroelectronics/meta-st-stm32mp/tree/kirkstone/recipes-graphics/gcnano-userland

Specifically from: `gcnano-userland-multi-binary-stm32mp.bb` recipe.

nandra, 2023-02-07:
I poked a bit and overcome this error but when start browser it says:
```qt-wpe-simple-browser -platform eglfs https://google.com
Failed to move cursor on screen DSI1: -13
Failed to move cursor on screen DSI1: -13
qt.qpa.input: xkbcommon not available, not performing key mapping
QQmlApplicationEngine failed to load component
qrc:/main.qml:3:1: module "org.wpewebkit.qtwpe" is not installed
root@stm32:~# find / -name *wpewebkit*
/usr/lib/qt5/qml/org/wpewebkit
root@stm32:~# ls -la /usr/lib/qt5/qml/org/wpewebkit
drwxr-xr-x    3 root     root          1024 Sep 20 15:07 .
drwxr-xr-x    3 root     root          1024 Sep 20 15:07 ..
drwxr-xr-x    2 root     root          1024 Sep 20 15:07 qtwpe
root@stm32:~# ls -la /usr/lib/qt5/qml/org/wpewebkit/qtwpe/
drwxr-xr-x    2 root     root          1024 Sep 20 15:07 .
drwxr-xr-x    3 root     root          1024 Sep 20 15:07 ..
-rwxr-xr-x    1 root     root         43332 Sep 20 15:07 libqtwpe.so
-rw-r--r--    1 root     root            40 Sep 20 15:07 qmldir```

and qtwpe exists as show above. Any ideas? Thanks.

nandra, 2023-02-07:
ok resolved by: `export QML2_IMPORT_PATH='/usr/lib/qt5/qml/'`

psaavedra, 2023-02-07:
The first comment in the issue makes reference to the DRM cog platform but at some point the issue turned to QT. What is the platform where you want to run Cog?

nandra, 2023-02-07:
@psaavedra sorry fon cofucion. I've basically want to compile qt-wpe-webkit-browser and first hit issue with compiling cog. Then was able to resolve it and have another issue with running browser. Should I close this and create other one pls? Thanks.

closed 2023-05-18 by github-actions[bot]