Igalia/meta-webkit #449: video src=rtsp://

mickdev, 2023-02-22:
Hello,

With a raspberry pi 4 build.

I try to get a page to display html with cog a <video src="rtsp://..."> ( normal video file:// work great )
But the  it show nothing in cog (i run cog without weston with render gles).

XDG_RUNTIME_DIR=/var/run/user/root cog -O renderer=gles file:///storage/content/index.html

index.html
<video style="width:100%;height:100%;" autostart autplay muted src="rtsp://xxx:xxxx@192.168.1.19/axis-media/media.amp" ></video>

**It does not work !**

if i do a run of gstreamer play

XDG_RUNTIME_DIR=/var/run/user/root gst-play-1.0 rtsp://xxx:xxxx@192.168.1.19/axis-media/media.amp

**It work !**

Note, if i set "export GST_DEBUG=3", i have no message from gstreamer (only this) when i run cog
0:00:00.063635951 13762  0x16f5120 WARN    webkitregistryscanner GStreamerRegistryScanner.cpp:211:hasElementForCaps: All video encoder elements matching caps video/x-av1 are disallowed
0:00:00.073386272 13762  0x16f5120 WARN    webkitregistryscanner GStreamerRegistryScanner.cpp:211:hasElementForCaps: All video encoder elements matching caps video/x-av1 are disallowed
when i rung gst-play-1.0 i got a lot of message

---------------------------

Any idea ? it was working on old version with WPELauncher ..with raspberry pi 3 ( sometimes ago :) ).

clopez, 2023-02-23:
does it work if you upload that `index.html` file to a webserver and try to access via `http://` ?
If that is the case then you are likely missing to install `shared-mime-info` on your build

mickdev, 2023-02-23:
I will test .. but :
I can perfectly play video (local mp4) in cog with file:///storage/content/index.html ( and <video src="file:///storage/content/mymp4.mp4" in the index.html), and yes i have shared-mime-info ( to play local file )

The problem is specific with rtsp:// ... i will also test with udp:// since i know it was also working with WPELauncher ( video (mp4/h264)  work with http:// and file://

philn, 2023-03-14:
https://bugs.webkit.org/show_bug.cgi?id=227904

closed 2023-06-20 by github-actions[bot]