Igalia/meta-webkit #459: core-image-weston build failure when adding wpewebkit and cog to IMAGE_INSTALL in local.conf

baibaidashixiong, 2023-03-21:
Hi, 
I want to integrate a browser into image, so I followed the tutorial of WPE wiki to add to following to local.conf:
```
IMAGE_INSTALL:append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
```
then `bitbake core-image-weston`, then it builds failed when do_rootfs, here is the specific error logs:
```
ERROR: Unable to install packages. Command '/home/zzz/yoccto/os/poky/build-x86/tmp-glibc/work/qemux86_64-oe-linux/core-image-weston/1.0-r0/recipe-sysroot-native/usr/bin/opkg --volatile-cache -f /home/zzz/yoccto/os/poky/build-x86/tmp-glibc/work/qemux86_64-oe-linux/core-image-weston/1.0-r0/opkg.conf -t /home/zzz/yoccto/os/poky/build-x86/tmp-glibc/work/qemux86_64-oe-linux/core-image-weston/1.0-r0/temp/ipktemp/ -o /home/zzz/yoccto/os/poky/build-x86/tmp-glibc/work/qemux86_64-oe-linux/core-image-weston/1.0-r0/rootfs  --force_postinstall --prefer-arch-to-version   install base-passwd chromium-x11 cog gtk+3-demo loongson matchbox-terminal opkg packagegroup-base-extended packagegroup-core-boot packagegroup-core-ssh-dropbear packagegroup-core-weston psplash run-postinsts shadow weston-xwayland wpewebkit' returned 2:
 * Solver encountered 2 problem(s):
 * Problem 1/2:
 *   - conflicting requests
 *   - nothing provides gstreamer1.0-plugins-base-videoconvert needed by wpewebkit-2.38.5-r0.core2-64
 * Solution 1:
 *   - do not ask to install a package providing wpewebkit

 * Problem 2/2:
 *   - package cog-0.14.1-r0.core2-64 requires wpewebkit >= 2.36, but none of the providers can be installed
 *   - conflicting requests
 *   - nothing provides gstreamer1.0-plugins-base-videoconvert needed by wpewebkit-2.38.5-r0.core2-64
 *
 * Solution 1:
 *   - do not ask to install a package providing cog
```

**To Reproduce**

Fikes to reproduce the behavior:
1. local.conf:add following code
```
IMAGE_INSTALL:append = " wpewebkit cog"
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
```
2. bblayers.conf: has following layers:
```
meta/ meta-poky/  meta-yocto-bsp/ meta-qt5/ meta-openembedded/meta-oe/ meta-webkit/
```
3. hash references of each repository of each used layer:
```
meta-yocto-bsp       = "master:bb839ca78db99b8a4338cf4f072fd62af7916b53"
meta-qt5             = "master:190e326fb3f15f2d0ca93dfa4815f4050cf44f25"
meta-oe              = "master:093077adc12e8b82eefe8feaae2c07224fd68a34"
meta-webkit          = "main:84ee06d1df77801984b0c16f9f17edc0a83257fa"
```

**Expected behavior**
can successfully build the core-image-weston image.

**Environment (please complete the following information):**
 - yocto release: poky master branch 
 - hardware : Linux 5.15.0-48-generic 20.04.1-Ubuntu x86_64 GNU/Linux

What caused these errors and how can i fix them?

clopez, 2023-03-21:
Seems this package was renamed on poky commit 4e07f56dced0b06d2cb60530e3989f656eede2cd (OE-Core rev: fb2d28e0315ece6180c87c7047587673024a09f7)

Can you try patch from https://github.com/Igalia/meta-webkit/pull/460 ?

baibaidashixiong, 2023-03-22:
> Seems this package was renamed on poky commit 4e07f56dced0b06d2cb60530e3989f656eede2cd (OE-Core rev: fb2d28e0315ece6180c87c7047587673024a09f7)
> 
> Can you try patch from #460 ?

Yes, this patch works for me now, thank you very much!

commit f688fe55 (2023-03-23): wpewebkit: gst videoconvert/videoscale plugins were merged into one.

A new plugin named gstreamer1.0-plugins-base-videoconvertscale contains
the previous plugins videoconvert/videoscale merged.

Since Yocto >= Mickledore (4.2)
( - Yocto commit 4e07f56dced0b06d2cb60530e3989f656eede2cd or
  - OE-Core rev: fb2d28e0315ece6180c87c7047587673024a09f7 )

Fixes: #459