Igalia/meta-webkit #469: `libavif` does not build with LLVM/clang as the default toolchain

jaskij, 2023-05-13:
**Bug description**

`libavif` does not build with LLVM/clang as the default toolchain.

**How To Reproduce**

Try to build using clang as the default toolchain, from [`meta-clang`](https://github.com/kraj/meta-clang), Kirkstone branch.

(internal layers removed, `meta-rust` is using an internal fork which, among other things, is updated to 1.69)
```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-ultima-linux"
MACHINE              = "imx8mprsb3720a1"
DISTRO               = "ultima"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "aarch64 armv8a crc cortexa53 crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "kirkstone:f53ab3a2ff206a130cdc843839dd0ea5ec4ad02f"
meta-oe              
meta-filesystems     
meta-networking      
meta-python          = "kirkstone:df452d9d98659b49888fa8a5428a0a8bd3e3aaec"
meta-freescale       = "kirkstone:adff7626fdbcf789a3ce5781cab0385c1e831717"
meta-intel           = "kirkstone:f709b3f2b05e2ac73c6734dc2a213382b5eaf579"
meta-clang           = "kirkstone:68ec449f97ffa58d835163581fc72afcb08f027b"
meta-rust            = "HEAD:0efddd14653ab4c08f81aa81c720c93b4841ab77"
meta-webkit          = "master:acd126e3b3e95f198dbbb9e48dc9bf2993d29b18"
```

**Expected behavior**

`libavif` builds normally

**Bug behavior**

A lot of errors like below. I am using a custom distro, replaced with `poky` because weird workplace rules.

```
| /home/jaskij/yocto/build/tmp-glibc/work/cortexa53-crypto-poky-linux/libavif/0.9.0-r0/libavif-0.9.0/src/reformat.c:277:40: error: mixing declarations and code is incompatible with standards before C99 [-Werror,-Wdeclaration-after-statement]
|                             uint16_t * pV = (uint16_t *)&yuvPlanes[AVIF_CHAN_V][(i * 2) + (j * yuvRowBytes[AVIF_CHAN_V])];
|                                        ^
| /home/jaskij/yocto/build/tmp-glibc/work/cortexa53-crypto-poky-linux/libavif/0.9.0-r0/libavif-0.9.0/src/reformat.c:317:32: error: mixing declarations and code is incompatible with standards before C99 [-Werror,-Wdeclaration-after-statement]
|                     uint16_t * pV = (uint16_t *)&yuvPlanes[AVIF_CHAN_V][(uvI * 2) + (uvJ * yuvRowBytes[AVIF_CHAN_V])];
```

**Environment (please complete the following information):**

 - yocto release: kirkstone
 - hardware target: i.MX8 MP

**Additional context**

Forcing the toolchain to GNU by adding `TOOLCHAIN = "gcc"` in a `.bbappend` is a valid workaround I'm currently employing.

clopez, 2023-05-13:
Can you try this patch https://github.com/AOMediaCodec/libavif/pull/886 to see if it fixes the issue?

jaskij, 2023-05-15:
@clopez I can not apply that patch. It does not apply to either `main` or `0.11.1`, and the PR itself is against `master` which seems to no longer exist.

Do you think I should try compiling from the PR branch directly?

jaskij, 2023-05-15:
After further testing, it seems merely bumping `libavif` to `0.11.1` fixes the bug, without a need for a patch.

On that note, I created #470

closed 2023-08-15 by jaskij