Igalia/meta-webkit #476: EGL_NOT_INITIALIZED issue

matteolisi, 2023-06-19:
Hi, 

I'm testing the wekbit on rockchip px30 with yocto kirkstone.

When launching cog, it returns an EGL_NOT_INITIALIZED error: 
````
root@rockchip-px30-evb:~# cog --platform=fdo www.google.com 
Rga built version:274b345f976a 

(cog:529): Cog-Core-WARNING **: 09:24:35.947: Platform module name 'fdo' is deprecated, please use 'wl' instead.
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.

(cog:529): GLib-GIO-WARNING **: 09:24:36.030: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
Rga built version:274b345f976a 
Rga built version:274b345f976a 
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.
EGLDisplay Initialization failed: EGL_NOT_INITIALIZED
Cannot create EGL context: invalid display (last error: EGL_SUCCESS)
Cog-Core-Message: 09:24:36.487: <http://www.google.com/> Load started.
Cog-Core-Message: 09:24:36.502: <https://www.google.com/> Redirected.
Cog-Core-Message: 09:24:36.784: <https://www.google.com/> Loading...
Cog-Core-Message: 09:24:38.639: <https://www.google.com/> Loaded successfully.
````
The system variables are:

````
export COG_PLATFORM_FDO_VIEW_FULLSCREEN="1"
export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/0/bus"
export EDITOR="vi"
export HOME="/home/root"
export HUSHLOGIN="FALSE"
export LANG="C"
export LOGNAME="root"
export MAIL="/var/spool/mail/root"
export MOTD_SHOWN="pam"
export OLDPWD
export OPIEDIR
export PATH="/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin"
export PS1="\\u@\\h:\\w\\\$ "
export PWD="/home/root"
export QPEDIR
export QTDIR
export SHELL="/bin/sh"
export SHLVL="1"
export SYSTEMD_PAGER=""
export TERM="linux"
export USER="root"
export XDG_RUNTIME_DIR="/run/user/1000/"
export XDG_SESSION_CLASS="user"
export XDG_SESSION_ID="c2"
export XDG_SESSION_TYPE="tty"
````

The local.conf variables are:
````
MACHINE = "rockchip-px30-evb"

DISTRO_FEATURES:remove += " x11"

IMAGE_INSTALL:append = " wpewebkit cog "
PREFERRED_PROVIDER_virtual/wpebackend = "wpebackend-fdo"
PREFERRED_PROVIDER_virtual/libwpe = "wpebackend"

PARALLEL_MAKE = "-j 8"
DISTRO_FEATURES:append = " gles2 opengl wayland "
````

My linux version is:
````
Linux rockchip-px30-evb 5.10.110-rockchip-standard #1 SMP Mon Jun 5 14:26:06 UTC 2023 aarch64 GNU/Linux
````

What it's wrong ?

matteolisi, 2023-06-19:
I'm Sorry ,  I forget the cog and webkit versions:

````
root@rockchip-px30-evb:~# cog --version
Rga built version:274b345f976a 
0.16.1 (WPE WebKit 2.38.2)
````

philn, 2023-06-19:
Are you running this in a wayland compositor?

matteolisi, 2023-06-19:
Yes , cog is running on wayland composerInviato dal mio Galaxy
-------- Messaggio originale --------Da: Philippe Normand ***@***.***> Data: 19/06/23  16:48  (GMT+01:00) A: Igalia/meta-webkit ***@***.***> Cc: MATTEO LISI ***@***.***>, Author ***@***.***> Oggetto: Re: [Igalia/meta-webkit] EGL_NOT_INITIALIZED issue (Issue #476) 
Are you running this in a wayland compositor?

clopez, 2023-06-19:
I don't see a `WAYLAND_DISPLAY`  variable in the list of environment variables you posted above. Defining this one is key to be able to connect to the compositor.

matteolisi, 2023-06-20:
Hi,
I set the WAYLAND_DISPLAY as:

export WAYLAND_DISPLAY=wayland-0

but cog still give me the same error

many thanks

clopez, 2023-06-20:
Export the env var `G_MESSAGES_DEBUG=all` and run Cog again. Paste here the output

matteolisi, 2023-06-20:
Hi 

I lauched the application with commands

````
export XDG_RUNTIME_DIR=/run/user/1000/
export COG_PLATFORM_FDO_VIEW_FULLSCREEN=1
export WAYLAND_DISPLAY=wayland-0
cog --platform=fdo www.google.com  >file.txt 2>&1
````

And this is the log:

````
Rga built version:274b345f976a 
(process:567): Cog-Core-DEBUG: 15:56:00.038: ensure_extension_points: Extension points registered.
(process:567): Cog-Core-DEBUG: 15:56:00.039: cog_modules_add_directory: Scanning '/usr/lib/cog/modules'
Cog-INFO: 15:56:00.062: com.igalia.Cog.Platform:
Cog-INFO: 15:56:00.063:   wl - 500/CogWlPlatform
Cog-INFO: 15:56:00.063:   drm - 200/CogDrmPlatform
Cog-INFO: 15:56:00.063:   headless - 100/CogHeadlessPlatform
(cog:567): GLib-GIO-DEBUG: 15:56:00.076: _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
(cog:567): Cog-DEBUG: 15:56:00.093: platform_setup: Platform name: fdo

(cog:567): Cog-Core-WARNING **: 15:56:00.094: Platform module name 'fdo' is deprecated, please use 'wl' instead.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.101: Initializing Wayland...
(cog:567): Cog-Wayland-DEBUG: 15:56:00.102: Using 'wl_compositor' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Using 'wl_subcompositor' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Ignoring 'wp_viewporter' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Ignoring 'zxdg_output_manager_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Using 'wp_presentation' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Ignoring 'zwp_relative_pointer_manager_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Ignoring 'zwp_pointer_constraints_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Ignoring 'zwp_input_timestamps_manager_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Ignoring 'wl_data_device_manager' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.103: Using 'wl_shm' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'mali_buffer_sharing' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Using 'wl_seat' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'zwp_linux_dmabuf_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'weston_direct_display_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'weston_content_protection' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Using 'wl_output' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'zwp_input_panel_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'zwp_input_method_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Using 'zwp_text_input_manager_v1' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Using 'xdg_wm_base' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.104: Ignoring 'weston_desktop_shell' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.105: Ignoring 'weston_screenshooter' interface obtained from the Wayland registry.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.105: Initializing EGL...
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.
Cog-Wayland-INFO: 15:56:00.131: EGL version 1.4 initialized.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.131: Creating Wayland surface...
(cog:567): Cog-Core-DEBUG: 15:56:00.157: gamepad setup: none
(cog:567): Cog-DEBUG: 15:56:00.157: platform_setup: Selected CogWlPlatform @ 0x55aaa16260

(cog:567): GLib-GIO-WARNING **: 15:56:00.169: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.169: Enumerating seat capabilities:
(cog:567): Cog-Wayland-DEBUG: 15:56:00.169:   - Keyboard
(cog:567): Cog-Wayland-DEBUG: 15:56:00.170:   - Touch
(cog:567): Cog-Wayland-DEBUG: 15:56:00.170: Done enumerating seat capabilities.
(cog:567): Cog-Wayland-DEBUG: 15:56:00.170: Seat name: 'default'
Cog-Wayland-INFO: 15:56:00.171: Got scale factor 1 for output 0x55aaa20140

Cog-Wayland-INFO: 15:56:00.171: Output 0x55aaa20140 is 1024x600 @ 60.72Hz
(cog:567): Cog-Wayland-DEBUG: 15:56:00.199: New XDG toplevel configuration: (0, 0)
(cog:567): Cog-Wayland-DEBUG: 15:56:00.200: Resized EGL buffer to: (1024, 768) @1x

Rga built version:274b345f976a 
(process:581): GLib-GIO-DEBUG: 15:56:00.329: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(process:581): GLib-GIO-DEBUG: 15:56:00.354: Failed to initialize networkmanager (GNetworkMonitorNM) for gio-network-monitor: NetworkManager not running
[... 1574 lines cut ...]
(process:581): GLib-Net-DEBUG: 15:56:02.213: CLIENT[0x55a00229b0]: yielding operation OP_READ
[... cut ...]
(process:581): GLib-Net-DEBUG: 15:56:02.222: CLIENT[0x55a00229b0]: starting to write 13 bytes to TLS connection
(process:581): GLib-Net-DEBUG: 15:56:02.222: CLIENT[0x55a00229b0]: claiming operation OP_WRITE
(process:581): GLib-Net-DEBUG: 15:56:02.222: CLIENT[0x55a00229b0]: claiming operation OP_WRITE succeeded
(process:581): GLib-Net-DEBUG: 15:56:02.222: CLIENT[0x55a00229b0]: yielding operation OP_WRITE
(process:581): GLib-Net-DEBUG: 15:56:02.222: CLIENT[0x55a00229b0]: successfully write 13 bytes to TLS connection
Cog-Core-Message: 15:56:02.529: <https://www.google.com/> Loaded successfully.
````

clopez, 2023-06-20:
According to that log Cog is able to open succesfully the EGLDisplay but then the WebProcess fails to do that.

This comes from Cog:
```
Cog-Wayland-INFO: 15:56:00.171: Got scale factor 1 for output 0x55aaa20140

Cog-Wayland-INFO: 15:56:00.171: Output 0x55aaa20140 is 1024x600 @ 60.72Hz
(cog:567): Cog-Wayland-DEBUG: 15:56:00.199: New XDG toplevel configuration: (0, 0)
(cog:567): Cog-Wayland-DEBUG: 15:56:00.200: Resized EGL buffer to: (1024, 768) @1x
```

And the fatal error comes later from the WPEWebProcess at Source/WebCore/platform/graphics/PlatformDisplay.cpp
```
EGLDisplay Initialization failed: EGL_NOT_INITIALIZED
```

Not sure why.

Can you please try with the last version of WPEWebKit? you are using 2.38.x but the last stable is 2.40.x

Try to apply also this patch to wpewebkit that looks relevant:
https://github.com/WebKit/WebKit/commit/6410ba3f6f3bdc37e5bdc53803b8418a83e3c433.patch

And if it continues to fails can you provide this extra info?:

1) build into the image the utility `waylandeglinfo` (the recipe is provided in this layer), execute it under the wayland environment and paste the output of it?
2) what is the output of the command `ls -l /dev/dri/` ?

philn, 2023-06-20:
i think we had bad experiences in the past with the mali drivers, so this is not very surprising to me...

matteolisi, 2023-06-21:
Hi,

I applied the patch :

https://github.com/WebKit/WebKit/commit/6410ba3f6f3bdc37e5bdc53803b8418a83e3c433.patch

But it doesn't resolt the issue , cog gives the same error.

Next , I tried to build the wpewebkit 2.40 version but I receive some linking error on gbm_create_device function, and I need to investigare on it...

Below the output of commands that you asked:

````
root@px30-evb:~# ls -l /dev/dri/
drwxr-xr-x    2 root     root            80 Jun 21 09:07 by-path
crw-rw----    1 root     video     226,   0 Jun 21 09:07 card0
crw-rw-rw-    1 root     render    226, 128 Jun 21 09:07 renderD128

root@px30-evbt:~# waylandeglinfo
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.
EGL_VERSION = 1.4 Bifrost-"g2p0-01eac0"
EGL_VENDOR = ARM
EGL_EXTENSIONS = EGL_WL_bind_wayland_display EGL_KHR_partial_update EGL_EXT_swap_buffers_with_damage EGL_KHR_swap_buffers_with_damage EGL_KHR_config_attribs EGL_KHR_image EGL_KHR_image_base EGL_KHR_fence_sync EGL_KHR_wait_sync EGL_KHR_gl_colorspace EGL_KHR_get_all_proc_addresses EGL_IMG_context_priority EGL_KHR_no_config_context EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers EGL_EXT_yuv_surface EGL_EXT_pixel_format_float EGL_ARM_pixmap_multisample_discard EGL_KHR_gl_texture_2D_image EGL_KHR_gl_renderbuffer_image EGL_KHR_create_context EGL_KHR_surfaceless_context EGL_KHR_gl_texture_cubemap_image EGL_EXT_image_gl_colorspace EGL_EXT_create_context_robustness
EGL_CLIENT_APIS = OpenGL_ES
GL_VERSION: OpenGL ES-CM 1.1 v1.g2p0-01eac0.7abca4a459ce7997f20ab707543ff422
GL_RENDERER: Mali-G31
GL_EXTENSIONS:
    GL_OES_byte_coordinates, GL_OES_fixed_point, GL_OES_single_precision, 
    GL_OES_matrix_get, GL_OES_compressed_paletted_texture, 
    GL_OES_point_size_array, GL_OES_point_sprite, GL_OES_read_format, 
    GL_OES_compressed_ETC1_RGB8_texture, GL_OES_depth24, GL_OES_stencil8, 
    GL_OES_framebuffer_object, GL_OES_packed_depth_stencil, GL_OES_rgb8_rgba8, 
    GL_EXT_read_format_bgra, GL_OES_matrix_palette, 
    GL_OES_extended_matrix_palette, GL_OES_draw_texture, 
    GL_OES_blend_equation_separate, GL_OES_blend_func_separate, 
    GL_OES_blend_subtract, GL_OES_stencil_wrap, 
    GL_OES_texture_mirrored_repeat, GL_EXT_texture_format_BGRA8888, 
    GL_OES_query_matrix, GL_OES_EGL_image, GL_OES_EGL_image_external, 
    GL_OES_EGL_sync, GL_OES_texture_npot, GL_OES_vertex_half_float, 
    GL_OES_required_internalformat, GL_OES_vertex_array_object, 
    GL_OES_mapbuffer, GL_OES_fbo_render_mipmap, GL_OES_element_index_uint, 
    GL_ARM_rgba8, GL_EXT_blend_minmax, GL_EXT_discard_framebuffer, 
    GL_EXT_texture_storage, GL_OES_texture_compression_astc, 
    GL_KHR_texture_compression_astc_ldr, GL_KHR_texture_compression_astc_hdr, 
    GL_KHR_texture_compression_astc_sliced_3d, 
    GL_EXT_texture_compression_astc_decode_mode, 
    GL_EXT_texture_compression_astc_decode_mode_rgb9e5, 
    GL_OES_surfaceless_context, GL_EXT_multisampled_render_to_texture, 
    GL_OES_texture_cube_map, GL_KHR_debug, GL_EXT_sRGB, GL_EXT_robustness
````

many thanks

clopez, 2023-06-22:
Can you try the patch from https://github.com/Igalia/meta-webkit/pull/477  to see if that allows building 2.40? Note: disable the `gbm` `PACKAGECONFIG` option on your build

matteolisi, 2023-06-22:
With the previous patch I have another error:
````
| /home/matteo/thirdroot/yocto/px30-kirkstone/yocto/build/tmp/work/armv8a-poky-linux/wpewebkit/2.40.2-r0/wpewebkit-2.40.2/Source/WTF/wtf/glib/Sandbox.cpp:49:13: error: 'BWRAP_EXECUTABLE' was not declared in this scope
|    49 |             BWRAP_EXECUTABLE,
|       |             ^~~~~~~~~~~~~~~~
````
Many thanks

matteolisi, 2023-06-22:
After some tricks I was able to build the 2.40.2 webkit version... 

Unfortunatelly it still not working...

The error seems to be the same:

````
root@px30-evb:~# cog --platform=fdo www.google.com  
Rga built version:274b345f976a 

(cog:327): Cog-Core-WARNING **: 16:16:32.081: Platform module name 'fdo' is deprecated, please use 'wl' instead.
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.
Rga built version:274b345f976a 

(cog:327): GLib-GIO-WARNING **: 16:16:32.521: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
Rga built version:274b345f976a 
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '7'.
EGLDisplay Initialization failed: EGL_NOT_INITIALIZED
Cannot create EGL sharing context: invalid display (last error: EGL_SUCCESS)
Cannot create EGL context: invalid display (last error: EGL_SUCCESS)
````

clopez, 2023-06-29:
Can you try with the DRM plugin? For that do the following:

1. Stop wayland/weston so you only see the default text-based prompt login on the board
2. Start cog like this:
```
cog --platform=drm https://google.com
```

If it fails to display anything then try to start cog like this:
```
cog --platform=drm --platform-params="renderer=gles" https://google.com
```

russdill, 2023-07-04:
I seem to be having similar issues. enabling debugging shows that cog starts up EGL ok, but the EGL_NOT_INITIALIZED errors come later. If I run just with drm:

(cog:31286): Cog-DRM-WARNING **: 00:22:56.580: failed to import a wl_buffer resource into gbm_bo

(cog:31286): Cog-DRM-WARNING **: 00:22:59.039: failed to import a wl_buffer resource into gbm_bo
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display

(block screen, I'm not sure why it's giving the numeric code 12289, aka 0x3001, aka EGL_NOT_INITIALIZED, rather than the string)

and If I run with renderer=gles

(cog:31286): Cog-DRM-WARNING **: 00:22:56.580: failed to import a wl_buffer resource into gbm_bo

(cog:31286): Cog-DRM-WARNING **: 00:22:59.039: failed to import a wl_buffer resource into gbm_bo
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display
ERR: Display.cpp:1014 (initialize): ANGLE Display::initialize error 12289: Failed to get system egl display

(Working display)

russdill, 2023-07-06:
It seems that when first starting, the wayland backend gets used. But at some point some element requires accelation, (webgl?) and something tries to open EGL with EGL_PLATFORM_SURFACELESS_MESA which fails on my platform.

woutervanh, 2023-09-12:
I'll jump in on this one. I have similar issues on rk3288 (Mali T760)
At compile time, I had the same ` linking error on gbm_create_device function` errors. libgbm was in the sysroot, but the gbm functions were in libmali and libmali-hook. After some fiddling around, I ended up compiling webkit. But I face the same issues as @russdill. Is there any progress on this one?
Weston is running, gstreamer is running on top, glmark2 is running. Only webkit is lacking me

simonchen007, 2023-09-28:
@woutervanh  
Hi,
how can you fix-up `linking error on gbm_create_device function` errors  with libmali ? I'm facing the same issue when compiling.

woutervanh, 2023-09-28:
> @woutervanh Hi, how can you fix-up `linking error on gbm_create_device function` errors with libmali ? I'm facing the same issue when compiling.

It was a hacky way. If I remember well it was a matter of renaming libmali to libgbm etc (symbols are in there). But was a wrong approach, didn't work anyway.
Really take mainline, it's dead simple

simonchen007, 2023-09-28:
> It was a hacky way. If I remember well it was a matter of renaming libmali to libgbm etc (symbols are in there). But was a wrong approach, didn't work anyway. Really take mainline, it's dead simple

But the mainline panfrost + mesa3d + wayland  having poor performance then libmali. So, I want to have the libmali + wpewebkit.

woutervanh, 2023-09-28:
Did you test it? For me it was working very well on rk3288.
Nevertheless, I did not get wpewebkit to work on libmali

simonchen007, 2023-09-28:
> Did you test it? For me it was working very well on rk3288. Nevertheless, I did not get wpewebkit to work on libmali

Yes, I tested it for both rk3288 and rk3326(PX30) :
1. panfrost + mesa3d + wayland + wpewebkit(0.42.0) + cog(0.18.0) + libwpe( 1.15.1 ).
2. libmali + wayland + wpewebkit(2.38.6) + cog(0.16.0) + libwpe( 1.12.3 )

I point out one of  test result for pointing out the  performance match:
The rk3326(PX30)  glmark2-es2-wayland testing result are:
```txt
panfrost:
kernel 5.10 glmark2 Score: 117 
new version buildroot + kernel 5.10 glmark2 Score: 115 
new version buildroot + kernel 5.10 with new mesa3D 23.1.6 glmark2 Score: 130 
new version buildroot + kernel 5.10(upgrade panfrost) with new mesa3D 23.1.6 glmark2 Score: 139
new version buildroot + kernel 5.10(upgrade panfrost) with new mesa3D 23.1.6 + CPU performance glmark2 Score: 146
new version buildroot + kernel 5.10(upgrade panfrost) with new mesa3D 23.1.6 + CPU performance + GPU performance glmark2 Score: 209

Mali:
kernel 4.4 g2p0  glmark2 Score: 370 
kernel 5.10 g2p0 glmark2 Score: 197 
kernel 5.10 g13p0 glmark2 Score: 193 
kernel 5.10 g13p0 + CPU performance glmark2 Score: 256
kernel 5.10 g13p0 + GPU performance glmark2 Score: 193
kernel 5.10 g13p0 + CPU performance + GPU performance glmark2 Score: 256
```

For the browser（cog）, running this link https://threejs.org/examples/physics_ammo_cloth.html
```txt
panfrost + mesa3d + wayland + wpewebkit(0.42.0) + cog(0.18.0) + libwpe( 1.15.1 ) in wayland mode: 30 FPS

libmali + wayland + wpewebkit(2.38.6) + cog(0.16.0) + libwpe( 1.12.3 ) in drm mode: 60 FPS
```

I must use the wayland mode , because the fllowing:
1. I have some wayland patchs from rockchip offical. I can speed up the software running faster.
2. for some vertical display screen, we can use the RGA for rotation.

However,  wpewebkit(2.38.6) + cog(0.16.0) + libwpe( 1.12.3 ) not support the wayland mode, so I have to upgrade wpewebkit + cog + libwpe to last. when I compile it, I got the 'linking error on gbm_create_device function'.

closed 2024-01-04 by github-actions[bot]