Igalia/meta-webkit #503: Cog supports X11 but recipe not

Oquirella, 2024-07-03:


Oquirella, 2025-01-14:
Can someone answer?

psaavedra, 2025-05-20:
Yes, we should to add a 

```
PACKAGECONFIG[x11] = ",, \
    libxcb \
    xkeyboard-config \
    libxkbcommon \
    libx11 \
    xcb-util-cursor \
    wpebackend-fdo 
"
```

and maybe something like `PACKAGECONFIG ?= "${@bb.utils.filter('DISTRO_FEATURES', 'x11', d)} ..."`

clopez, 2025-09-02:
re-opening

clopez, 2025-11-21:
I have a WIP patch https://paste.rs/5LTAt (with color syntax https://paste.rs/5LTAt.diff )
But currently I don't have time to build and test it.
If you can try the patch and report if it works or not then I'm happy to merge it.

Oquirella, 2026-03-26:
root@0030D6453F19:~# ls -l /dev/dri/
total 0
drwxr-xr-x 2 root root         80 May 29 19:48 by-path
crw-rw---- 1 root video  226,   0 May 29 19:48 card0
crw-rw-rw- 1 root render 226, 128 May 29 19:48 renderD128

root@atlasco0030D6453F19:~# glxinfo  
name of display: :0 -> stuck here

root@0030D6453F19:~# es2_info 
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: MESA-LOADER: failed to open zink: /usr/lib/dri/zink_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/dri, suffix _dri)

EGL_VERSION: 1.4
EGL_VENDOR: Mesa Project
EGL_EXTENSIONS:
    EGL_ANGLE_sync_control_rate, EGL_KHR_cl_event2, EGL_KHR_config_attribs, 
    EGL_KHR_context_flush_control, EGL_KHR_create_context, 
    EGL_KHR_create_context_no_error, EGL_KHR_fence_sync, 
    EGL_KHR_get_all_proc_addresses, EGL_KHR_gl_colorspace, 
    EGL_KHR_gl_renderbuffer_image, EGL_KHR_gl_texture_2D_image, 
    EGL_KHR_gl_texture_3D_image, EGL_KHR_gl_texture_cubemap_image, 
    EGL_KHR_image_base, EGL_KHR_no_config_context, EGL_KHR_reusable_sync, 
    EGL_KHR_surfaceless_context, EGL_EXT_pixel_format_float, 
    EGL_KHR_wait_sync, EGL_MESA_configless_context, EGL_MESA_drm_image, 
    EGL_MESA_query_driver
EGL_CLIENT_APIS: OpenGL OpenGL_ES 
GL_VENDOR: Mesa
GL_VERSION: OpenGL ES 3.1 Mesa 24.0.7
GL_SHADING_LANGUAGE_VERSION: OpenGL ES GLSL ES 3.10
GL_RENDERER: softpipe
GL_EXTENSIONS:
    GL_EXT_blend_minmax, GL_EXT_multi_draw_arrays, 
    GL_EXT_texture_filter_anisotropic, GL_EXT_texture_compression_s3tc, 
    GL_EXT_texture_compression_dxt1, GL_EXT_texture_compression_rgtc, 
    GL_EXT_texture_format_BGRA8888, GL_OES_compressed_ETC1_RGB8_texture, 
    GL_OES_depth24, GL_OES_element_index_uint, GL_OES_fbo_render_mipmap, 
    GL_OES_mapbuffer, GL_OES_rgb8_rgba8, GL_OES_standard_derivatives, 
    GL_OES_stencil8, GL_OES_texture_3D, GL_OES_texture_float, 
    GL_OES_texture_float_linear, GL_OES_texture_half_float, 
    GL_OES_texture_half_float_linear, GL_OES_texture_npot, 
    GL_OES_vertex_half_float, GL_EXT_draw_instanced, 
    GL_EXT_texture_sRGB_decode, GL_OES_EGL_image, GL_OES_depth_texture, 
    GL_OES_packed_depth_stencil, GL_EXT_texture_type_2_10_10_10_REV, 
    GL_NV_conditional_render, GL_OES_get_program_binary, 
    GL_APPLE_texture_max_level, GL_EXT_discard_framebuffer, 
    GL_EXT_read_format_bgra, GL_NV_pack_subimage, GL_EXT_frag_depth, 
    GL_NV_fbo_color_attachments, GL_OES_EGL_image_external, GL_OES_EGL_sync, 
    GL_OES_vertex_array_object, GL_OES_viewport_array, 
    GL_ANGLE_pack_reverse_row_order, GL_ANGLE_texture_compression_dxt3, 
    GL_ANGLE_texture_compression_dxt5, GL_EXT_occlusion_query_boolean, 
    GL_EXT_texture_rg, GL_EXT_unpack_subimage, GL_NV_draw_buffers, 
    GL_NV_read_buffer, GL_NV_read_depth, GL_NV_read_depth_stencil, 
    GL_NV_read_stencil, GL_APPLE_sync, GL_EXT_draw_buffers, 
    GL_EXT_instanced_arrays, GL_EXT_map_buffer_range, GL_KHR_debug, 
    GL_KHR_texture_compression_astc_ldr, GL_NV_generate_mipmap_sRGB, 
    GL_NV_pixel_buffer_object, GL_OES_depth_texture_cube_map, 
    GL_OES_required_internalformat, GL_OES_surfaceless_context, 
    GL_EXT_color_buffer_float, GL_EXT_debug_label, GL_EXT_sRGB_write_control, 
    GL_EXT_separate_shader_objects, GL_EXT_shader_implicit_conversions, 
    GL_EXT_shader_integer_mix, GL_EXT_base_instance, 
    GL_EXT_compressed_ETC1_RGB8_sub_texture, GL_EXT_copy_image, 
    GL_EXT_draw_buffers_indexed, GL_EXT_draw_elements_base_vertex, 
    GL_EXT_gpu_shader5, GL_EXT_primitive_bounding_box, GL_EXT_render_snorm, 
    GL_EXT_shader_io_blocks, GL_EXT_texture_border_clamp, 
    GL_EXT_texture_buffer, GL_EXT_texture_cube_map_array, 
    GL_EXT_texture_norm16, GL_EXT_texture_view, GL_KHR_context_flush_control, 
    GL_NV_image_formats, GL_NV_shader_noperspective_interpolation, 
    GL_OES_copy_image, GL_OES_draw_buffers_indexed, 
    GL_OES_draw_elements_base_vertex, GL_OES_gpu_shader5, 
    GL_OES_primitive_bounding_box, GL_OES_shader_io_blocks, 
    GL_OES_texture_border_clamp, GL_OES_texture_buffer, 
    GL_OES_texture_cube_map_array, GL_OES_texture_stencil8, 
    GL_OES_texture_storage_multisample_2d_array, GL_OES_texture_view, 
    GL_EXT_blend_func_extended, GL_EXT_buffer_storage, GL_EXT_float_blend, 
    GL_EXT_geometry_point_size, GL_EXT_geometry_shader, 
    GL_EXT_texture_sRGB_R8, GL_EXT_texture_sRGB_RG8, GL_KHR_no_error, 
    GL_KHR_texture_compression_astc_sliced_3d, GL_OES_EGL_image_external_essl3, 
    GL_OES_geometry_point_size, GL_OES_geometry_shader, 
    GL_OES_shader_image_atomic, GL_EXT_clear_texture, 
    GL_EXT_clip_cull_distance, GL_EXT_disjoint_timer_query, 
    GL_EXT_texture_compression_s3tc_srgb, GL_MESA_shader_integer_functions, 
    GL_EXT_clip_control, GL_EXT_color_buffer_half_float, 
    GL_EXT_texture_compression_bptc, GL_EXT_texture_mirror_clamp_to_edge, 
    GL_KHR_parallel_shader_compile, GL_EXT_EGL_image_storage, 
    GL_MESA_framebuffer_flip_y, GL_EXT_depth_clamp, GL_EXT_texture_query_lod, 
    GL_MESA_sampler_objects, GL_MESA_bgra

clopez, 2026-03-27:
> I installed libgles2-mesa now the minibrowser works, is this a missing dependency?

AFAIK Yocto should automatically pull any library that the binary links against, I have no idea why it didn't did do that in your case.